arendst/Tasmota #17896: GPS not registering correct date when no network present

pkkrusty, 2023-02-06:
### PROBLEM DESCRIPTION
Connected a GPS module to ESP8266. When ESP8266 is restarted and the GPS module already has a signal, the GPS will correctly set the date/time of the system on startup. However, if the ESP8266 is cold started (and thus the GPS is cold-started), the system date remains at 1-Jan-1970 (as expected) until GPS receives a good signal, then turns into 7-Jan-1980 with the correct time (issue opened on 6-Feb-2023, 2248 weeks or 1,359,590,400 seconds from the 1980 date).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):Wemos D1 Mini with ublox 6M GNSS module
- [x] Tasmota binary firmware version number used: 12.3.1.5
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:00:24.322 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
00:00:24.325 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:00:24.341 WIF: DNS resolved '192.168.1.1' (192.168.1.1) in 0 ms
00:00:25.349 NTP: No reply from 192.168.1.1
00:00:25.512 SRC: Backlog
00:00:25.514 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
00:00:25.516 RSL: RESULT = {"Module":{"18":"Generic"}}
00:00:25.518 RTC: UTC 1980-01-07T14:33:00, DST 1980-03-30T02:00:00, STD 1980-10-26T03:00:00
14:33:00.000 RTC: Synced by UBX
14:33:00.215 SRC: Backlog
14:33:00.217 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
14:33:00.222 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"4352":"GPS RX"},"GPIO5":{"4384":"GPS TX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:33:38.763 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"GPS Test","FriendlyName":["GPS Test"],"Topic":"gps-test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:33:38.766 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:00:58","StartupUTC":"1980-01-07T14:32:40","Sleep":50,"CfgHolder":4802,"BootCount":8,"BCResetTime":"2023-02-05T14:37:41","SaveCount":16,"SaveAddress":"3FB000"}}
14:33:38.769 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1.5(tasmota)","BuildDateTime":"2023-02-06T14:07:57","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"404/699"}}
14:33:38.772 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":3,"LogHost":"192.168.1.23","LogPort":514,"SSId":["Roatan","pkkrusty"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00208000","2805C80001000600003C5A00192800000000","000200C9","08006000","00004000","00000000"]}}
14:33:38.777 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":547,"Free":476,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","05000482","04080001","00000003","00000000","40400001","00000080","00200000","50000000","00000080"],"Drivers":"1,2,4,9,10,24,38,50","Sensors":"1,2,60"}}
14:33:38.780 RSL: STATUS5 = {"StatusNET":{"Hostname":"gps-test","IPAddress":"192.168.1.55","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.23","DNSServer2":"0.0.0.0","Mac":"8C:CE:4E:CF:21:B8","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:33:38.782 RSL: STATUS7 = {"StatusTIM":{"UTC":"1980-01-07T14:33:38","Local":"1980-01-07T14:33:38","StartDST":"1980-03-30T02:00:00","EndDST":"1980-10-26T03:00:00","Timezone":"+00:00"}}
14:33:38.785 RSL: STATUS10 = {"StatusSNS":{"Time":"1980-01-07T14:33:38","GPS":{"lat":0.0000000,"lon":0.0000000,"alt":0.000,"hAcc":0.000,"vAcc":0.000}}}
14:33:38.788 RSL: STATUS11 = {"StatusSTS":{"Time":"1980-01-07T14:33:38","Uptime":"0T00:00:58","UptimeSec":58,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Roatan","BSSId":"BC:A5:11:04:25:8D","Channel":11,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
- [ ] Normal operation
```lua
00:00:00.002 HDW: ESP8266EX
00:00:00.016 UFS: FlashFS mounted with 2992 kB free
00:00:00.051 CFG: Loaded from File, Count 20
00:00:00.054 QPC: Count 1
00:00:00.059 Project pkkrusty - GPS Test Version 12.3.1.5(tasmota)-2_7_4_9(2023-02-06T14:07:57)
00:00:00.060 UFILESYSTEM OK!
00:00:00.449 WIF: Checking connection...
00:00:00.458 WIF: Attempting connection...
00:00:00.479 WIF: Network (re)scan started...
00:00:00.642 RTC: UTC 2023-02-06T14:41:42, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
14:41:42.000 RTC: Synced by UBX
14:41:42.394 WIF: Checking connection...
14:41:42.404 WIF: Attempting connection...
14:41:43.396 WIF: Checking connection...
14:41:43.404 WIF: Attempting connection...
[... 99 lines cut ...]
00:00:23.188 WIF: DNS resolved '192.168.1.23' (192.168.1.23) in 0 ms
00:00:24.191 NTP: No reply from 192.168.1.23
00:00:24.256 NTP: Sync time...
00:00:24.257 WIF: DNS resolved '192.168.1.1' (192.168.1.1) in 0 ms
00:00:25.269 NTP: No reply from 192.168.1.1
00:00:25.272 RTC: UTC 1980-01-07T14:43:41, DST 1980-03-30T02:00:00, STD 1980-10-26T03:00:00
14:43:41.001 RTC: Synced by UBX
```
Note that this UBX sync happens before the LED on the GPS module starts blinking.
### TO REPRODUCE
_Steps to reproduce the behavior:_
Restart 1 with GPS already sending data gives normal date/time. Detaching and reattaching power results in GPS module reporting 7-Jan-1980 date with correct time, even after GPS starts sending location data. Maybe due to time sync before full fix is available? The on-board battery in the module is holding only time and not date?

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
UBX sync gives correct date/time whether hot start or cold start.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Webserver seems to crash regularly while running the GPS driver. Host is still alive, responds to ping, logs data, but can't load any pages.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-02-06:
Could you please compile with `#define DEBUG_TASMOTA_SENSOR` and re-post the logs with the additional informations ?

Staars, 2023-02-06:
Addionally you could try to restrict the NTP server to only reply with a valid GPS fix by changing line 725 to:
` if(!TasmotaGlobal.global_state.network_down && UBX.mode.runningNTP && UBX.state.gpsFix > 1){`

pkkrusty, 2023-02-06:
In this use case, I just need the ESP8266 to set its own time correctly, since I'm logging GPS data and need the time stamp. Not using the NTP feature. Maybe this line affects just internal usage also?

pkkrusty, 2023-02-08:
OK, more logging info:

Cold start, GPS reports time as 3236-1-12 23:59:48, then rolls over at "midnight" to 1980-01-06 00:00:00:
```
00:00:01.098 UBX: got 34 bytes, non-empty-loop: 1

00:00:01.399 WIF: Checking connection...

00:00:01.400 WIF: Attempting connection...

00:00:01.910 UBX: got NAV_STATUS

00:00:01.918 UBX: gpsFix: 0, valid: 0

00:00:02.004 UBX: got NAV_POSLLH

00:00:02.005 UBX: got NAV_TIME_UTC

00:00:02.005 UBX: got 95 bytes, non-empty-loop: 1

00:00:02.405 WIF: Checking connection...

00:00:02.406 WIF: Attempting connection...

00:00:02.912 UBX: got NAV_STATUS

00:00:02.921 UBX: gpsFix: 0, valid: 0

00:00:03.007 UBX: got NAV_POSLLH

00:00:03.007 UBX: iTOW: 3000

00:00:03.008 UBX: no valid position data

00:00:03.106 UBX: got NAV_TIME_UTC

00:00:03.107 UBX: UTC-Time: 3236-1-12 23:59:48

00:00:03.406 WIF: Checking connection...

00:00:03.406 WIF: Attempting connection...

00:00:03.407 WIF: Network 0, AP1, SSId Roatan, Channel 1, BSSId 28:DE:A8:AB:9D:68, RSSI -58, Encryption 1

00:00:03.412 WIF: Network 1, AP-, SSId Roatan-2, Channel 6, BSSId C0:4A:00:40:5F:E7, RSSI -69, Encryption 1

00:00:03.758 WIF: Connecting to AP1 Roatan Channel 1 BSSId 28:DE:A8:AB:9D:68 in mode 11n as gps-test...

00:00:03.936 UBX: got NAV_STATUS

00:00:03.937 UBX: gpsFix: 0, valid: 0

00:00:04.036 UBX: got NAV_TIME_UTC

00:00:04.037 UBX: UTC-Time: 3236-1-12 23:59:49

00:00:04.639 WIF: Checking connection...

00:00:04.640 WIF: Attempting connection...

00:00:05.002 UBX: got NAV_STATUS

00:00:05.003 UBX: gpsFix: 0, valid: 0

00:00:05.102 UBX: got NAV_POSLLH

00:00:05.103 UBX: iTOW: 5000

00:00:05.103 UBX: no valid position data

[... 482 lines cut ...]

10:38:59.225 UBX: gpsFix: 0, valid: 0

10:38:59.303 UBX: got NAV_POSLLH

10:38:59.304 UBX: iTOW: 297554000

10:38:59.305 UBX: no valid position data

10:38:59.403 UBX: got NAV_TIME_UTC

10:38:59.404 UBX: UTC-Time: 2023-2-8 10:38:59

10:38:59.406 UBX: UTC-Time is valid

10:39:00.213 UBX: got NAV_STATUS

10:39:00.232 UBX: gpsFix: 0, valid: 0

10:39:00.307 UBX: got NAV_POSLLH

10:39:00.308 UBX: iTOW: 297555000
```

I suppose I could enable the option to force time update on every valid time received (Sensor60 11), but seems like ideal would be to restrict valid time to >1554595182 (2019-04-06 23:59:42, the latest GPS week rollover).

Staars, 2023-02-08:
Makes sense to me.
Next rollover is a bit to go, I just think about to deny times, that are older than the firmware build time in general.

pkkrusty, 2023-02-11:
Do you want to submit a PR for that? I'm not sure where exactly in the code that check would go.

Separately, I noticed in your drift filter Line 632, you have a check:

` if ((UBX.Message.navPosllh.lat-UBX.rec_buffer.values.lat<abs(UBX_LAT_LON_THRESHOLD))||(UBX.Message.navPosllh.lon-UBX.rec_buffer.values.lon<abs(UBX_LAT_LON_THRESHOLD))) {`

but I think maybe it should be:

`  if ((abs(UBX.Message.navPosllh.lat-UBX.rec_buffer.values.lat)<UBX_LAT_LON_THRESHOLD)||(abs(UBX.Message.navPosllh.lon-UBX.rec_buffer.values.lon)<UBX_LAT_LON_THRESHOLD)) {`

 since you want to compare movement in any direction to the threshold.

barbudor, 2023-02-11:
> deny times, that are older than the firmware build time in general.

I think making the test based on the build date is perfect but over-engineered
Testing against <2023 should be fine and could updated in a few years if needed

Staars, 2023-02-11:
@pkkrusty I agree, your change should be an improvement.

@barbudor I agree too, hardcoding 2023 should be perfectly fine.

Feel free to open a PR. My spare time is completely absorbed by a pretty large subproject for Tasmota at the moment, that’s why I can not really do testing in this case in the next days.

barbudor, 2023-02-11:
Done for both issue : https://github.com/arendst/Tasmota/pull/17932

Your review is welcome @pkkrusty and @Staars

pkkrusty, 2023-02-11:
Was also thinking about making new_position available to users (getting passed into JSON messages like lat and lon) so logic could be performed based on it (i.e. only logging if new_position is true). I adjusted the code to make new_position global in the driver, and it takes its value from the bool function `UBXHandlePOSLLH()` so I would assume new_position would be either true or false (1/0), but in WebUI, it shows as 1073422080. JSON is displayed correctly as 1/0. I'll PR after yours is merged so you can take a look at it and see what I'm missing.

Also added a dynamic adjustment for `UBX_LAT_LON_THRESHOLD`, since common cheap GPS boards will get down to 2m horizontal accuracy and 5 decimal digits in GPS coordinates is 1.11m, so a multiplier of 50 seems to be a good threshold. That filters any movement below (above?) 5 decimal digits, which is noise anyway. In my tests with 5m accuracy, a threshold of 100 was very stable in detecting movement correctly.

barbudor, 2023-02-11:
Assuming a bool is 0 or 1 is a shortcut I wouldn't take
Per C definition false is 0, true is anything non 0
Depending how you are using the return value of `UBXHandlePOSLLH()` for the JSON and for the GUI, you may get different values depending how the compiler is re-casting
To be 100% sure `(mybool ? 1: 0)` will give you the expected result

closed 2023-02-12 by pkkrusty