arendst/Tasmota #17905: Websend action execution randomly delayed and very inconsistent.

digital-spinner, 2023-02-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 2.5
- [X] Tasmota binary firmware version number used: 12.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:49:09.581 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"193":"Switch_n2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"192":"Switch_n1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ1"},"GPIO17":{"4736":"ADC Temp"}}
08:49:22.005 RSL: STATE = {"Time":"2023-02-07T08:49:22","Uptime":"0T08:51:08","UptimeSec":31868,"Heap":21,"SleepMode":"Normal","Sleep":0,"LoadAvg":999,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"DigitallHOME-R","BSSId":"C0:C9:E3:3B:5A:4E","Channel":7,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}
08:49:22.017 RSL: SENSOR = {"Time":"2023-02-07T08:49:22","Switch1":"ON","Switch2":"ON","ANALOG":{"Temperature":57.0},"ENERGY":{"TotalStartTime":"2023-02-06T23:57:43","Total":0.107,"Yesterday":0.095,"Today":0.011,"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":94,"Free":417,"Rules":"ON Switch2#State==2 DO Websend [192.168.102.32] /?device=relay&name=light&command=toggle ENDON"}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:50:53.681 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_82915D","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":0,"SwitchTopic":"0","SwitchMode":[6,6,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:50:53.692 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T08:52:39","StartupUTC":"2023-02-06T22:58:14","Sleep":0,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2023-02-06T23:57:43","SaveCount":7,"SaveAddress":"F5000"}}
08:50:53.700 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:28:17","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"406/699"}}
08:50:53.708 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DigitallHOME-R",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008000","1405C80001000600003C5A0A192800000000","00000480","00006000","00004000","00000000"]}}
08:50:53.723 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
08:50:53.734 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-82915D-4445","IPAddress":"192.168.102.31","Gateway":"192.168.100.1","Subnetmask":"255.255.252.0","DNSServer1":"192.168.100.1","DNSServer2":"0.0.0.0","Mac":"48:3F:DA:82:91:5D","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:50:53.744 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-02-07T07:50:53","Local":"2023-02-07T08:50:53","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:11","Sunset":"17:56"}}
08:50:53.751 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:50:53.764 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-02-07T08:50:53","Switch1":"ON","Switch2":"ON","ANALOG":{"Temperature":57.0},"ENERGY":{"TotalStartTime":"2023-02-06T23:57:43","Total":0.107,"Yesterday":0.095,"Today":0.011,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}}
08:50:53.774 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-02-07T08:50:53","Uptime":"0T08:52:39","UptimeSec":31959,"Heap":21,"SleepMode":"Normal","Sleep":0,"LoadAvg":999,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"DigitallHOME-R","BSSId":"C0:C9:E3:3B:5A:4E","Channel":7,"Mode":"11n","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

08:53:39.412 ADE: ACCMODE 0x2D1C00, VRMS 6221316, 6221125, Period 4474, 4474, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:40.412 ADE: ACCMODE 0x2D1C00, VRMS 6218538, 6217910, Period 4474, 4474, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:41.412 ADE: ACCMODE 0x2D1C00, VRMS 6218244, 6216633, Period 4474, 4474, IRMS 1773, 1773, WATT 0, 0, VA 28, 29, VAR 0, 0
08:53:42.412 ADE: ACCMODE 0x2D1C00, VRMS 6217546, 6216069, Period 4474, 4474, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:43.412 ADE: ACCMODE 0x2D1C00, VRMS 6217880, 6215831, Period 4474, 4474, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:44.412 ADE: ACCMODE 0x2D1C00, VRMS 6227744, 6225268, Period 4474, 4474, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:45.412 ADE: ACCMODE 0x2D1C00, VRMS 6217980, 6215273, Period 4474, 4474, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:46.412 ADE: ACCMODE 0x2D1C00, VRMS 6215761, 6212896, Period 4473, 4473, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:47.412 ADE: ACCMODE 0x2D1C00, VRMS 6218623, 6214911, Period 4473, 4473, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:48.412 ADE: ACCMODE 0x2D1C00, VRMS 6212297, 6208432, Period 4473, 4473, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:49.411 ADE: ACCMODE 0x2D1C00, VRMS 6221049, 6217161, Period 4473, 4473, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:49.621 RUL: SWITCH2#STATE==2 performs "Websend [192.168.102.32] /?device=relay&name=light&command=toggle"
08:53:49.623 SRC: Rule
08:53:49.626 CMD: Grp 0, Cmd 'WEBSEND', Idx 1, Len 57, Pld -99, Data '[192.168.102.32] /?device=relay&name=light&command=toggle'
08:53:50.412 ADE: ACCMODE 0x2D1C00, VRMS 6217970, 6213865, Period 4472, 4473, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:51.264 RSL: RESULT = {"WebSend":"Done"}
08:53:51.412 ADE: ACCMODE 0x2D1C00, VRMS 6218963, 6215299, Period 4473, 4473, IRMS 1773, 1773, WATT 0, 0, VA 28, 28, VAR 0, 0
08:53:51.792 RUL: SWITCH2#STATE==2 performs "Websend [192.168.102.32] /?device=relay&name=light&command=toggle"
08:53:51.794 SRC: Rule
[... 70 lines cut ...]

# apply template and best settings
Backlog Template {"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18};
 SwitchMode1 6; SwitchMode2 6; Backlog SetOption0 0; SetOption3 0; PowerOnState 0; SetOption32 20; SaveData 0; Module 0; Restart 1

# apply rule to be used with pushbutton 2
Rule1 ON Switch2#State==2 DO Websend [192.168.102.32] /?device=relay&name=light&command=toggle ENDON
Rule1 1

[... cut ...]
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I need to control another device which can accept HTTP GET requests. Since my Blebox device died I switched to Shelly 2.5 and Tasmota 12.3.1 for it (also tested with 9.2.0) but the described issue raised. I have also tested WebQuery command and it behaves in the same way.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-02-07:
Tasmota is designed to be used with mqtt. Do not expect near real time behaviour when using http.

closed 2023-02-07 by Jason2866

digital-spinner, 2023-02-07:
Hola, wait a minute! I didn't say anything about the real time behavior and don't be rude man. The WebSend is just slow as hell comparing to other solutions and I'm not even mentioning the random delay of it's processing capabilities and occasional connection issues of that. If this is the case just remove the whole websend / webquery stuff from it so there will be no confusion of any kind. I just bought those devices only because of this functionality. And yea, I can go with MQTT later, but I hope you understand the confusion?

And the last one thing at the end: Tasmota is great without the doubts and it's power comes from "simplicity".

barbudor, 2023-02-07:
Hi @digital-spinner
I see 2 parts from you issue
- delayed rule execution
- websend execution

I would suggest first to treat them separately starting with the rule
If you replace the websend by something purely internal to Tasmota like incrementing a variable or toggling a native relay on gpio, what is your experience?
Do you see delay? Randomness?

Regarding the delay on websend (WebQuery could seem more logical for non tasmota target although it's a matter or syntax as it's the same code behind) you would need an analysis at a network level which is not easy to qualify if the delay is from Tasmota or the device or the network

digital-spinner, 2023-02-07:
If i just type the Websend in the console it has the same random delay behave. I mean, the first time after long period of not doing anything it *usually* fires immediately, but after even like 15 seconds next press (or next console command) is a lotery - it may execute like within 2-3 seconds or it will execute like after 5 or so. It looks like it can't be used more often than once per 10 seconds at least. And again it doesn't happen with some other devices (Blebox with thier API capabilities) so that is why I'm mentioning it here at first place. It is a possibility to improve if of course can be done.

barbudor, 2023-02-07:
If I understand well what you describe, it's mostly pointing in delay in the whole http communication
Definitively needs network analysis (tcpdump/wireshark). If you have an http workflow from another device than Tasmota, both need capture to allow comparison

Improvement can only occur if something missing is identified, and identification can only come from your size (your network, you device)

digital-spinner, 2023-02-07:
I think the issue is with scheduling internal execution with Tasmota itself not the network issues, but for my curiosity I will check another Tasmota device if this is a case and let know if anything is changed for worse or better but.. when I click toggle on web panel it reacts immediately for those power outputs which it has (all my devices reacts momentarily), also relays reacts immediately on buttons reactions. Maybe there is a way to prioritize the HTTP over MQTT (an option switch for that) so the HTTP will have a precedence if it consumes some CPU cycles which is less and less after each action because MQTT is working or some other stuff (but I have the MQTT disable on my devices currently).

Anyway it makes sense to have a low priority for the HTTP API support if the MQTT solution is aimed for speed and reliability..
I think I'm gonna get rid of the HTTP devices and put the MQTT to use but i need some time to do that and I have already like 3 other projects to work on :-(

sfromis, 2023-02-07:
The "low priority" for HTTP is not about scheduling or CPU cycles, but about implementation. It is "tacked on", but by far not seen as a "primary strategy"

Not much has been done to create "optimal" implementations of using HTTP, even on the inherently slower terms of using this protocol, which is based on starting and closing sessions per request, instead of MQTT using a persistent session, leading to fewer network round trips.

digital-spinner, 2023-02-07:
@sfromis thanks for the explanation, good to know that. Just one additional question to that - just wonder about one thing (don't know how it is done here) are those MQTT messeges acknowledged or not (sent by Tasmota). I know it can be both ways with MQTT right? So the brocker ACK can also  be problematic if I understand correctly. Also MQTT uses TCP so the connection itself needs to be acknowledged.

sfromis, 2023-02-07:
While MQTT uses TCP, each client has one persistent connection while available, until outages of either end or network. Ideally up to years, but at the least it should be hours.

As Tasmota only uses QoS 0 "at most once" with MQTT, delivery is not assured.
> The minimal QoS level is zero. This service level guarantees a best-effort delivery. There is no guarantee of delivery. The recipient does not acknowledge receipt of the message and the message is not stored and re-transmitted by the sender. QoS level 0 is often called “fire and forget” and provides the same guarantee as the underlying TCP protocol.

In practice, HTTP provides no better delivery assurance, but the requestor does have the option of checking for a response.

barbudor, 2023-02-07:
I just played alittle game between an ESP32-DevKit and an ESP8266-NoceMCU
Each have a relay
Each have a rule like `on power1#state do websend [ip of the ohter] power toggle endon` 
That's the Tamsota version of the stupid machine

Here is the log I have:
```
21:58:07.387 RUL: POWER1#STATE performs "websend [192.168.168.195] power toggle"
21:58:07.508 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{"POWER":"OFF"}}
21:58:07.516 MQT: stat/esp32dev3/WEBSEND = {"WebSend":"Done"}
21:58:07.595 MQT: stat/esp32dev3/POWER = {"POWER":"ON"}
21:58:07.601 MQT: stat/esp32dev3/POWER = ON (retained)
21:58:07.640 RUL: POWER1#STATE performs "websend [192.168.168.195] power toggle"
21:58:07.753 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{"POWER":"ON"}}
21:58:07.761 MQT: stat/esp32dev3/WEBSEND = {"WebSend":"Done"}
21:58:07.839 MQT: stat/esp32dev3/POWER = {"POWER":"OFF"}
21:58:07.842 MQT: stat/esp32dev3/POWER = OFF (retained)
21:58:07.875 RUL: POWER1#STATE performs "websend [192.168.168.195] power toggle"
21:58:08.014 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{"POWER":"OFF"}}
21:58:08.023 MQT: stat/esp32dev3/WEBSEND = {"WebSend":"Done"}
21:58:08.107 MQT: stat/esp32dev3/POWER = {"POWER":"ON"}
21:58:08.112 MQT: stat/esp32dev3/POWER = ON (retained)
21:58:08.146 RUL: POWER1#STATE performs "websend [192.168.168.195] power toggle"
21:58:08.263 MQT: stat/esp32dev3/WEBQUERY = {"WebQuery":{"WebQuery":}
21:58:08.272 MQT: stat/esp32dev3/WEBSEND = {"WebSend":"Done"}
```
So the round-trip-time is less than 300 ms for one tasmota to send the command and the other to sendback the command
And roughly less htan 100ms between the "WebSend Done" and receiving the reply command which is rule execution + Websend execution.

I don't see any performance problem on Tasmota
The results are consistent mnutes after minutes

So as sfromis pointed out, it's most likely due to some delay in the network or the other device

Now, we don't know what your Shelly is doing at the same time, so can't judge is tasmota could be busy doing something else.
My devkit+nodemcu are only doing that

May be you can try same test between your Shelly and anohter Tamsota ?

barbudor, 2023-02-07:
And for the completness of the test, let's do it replacing the WebSend by a ` publish cmnd/othertopic/power toggle`

```
22:05:33.311 MQT: cmnd/nodemcu/power = toggle
22:05:33.487 MQT: stat/esp32dev3/POWER = {"POWER":"ON"}
22:05:33.490 MQT: stat/esp32dev3/POWER = ON (retained)
22:05:33.502 RUL: POWER1#STATE performs "publish cmnd/nodemcu/power toggle"
22:05:33.509 MQT: cmnd/nodemcu/power = toggle
22:05:33.687 MQT: stat/esp32dev3/POWER = {"POWER":"OFF"}
22:05:33.690 MQT: stat/esp32dev3/POWER = OFF (retained)
22:05:33.702 RUL: POWER1#STATE performs "publish cmnd/nodemcu/power toggle"
22:05:33.709 MQT: cmnd/nodemcu/power = toggle
22:05:33.788 MQT: stat/esp32dev3/POWER = {"POWER":"ON"}
22:05:33.795 MQT: stat/esp32dev3/POWER = ON (retained)
22:05:33.807 RUL: POWER1#STATE performs "publish cmnd/nodemcu/power toggle"
22:05:33.813 MQT: cmnd/nodemcu/power = toggle
22:05:33.987 MQT: stat/esp32dev3/POWER = {"POWER":"OFF"}
22:05:33.994 MQT: stat/esp32dev3/POWER = OFF (retained)
22:05:34.002 RUL: POWER1#STATE performs "publish cmnd/nodemcu/power toggle"
22:05:34.010 MQT: cmnd/nodemcu/power = toggle
22:05:34.083 MQT: stat/esp32dev3/POWER = {"POWER":"ON"}
22:05:34.091 MQT: stat/esp32dev3/POWER = ON (retained)
22:05:34.102 RUL: POWER1#STATE performs "publish cmnd/nodemcu/power toggle"
22:05:34.106 MQT: cmnd/nodemcu/power = toggle
```

200ms round-trip-time
So a ~30% performance improvement of MQTT over HTTP
Which is not suprising based on the asynchonicity of MQTT vs HTTP

barbudor, 2023-02-07:
[image]

digital-spinner, 2023-02-07:
OK, thank you.

One thing I have tested additionally is another Shelly 2 with Tasmota 9.2.0 and it behave in the same way. There is also a random lag after button press. For comparison Blebox switches also are executing api endpoint in the same way but I did not noticed any random lag with those. Of course when I click like crazy 3 times fast then even those will loose one payload etc. but there was no random lag, it was actually expected after multiple presses (I guess it was doing also something) but not like in my Shellies now.

I think I will migrate to MQTT someday, for now I can live with that as it is.

If you can point me what network tests I can do than I will try to do it.
My network has fiber WAN connection attached to OPNSense gateway, which serves through the switch 3 APs and some other cable clients. Never had an issue with that - it is rock solid and serves also as a VPN gateway (this is the main reason why it is at the first place). My work laptops have 200+Mbps over WiFi in every case and the workstation easily can utilize 930Mbps over cable. The DNS is dynamically managed but I set the manual assignations, so every device gets its own IP address. Ah, the subnet is 22 if this matter. The APs are just serving WiFi in FAST ROAM mode using OpenWRT firmware and doing nothing more. I have over 40 wifi IOT devices if I remember. I hope that I won't start glowing any soon.

barbudor, 2023-02-07:
The best would be to be able to do tcpdump/wirechark captures (I believe I already said that above)
But if the addressed device is an embedded device and not a server, I don't know how to achieve that
In some case, it is possible to use some wifi card in PCs in "promiscuous mode" which is a kind of "spy" mode where the would capture everything that goes on the wifi network.
But my understanding that this feature is not available with every wifi card

digital-spinner, 2023-02-17:
OK, one observation extra to that..
When I press both buttons simultaneously, then the random delay applies for both, the physical one and the WebSend wired button also. There is definitely some issue here.

When I have both buttons wired physically (without any rules) then there is no delay.