arendst/Tasmota #17922: ESP32 Temp Sensor for Sonoff Dual R3 Removed?

stoffies00711, 2023-02-08:
### PROBLEM DESCRIPTION
I just upgraded one of my Sonoff DualR3's to the latest release from 12.1.1 to 12.3.1 via OTA
I now noticed that the ESP Temperature Sensor is not available anymore. 
The temperature sensor has never been available on the POWR2 and POWR3 units. Only the DualR3 but now that's missing too.

Please forgive a newb here but has this been reported before?  Will it be implemented again and on the other models as well?

Thanks and my apologies if I am not using this template correctly.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

joba-1, 2023-02-08:
Use setoption146 1 to reenable it.
Display was disabled by default due to being notoriously inaccurate.

sfromis, 2023-02-09:
In most cases, not just "inaccurate", but just not working. Useless.

stoffies00711, 2023-02-09:
Hey @joba-1 
**Thank you.  It works.** 

I have automations to switch off outlets in case they overheat for some reason. Hasn't happened yet but you never know...
I am willing to accept "inaccurate" for now.  I am slowly replacing them with Shelly's.

sfromis, 2023-02-09:
Again, it is not merely "inaccurate", but actively misleading.

joba-1, 2023-02-09:
well, stoffies had it working. Same with all my esp32 in different devices (from esp32-dev, esp32-cam, ... to DualR3s) have the same problem of being extremely inaccurate, but never not working at all. Of course that still can be luck.

To give an example: they mostly show temperatures from 50-70°C for slightly warm (max 30°C). And figures all rise when they really get warmer. I see no reason why they should change behaviour.
So establishing a "normal" temperature range separate for each device worked for me to notice nonstandard situations. 

Except for the tasmota-DualR3 that still sometimes goes into full isolation mode (no network access until power cycle) for no apparent reason.

Anyways, I would not use any temperature sensor for cirquits mounted to wood panels and bet my life on it (if I had to, I would use NTCs). And I would not rely on any esp for critical stuff like that - just as nice addition while it works). I just find a better, fireproof way of mounting.

sfromis, 2023-02-09:
With two ESP32 running at the same time having something like a 40 degrees deviation, it is way beyond saying "inaccurate".