arendst/Tasmota #17923: NOUS A1T can't resolve DNS requests

carow-it, 2023-02-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After setting up two NOUS A1T one of the devices can't resolve dns requests.
Defective device
```lua
00:22:11.287 CMD: webquery http://homeassistant.lan:8123 get
00:22:11.290 SRC: WebConsole from 192.168.69.70
00:22:11.293 CMD: Grp 0, Cmd 'WEBQUERY', Idx 1, Len 33, Pld -99, Data 'http://homeassistant.lan:8123 get'
00:22:21.674 RSL: RESULT = {"WebQuery":"Connect failed"}
```
Working Device:
```lua
23:03:03.397 CMD: webquery http://homeassistant.lan:8123 get
23:03:03.399 SRC: WebConsole from 192.168.69.70
23:03:03.402 CMD: Grp 0, Cmd 'WEBQUERY', Idx 1, Len 33, Pld -99, Data 'http://homeassistant.lan:8123 get'
23:03:03.424 MQT: stat/tasmota_181037/RESULT = {"WebQuery":"Done"}
```
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NOUS A1T
- [x] Tasmota binary firmware version number used: 12.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
00:03:43.887 SRC: WebConsole from 192.168.69.70
00:03:43.889 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
00:03:43.921 SRC: Backlog
00:03:43.924 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
00:03:43.929 RSL: RESULT = {"NAME":"NOUS A1T","GPIO":[32,0,0,0,2720,2656,0,0,2624,320,224,0,0,0],"FLAG":0,"BASE":49}
00:03:44.138 SRC: Backlog
00:03:44.141 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
00:03:44.144 RSL: RESULT = {"Module":{"0":"NOUS A1T"}}
00:03:44.200 HLW: EPC 244, CFlen 4011 usec
00:03:44.389 SRC: Backlog
00:03:44.392 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
00:03:44.400 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
```
- [x] Provide the output of this command: `Status 0`:
```lua
00:02:52.382 SRC: WebConsole from 192.168.69.70
00:02:52.385 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
00:02:52.390 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Stecker-EG-Server-Allnet","FriendlyName":["Stecker-EG-Server-Allnet"],"Topic":"tasmota_182BAD","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:02:52.424 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:45","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"1970-01-01T00:00:00","SaveCount":27,"SaveAddress":"F9000"}}
00:02:52.455 RSL: STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:28:17","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"429/699"}}
00:02:52.473 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CW-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
00:02:52.505 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1460C4","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
00:02:52.539 RSL: STATUS5 = {"StatusNET":{"Hostname":"Stecker-EG-Server-Allnet","IPAddress":"192.168.30.39","Gateway":"192.168.30.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.30.254","DNSServer2":"0.0.0.0","Mac":"48:55:19:18:2B:AD","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:02:52.563 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"homeassistant.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_182BAD","MqttUser":"mqtt","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:02:52.585 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:02:52","Local":"1970-01-01T00:02:52","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:02:52.602 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:02:52.620 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:02:52","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":18.459,"Yesterday":0.000,"Today":18.459,"Power":423,"ApparentPower":504,"ReactivePower":273,"Factor":0.84,"Voltage":300,"Current":1.682}}}
00:02:52.641 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:02:52","Uptime":"0T00:02:45","UptimeSec":165,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"CW-IoT","BSSId":"CC:88:C7:16:98:C2","Channel":1,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 26
00:00:00.056 QPC: Count 1
00:00:00.062 ROT: Mode 1
[... 196 lines cut ...]
00:01:48.200 HLW: EPC 239, CFlen 4147 usec
00:01:49.200 HLW: EPC 231, CFlen 4279 usec
00:01:50.200 HLW: EPC 233, CFlen 4277 usec
00:01:51.200 HLW: EPC 239, CFlen 4243 usec
00:01:52.200 HLW: EPC 231, CFlen 4397 usec
00:01:53.200 HLW: EPC 229, CFlen 4047 usec
00:01:54.200 HLW: EPC 242, CFlen 4170 usec
00:01:55.198 HLW: EPC 231, CFlen 4283 usec
00:01:56.200 HLW: EPC 229, CFlen 4334 usec
00:01:57.200 HLW: EPC 236, CFlen 4165 usec
00:01:58.200 HLW: EPC 232, CFlen 4284 usec
00:01:59.200 HLW: EPC 230, CFlen 4261 usec
00:02:00.200 HLW: EPC 242, CFlen 4106 usec
```

### TO REPRODUCE

### EXPECTED BEHAVIOUR
Both devices should resolve dns request the same.

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-02-08:
Are you using static IP or DHCP ? Are you sure they are not both on the same IP ? or does the problem occurs also if only one is powered ?

If it works with one and not with the other it either a configuration error or a faulty hardware
As the device seems to boot well and connect to wifi, I would likely vote for a misconfiguration error.

In case of a doubt `reset 1` and restart configuration from scratch

I see that you don't have an external DNS but only the one provided by your router.
I would advise to have an external DNS as backup such as 1.1.1.1 or 8.8.8.8
That would not help with the resolution of your local homehassistant but still could help with the NTP servers

carow-it, 2023-02-08:
I am using DHCP and set their respective MAC addresses to different IPs in the DHCP conf.
The Problem occurs if one or both are powered on.
I have a total of 12 devices, and three of those have the same issue.
I already tried ```reset 5``` and ```reset 1```.
Adding ```ipaddress5 8.8.8.8``` with ```restart 1``` didn't help with the external requests.
```lua
00:46:28.122 CMD: webquery http://ntp.org get
00:46:28.124 SRC: WebConsole from 192.168.69.70
00:46:28.127 CMD: Grp 0, Cmd 'WEBQUERY', Idx 1, Len 18, Pld -99, Data 'http://ntp.org get'
00:46:38.501 RSL: RESULT = {"WebQuery":"Connect failed"}
```

barbudor, 2023-02-08:
You first said 1 device out of 2, no you say 3 out of 12
Are they all Nous A1T ?

Could you try to upgrade to latest dev on one of the failing device ? OTA from http://ota.tasmota.com/tasmota/tasmota.bin.gz

carow-it, 2023-02-08:
Sorry, that was an example meant as 1 to 1 comparison.
They are all the same product, Nous A1T.

Would you look at that, the one with dev firmware now works. Even without a direct entry to an external DNS. You would not happen to know what code changed in regard to DNS?
````lua
23:52:36.522 CMD: webquery http://ntp.org get
23:52:37.099 MQT: stat/tasmota_17FB9C/RESULT = {"WebQuery":"Done"}
23:52:58.513 CMD: ipaddress
23:52:58.521 MQT: stat/tasmota_17FB9C/RESULT = {"IPAddress1":"0.0.0.0 (192.168.30.40)","IPAddress2":"192.168.30.254","IPAddress3":"255.255.255.0","IPAddress4":"192.168.30.254","IPAddress5":"0.0.0.0"}
23:54:43.180 CMD: webquery http://homeassistant.lan:8123 get
23:54:43.215 MQT: stat/tasmota_17FB9C/RESULT = {"WebQuery":"Done"}
````

All devices showed up in MQTT but curiously the other two erroneous devices they still can not resolve the home assistant URL and their Sensors are unavailable in Home Assistant. Also, they keep showing the mqt connect failed error message.
````lua
00:00:44.125 CMD: webquery http://homeassistant.lan:8123 get
00:00:54.465 RSL: RESULT = {"WebQuery":"Connect failed"}
00:01:09.061 MQT: Attempting connection...
00:01:10.064 MQT: Connect failed to homeassistant.lan:1883, rc -5. Retry in 30 sec
````

barbudor, 2023-02-08:
There have been some recent changes but I was under the impression that release 12.3.1 was ok
Since, the dev version had few changes on dns but normally only related to IPv6 and some partial reintroduction of mdns for matter protocol

The one that works doesn't need an external dns as your router (IPAddress4 ending .254) act as a dns

For the one that doesn't resolve, as they can't resolve homeassistant.lan they can't connect to HA mqtt

So I can't tell for sure which recent change solve your issue but at least you know now what to do

Still it doesn't explain why some devices work and some don't with the same 12.3.1

Jason2866, 2023-02-09:
It was a fix made regarding DNS name resolution of mqtt (after the release 12.3.1) that not always hits.

closed 2023-02-09 by Jason2866