arendst/Tasmota #17944: THR316 with THS-01 showing null temperature and humidity

gertjaap, 2023-02-14:
### PROBLEM DESCRIPTION

Flashed latest Tasmota32 onto THR316 with an attached THS01 sensor. Configured `GPIO25` as `SI7021`. 

Temperature and humidity readings show `null`

Console keeps showing:

```
DHT: Pin25 timeout waiting for low pulse
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff THR316
- [X] Tasmota binary firmware version number used: 12.3.1 and 12.3.1.6
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: `esptool` first and then OTA 
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

11:55:28.629 CMD: Backlog Template; Module; GPIO 255
11:55:28.632 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
11:55:28.666 SRC: Backlog
11:55:28.668 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
11:55:28.679 MQT: stat/tempsensor01/RESULT = {"NAME":"Sonoff THR316","GPIO":[32,0,0,0,0,0,0,0,0,321,0,576,320,0,0,0,0,224,0,0,0,1,0,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
11:55:28.912 SRC: Backlog
11:55:28.915 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
11:55:28.925 MQT: stat/tempsensor01/RESULT = {"Module":{"0":"Sonoff THR316"}}
11:55:29.163 SRC: Backlog
11:55:29.165 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
11:55:29.183 MQT: stat/tempsensor01/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"321":"Led_i2"},"GPIO14":{"0":"None"},"GPIO15":{"576":"LedLink_i"},"GPIO16":{"320":"Led_i1"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"224":"Relay1"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"1248":"SI7021"},"GPIO26":{"0":"None"},"GPIO27":{"3840":"Output Hi"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
11:55:30.982 DHT: Pin25 timeout waiting for low pulse

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

11:57:24.968 CMD: Status 0
11:57:24.973 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
11:57:24.983 MQT: stat/tempsensor01/STATUS = {"Status":{"Module":0,"DeviceName":"tempsensor01","FriendlyName":["tempsensor01"],"Topic":"tempsensor01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:57:25.027 MQT: stat/tempsensor01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:06:23","StartupUTC":"2023-02-14T10:51:02","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2023-02-14T10:19:41","SaveCount":49}}
11:57:25.072 MQT: stat/tempsensor01/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022-12-16T10:31:03","Core":"2_0_5_3","SDK":"v4.4.3","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"436/699"}}
11:57:25.092 MQT: stat/tempsensor01/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["House of Vision - Media",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00206000","00004000","00000000"]}}
11:57:25.132 MQT: stat/tempsensor01/STATUS4 = {"StatusMEM":{"ProgramSize":1479,"Free":1400,"Heap":132,"StackLowMark":1,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D43C482D","000000A1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
11:57:25.201 MQT: stat/tempsensor01/STATUS5 = {"StatusNET":{"Hostname":"tempsensor01-3480","IPAddress":"192.168.1.202","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"C0:49:EF:F6:AD:98","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:57:25.244 MQT: stat/tempsensor01/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.blocksource.nl","MqttPort":28883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F6AD98","MqttUser":"tempsensor01","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:57:25.268 BRY: GC from 7872 to 3920 bytes, objects freed 20/40 (in 0 ms) - slots from 77/122 to 38/122
11:57:25.276 MQT: stat/tempsensor01/STATUS7 = {"StatusTIM":{"UTC":"2023-02-14T10:57:25","Local":"2023-02-14T11:57:25","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:00","Sunset":"18:08"}}
11:57:25.300 MQT: stat/tempsensor01/STATUS10 = {"StatusSNS":{"Time":"2023-02-14T11:57:25","SI7021":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}}
11:57:25.316 BRY: GC from 4274 to 3689 bytes, objects freed 3/40 (in 0 ms) - slots from 39/122 to 38/122
11:57:25.326 MQT: stat/tempsensor01/STATUS11 = {"StatusSTS":{"Time":"2023-02-14T11:57:25","Uptime":"0T00:06:23","UptimeSec":383,"Heap":132,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Wifi":{"AP":1,"SSId":"House of Vision - Media","BSSId":"7A:AC:B9:B1:D3:2D","Channel":1,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:57:26.985 DHT: Pin25 timeout waiting for low pulse
11:57:28.396 WIF: Checking connection...

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

11:57:30.950 DHT: Pin25 timeout waiting for low pulse
11:57:34.966 DHT: Pin25 timeout waiting for low pulse
11:57:38.988 DHT: Pin25 timeout waiting for low pulse
11:57:42.953 DHT: Pin25 timeout waiting for low pulse
11:57:46.973 DHT: Pin25 timeout waiting for low pulse
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Set GPIO25 to SI7021, attach THS-01 sensor, restart.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Proper temperature values being shown

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-02-14:
Can you open the sensor and check if there is still a SI7021 inside?

gertjaap, 2023-02-14:
[image]

Jason2866, 2023-02-14:
That is not an SI7021. Older Sonoff Sensor with an SI7021
[image]
Now the question is what sensor is it?

barbudor, 2023-02-14:
The problem is the MCU in between the sensor and the protocol it implements (some kind of one wire as the si7021 is natively i2c) 

Whatever is the real sensor chip, the protocol could be using the same as the original Sonoff sensor, or not.

arendst, 2023-02-14:
I have the same THS-01 and it used to work until I openend the unit to find out if it was the same unit. Now it occasionally reports ``16:47:33.704 DHT: Pin25 timeout waiting for high pulse``.

Timing is critical and looking at the source I did some timing tweaking before.
Let me chk.

arendst, 2023-02-14:
BTW you might want to try at a lower CPU frequency with command ``CpuFrequency 80`` and see if that works (at least it solved my latest timeout).

gertjaap, 2023-02-14:
> BTW you might want to try at a lower CPU frequency with command CpuFrequency 80 and see if that works (at least it solved my latest timeout).

Did this, but it had no effect.

```
17:41:03.866 CMD: CpuFrequency 80
17:41:03.868 SRC: WebConsole from <redacted>
17:41:03.871 CMD: Grp 0, Cmd 'CPUFREQUENCY', Idx 1, Len 2, Pld 80, Data '80'
17:41:03.883 MQT: stat/tempsensor01/RESULT = {"CpuFrequency":80}
17:41:04.936 DHT: Pin25 timeout waiting for low pulse
17:41:08.909 DHT: Pin25 timeout waiting for low pulse
17:41:10.376 WIF: Checking connection...
17:41:12.928 DHT: Pin25 timeout waiting for low pulse
17:41:16.902 DHT: Pin25 timeout waiting for low pulse
17:41:20.919 DHT: Pin25 timeout waiting for low pulse
17:41:24.929 DHT: Pin25 timeout waiting for low pulse
17:41:28.944 DHT: Pin25 timeout waiting for low pulse
```

arendst, 2023-02-14:
Make sure the connector is inserted firmly. Try to measure resistance of the wires end to end for correct connection. Chk colors of wires in connector: red,white, empty,black

arendst, 2023-02-15:
Pls try latest driver (available as pre-compiled dev binary in an hour).

It adds more debugging and command ``DhtDelay<sensor> <high_delay>,<low_delay>`` to allow user control over high and low delay in microseconds.

My THS-01 now runs fine on CPUFrequency 160MHz with non-persistent command ``DhtDelay 500,40``.

gertjaap, 2023-02-16:
Thanks for the update! I have installed it and issued the DhtDelay command. Here's the debug output:

```
10:15:16.369 CMD: DhtDelay 500,40
10:15:16.370 SRC: WebConsole from 10.5.25.65
10:15:16.373 CMD: Grp 0, Cmd 'DHTDELAY', Idx 1, Len 6, Pld 500, Data '500,40'
10:15:16.385 MQT: stat/tempsensor01/RESULT = {"DhtDelay1":[500,40]}
10:15:17.807 DHT: Pin25 cycles (0/80)  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ..
10:15:17.809 DHT: Pin25 timeout waiting for pulse 0
10:15:19.132 WIF: Checking connection...
10:15:21.806 DHT: Pin25 cycles (0/80)  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ..
10:15:21.808 DHT: Pin25 timeout waiting for pulse 0
10:15:25.767 DHT: Pin25 cycles (0/80)  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ..
10:15:25.769 DHT: Pin25 timeout waiting for pulse 0
10:15:29.811 DHT: Pin25 cycles (0/80)  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ..
10:15:29.813 DHT: Pin25 timeout waiting for pulse 0
10:15:33.815 DHT: Pin25 cycles (0/80)  0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ..
10:15:33.817 DHT: Pin25 timeout waiting for pulse 0
```

As for your earlier question about the wires in the connector, they match up with what you said:

[image]

Pressed the connector firmly, but to no avail it seems. I got a multi meter coming in later today then I can measure the wire resistance for you.

sfromis, 2023-02-16:
The opportunity you get with the new `DhtDelay` command is to test if it works when tweaking the timing values up or down.

gertjaap, 2023-02-17:
Used the multimeter to measure connection between the VCC / DATA / GND pins on the THS-01 PCB and the THR316 PCB and they seem just fine (resistance drops to 0 when i push the multimeter on the connecting pins).

I tried DhtDelay up to `10000,10000` and it's still timing out. I don't know what I can try next.

Do I need some kind of logic analyzer to snoop on what's going over the data wire to figure this out?

I have an old DHT-11 sensor lying around, I could try connecting that to the RJ11 and see what happens?

sfromis, 2023-02-17:
Extreme values like 10000 is a pretty bad starting point in case of trying to make tweaks, where you have a much better starting point in what arendst used.

gertjaap, 2023-02-20:
Okay, I have tried a lot of different settings close to the initial settings of arendst. Tested on CpuFrequency 80 and 160, High cycle timeout 300-700 in steps of 50 and low cycle timeout 20-70 in steps of 10 - and every possible combination therein.

I made a small script to cycle through all the options to make sure I tested them all properly and captured the log for each. 

```bash
#!/bin/bash

CPUFREQ=(80 160)
DHTDELAYHIGH=(300 350 400 450 500 550 600 650 700 750)
DHTDELAYLOW=(20 30 40 50 60 70)

for freq in ${CPUFREQ[@]}; do
  echo "CpuFrequency $freq"
  curl "http://192.168.1.202/cm?cmnd=CpuFrequency%20$freq"
  sleep 1
  for high in ${DHTDELAYHIGH[@]}; do
    for low in ${DHTDELAYLOW[@]}; do
      echo "CpuFrequency $freq - DhtDelay $high,$low" | tee -a "DhtDelay.log"
      curl "http://192.168.1.202/cm?cmnd=DhtDelay%20$high%2C$low"
      sleep 12
      curl -s http://192.168.1.202/cs?c2=0 | tail -n 30 | grep -A 900 "DhtDelay1\":\[$high,$low" | tee -a "DhtDelay.log"
    done
  done
done
```

[DhtDelay.log.txt](https://github.com/arendst/Tasmota/files/10783107/DhtDelay.log.txt)

The fact that the cycles printout remains all zeroes on every setting makes me think that this sensor uses a different protocol all together? 

I ordered a very basic logic analyzer to try and make sense of what's going on. I have a couple more THR316s with the original firmware on it, so I'll try to capture what it is doing in the stock firmware when talking to the sensor.

arendst, 2023-02-20:
Let me chk your script against mine......

arendst, 2023-02-20:
I wonder if you need a logic analyzer. Something is wrong hardware wise.

What you could do is configure GPIO25 as a ``Relay3`` and measure the voltage on the input pin of the THS-01 when you toggle the new button in the GUI. 

First measure the voltage on two of the four flash connector pins you'll see in your picture above on the left called J2. The top hole is Gnd. The bottom hole is 3V3.

Then find a way to measure the signal on the white wire while toggling relay3.

gertjaap, 2023-02-20:
Okay, I measured using a multimeter between these points on the PCB (bottom view):

[image]

| + | - | Voltage (relay off) | Voltage (relay on)   |
|---|---|---------------------------|-----------------------------|
| 1 | 2 | 0.236 | 0.886 |
| 1 | 3 | 0.236 | 0.568 |
| 1 | 4 | 0.000 | -0.941 |
| 1 | 5 | 0.236 | 0.884 |
| 2 | 3 | 0.000 | -0.314 | 
| 2 | 4 | -0.237 | -1.826 |
| 2 | 5 | 0.000 | 0.000 |
| 3 | 4 | -0.237  | -1.509 |
| 3 | 5 | 0.000 | 0.314 |
| 4 | 5 | 0.237 | 1.822 |

arendst, 2023-02-20:
So obviously you're missing the power signal.

Let's chk our templates. This is mine:
```
{"NAME":"Sonoff THR316D","GPIO":[32,0,0,0,225,9280,0,0,0,321,0,576,320,9184,9216,0,0,224,0,9248,0,1,0,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```
Chk your GPIO27 in configure template menu option. It should be ``Output Hi``.

gertjaap, 2023-02-20:
I have plugged the sensor into a THR316 running on stock firmware, it shows proper 3v3 between 1 and 2.

I will check the template

gertjaap, 2023-02-20:
Template (from [here](https://templates.blakadder.com/sonoff_THR316.html) ) :
```
{"NAME":"Sonoff THR316","GPIO":[32,0,0,0,0,0,0,0,0,321,0,576,320,0,0,0,0,224,0,0,0,1,0,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

GPIO Template settings:

[image]

arendst, 2023-02-20:
Ah. You're having a THR316 without the D(isplay). In that case the template looks fine as I expect they didn't change GPIO usage between THR316 and THR316D.

gertjaap, 2023-02-20:
When I apply your template, the voltage on the THS-01 also isn't 3V3. I'm going to try and flash the unit I confirmed 3V3 on the sensor with Tasmota to see if perhaps it's something faulty in the THR316 unit I have flashed with Tasmota?

arendst, 2023-02-20:
OK. But make a backup of the current firmware first so you can go back if neded.

gertjaap, 2023-02-20:
Made backup, flashed the other THR316. Now works with the latest dev firmware without any tweaks to the DhtDelay or CpuFreq. 

I will flash the other THR316s i have. Looks like the one unit I was testing with is defect then?

arendst, 2023-02-20:
I guess so. Do you hear the relay when you push the button on the device?

cju-mt, 2023-02-21:
I have a TH16 with a sensor that has a 4C TRRS Plug. I have not popped it open. I am wondering if it is a good idea to adapt it for a test to see if this clears up the timeout issue? Thoughts? is it worth trying or are we close to a resolution? Thx

arendst, 2023-02-21:
The latest fixes (dhtdelay) IS the solution. From three years experience we now know there is no final timing solution for these crappy sensors. The latest addition allows all users to tweak their sensors to make them functional.

Jason2866, 2023-02-21:
Closing since the issue is solved.

closed 2023-02-21 by Jason2866

cju-mt, 2023-02-21:
That fixed my issue. Thank you very much!!!!!!!!!

cju-mt, 2023-02-26:
Is there a way to make dhtdelay persistent? After restarting HA several times while working on an unrelated sensor I realized that my thermostat stopped working. After some poking around I discovered that dhtdelay had reverted back to the original setting of 400,30. I had it set at 500,40 which was the only setting that would work.
Thanks

barbudor, 2023-02-26:
You can create a boot rule to set it back at each restart:
```
Rule1 on system#init do DhtDelay 500,40 endon
Rule1 1
```

sfromis, 2023-02-26:
Instead of a rule, it would be a bit simpler to include such commands in a file `autoexec.bat`

cju-mt, 2023-02-26:
Wow, that is exactly what I needed @barbudor . Could not have been easier. Working better then ever. I really appreciate the help. Hopefully this will help somebody else that is a newbie like me.

cods69, 2023-04-18:
FWIW, just found this bug after having the same issue, on original versions (from years ago) 2xTH10 and 1xTH16, all 3 with SI7021 sensors.  I'm getting both random NULL and random 100% Humidity along with roughly half temp, in 12.4.0.
After thinking this was just a FW issue, upgraded one TH10 to 12.5.0 and still has the same issue.  Hence I'm here now.
Is there a short fix to this issue? All the ideas above are a tad confusing without my morning coffee. :)
[image]

sfromis, 2023-04-19:
That bash script is just an example to try out various `DhtDelay` values to determine what fails (or not). You can also do that other ways, what counts is simply what values work. When you have decided what `DhtDelay` values works for you, you can have that command issued after each boot, simplest is to put it in a `autoexec.bat` file.

Defaults are `DhtDelay 400,30`; I saw better results with `DhtDelay 500,50`, but that may not be "especially good" values.

cods69, 2023-04-19:
Thanks for the quick reply - I got the DhtDelay script working ok the output seems indeterminate and sporadic.  i.e. not always getting temp values (bad or good) back.
As for the autoexec.bat, I'm using OpenHab on a pi, so I'd be able to do some init.d stuff I assume, when I get a value to work consistently.

sfromis, 2023-04-19:
`init.d` would be the totally wrong place, as `autoexec.bat` is a file in the Tasmota file system, used when Tasmota boots, unrelated to whatever happens on your Linux box.

And that bash script is not for getting a temperature, it is only for experimenting with `DhtDelay` values, nothing more. Thus, it only works properly when you are **not** getting "consistent" results.

cods69, 2023-04-19:
Ahh got it.  Thought it was strange.
I'll roll back the devices until I have time to play with them as it's triggering all sorts of reactions in OH.  
I appreciate the help.

plbartie, 2023-04-30:
> FWIW, just found this bug after having the same issue, on original versions (from years ago) 2xTH10 and 1xTH16, all 3 with SI7021 sensors. I'm getting both random NULL and random 100% Humidity along with roughly half temp, in 12.4.0. After thinking this was just a FW issue, upgraded one TH10 to 12.5.0 and still has the same issue. Hence I'm here now. Is there a short fix to this issue? All the ideas above are a tad confusing without my morning coffee. :) [image]
> 
> Is the script in the link below what I'm supposed to run from openhab, to find parameters? [#18438 (comment)](https://github.com/arendst/Tasmota/discussions/18438#discussioncomment-5639305)

Did you manage to solve it. My devices was working fine on old v 11 till i updated to latest 12.5.0. Randomly 2 sensors i have will go to 100% and null.

sfromis, 2023-04-30:
There is no script which you are "supposed to run", that is merely a helper script to to zone in on what `DhtDelay` values may work for you.

cods69, 2023-05-01:
> Did you manage to solve it. My devices was working fine on old v 11 till i updated to latest 12.5.0. Randomly 2 sensors i have will go to 100% and null.

No sorry.  I'm putting this on the back-burner as 12.3.1 works perfectly and never has these issues.  I suggest the same for you unless you want to delve in to dark arts and have the time.  It's an easy downgrade.

cods69, 2023-07-07:
> Did you manage to solve it. My devices was working fine on old v 11 till i updated to latest 12.5.0. Randomly 2 sensors i have will go to 100% and null.

FYI, V13.0.0 on one of my TH10's no longer has the issue.  I can only assume the bug was fixed at some point. 
Thanks devs.

mightymos, 2024-02-21:
Has anyone tried to dump the firmware from the Onbright microcontroller (8R08A1)?

I have written a tool that uses Arduino to read/write to these chips:
https://github.com/mightymos/OnbrightFlasher

It works for me and others report it working but it can be a little finicky.
The chip is likely protected from reads but it might be worth trying.
If anyone does want to try DO NOT ISSUE ANY ERASE OR SETFUSE commands because you will lose your stock firmware.

I will also need to expand the sketch to make reading the entire fuse and flash range easier so feel free to open an issue on that project page.

jlazkano, 2024-05-25:
Hello,

I flash with the last firmware version, but the wireless signal is not stable, do you recommend other version?

Regards.