arendst/Tasmota #17967: Switch not working after Upgrate from Percy to Peter

smartboart, 2023-02-17:
Hello,
after updating all my devices from Percy to Peter, All my configurated Switches didnt work anymore.
After downgrade from Peter to Percy it worked again..

arendst, 2023-02-17:
I suggest you fill in the mandatory fields requested when you openend this issue.

Currently it's a big guess what switches you have.

smartboart, 2023-02-17:
Sorry, i thought this is clear....I hjave about 20 devices.. Wemo D1 Mini, ESP32 Olimex and NodeMCU. In all devices i have standart exgternal switches / configurated as switch and motion sensors also conifgurated as switch. All these gpios which are configurated as switch are not working. All gpios configurated as buttons are still working. Even I²C Sensors and rules are still working...

Jason2866, 2023-02-17:
Please post one exact setup which is not working. Use the template, you have deleted, and provide ALL info asked there. Without providing this data we can't help.

smartboart, 2023-02-18:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is.
After the upgrade from Percy to Peter, all devices will not read gpios with connected switch. All action on the hardware switch is not recognized. It makes no difference if the switch is detached fromrelay or linked to a relay like it is in standard configuration.
Even if it is a wemo D1 mini, a Node MCU or a ESP32 Olimex POE Modul. All my devices have the same issue after upgrade to Peter. Here is a example of a Wemo Config which is linked to Relay...
[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): __Wemo D1 Mini - Node Mcu - Olimex ESP32 POE___
- [ x] Tasmota binary firmware version number used: _____12.3.1 to 12.4.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: _____OTA
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:06:09.007 CMD: Backlog Template; Module; GPIO 255
11:06:09.094 MQT: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:06:09.294 MQT: stat/sonoff/RESULT = {"Module":{"18":"Generic"}}
11:06:09.500 MQT: stat/sonoff/RESULT = {"GPIO0":{"640":"I2C SDA"},"GPIO1":{"0":"None"},"GPIO2":{"608":"I2C SCL"},"GPIO3":{"0":"None"},"GPIO4":{"257":"Relay_i2"},"GPIO5":{"256":"Relay_i1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"97":"Button_i2"},"GPIO13":{"0":"None"},"GPIO14":{"162":"Switch3"},"GPIO15":{"0":"None"},"GPIO16":{"226":"Relay3"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

11:08:22.370 CMD: rule1
11:08:22.378 MQT: stat/sonoff/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":142,"Free":369,"Rules":"on switch8#state=1 do publish stat/sonoff/RESULT {\"POWER8\":\"ON\"} endon on switch8#state=0 do publish stat/sonoff/RESULT {\"POWER8\":\"OFF\"} endon"}}

11:08:52.282 CMD: rule2
11:08:52.290 MQT: stat/sonoff/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":149,"Free":362,"Rules":"ON button2#state=10 DO power1 2 ENDON ON button2#state=11 DO power2 2 ENDON ON button2#state=3 DO power2 2 ENDON ON button2#state=3 DO power1 2 ENDON"}}

11:09:16.651 CMD: rule3
11:09:16.660 MQT: stat/sonoff/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":75,"Free":436,"Rules":"ON button2#state=10 DO power2 2 ENDON ON button2#state=11 DO power1 2 ENDON"}}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:29:30.363 CMD: status0
11:29:30.365 SRC: WebConsole from 192.168.1.75
11:29:30.368 CMD: Grp 0, Cmd 'STATUS', Idx 0, Len 0, Pld -99, Data ''
11:29:30.412 MQT: stat/sonoff/STATUS0 = {"Status":{"Module":18,"DeviceName":"Wemo-Ankleidezimmer","FriendlyName":["Rolladen_hoch","Rolladen_runter","Licht_Ankleidezimmer"],"Topic":"sonoff","ButtonTopic":"","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,1,1,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.42:8000/Tasmota/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:10","StartupUTC":"2023-02-18T10:22:20","Sleep":0,"CfgHolder":4617,"BootCount":83,"BCResetTime":"2020-03-24T14:36:31","SaveCount":5891,"SaveAddress":"F9000"},"StatusFWR":{"Version":"12.4.0(sensors)","BuildDateTime":"2023-02-16T12:12:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"485/699"},"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"192.168.1.39","LogPort":514,"SSId":["michaswlan",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000802B","0A05C80000000600003C5AFF002800000000","00808080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":698,"Free":304,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0000C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103"},"StatusNET":{"Hostname":"Wemo-Ankleidezimmer","IPAddress":"192.168.1.71","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:E3:EA:D7","Webserver" [... cut ...]
_Steps to reproduce the behavior:_
use external switch ...always no reaction

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
When i use my external Switch which is connected to gpio 14 D5 conigured as switch 3 it should switch on relay 3 in this example..

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-02-18:
Thx for the better description!

Originally the switch (and button) numbering was supposed to be sequential. So starting at 1 and up. Until v12.4 this was apparently not enforced. It is now to cater for virtual switches (and buttons).

I'll need to revisit my code changes leading to this as you seem to have a valid solution up until v12.4

Investigating.

barbudor, 2023-02-18:
Supporting scattered values would be a problem when adding port extenders for example that are expecting 8 or more switches to be available continuously like what was recently done on tm1638

The motto on support has always been : start at 1 and increment by 1.
This was a loophole

arendst, 2023-02-18:
This would indeed be the preffered and supported way of button and switch configuration:

[image]

In this case GPIO13 can still be unconnected while allowing the switch to control relay2.

barbudor, 2023-02-18:
Even is the switch on gpio14 was set to switch 1, it could still be redirected to relay 3 with a rule

sfromis, 2023-02-18:
Does this mean that out-of-sequence Switch inputs works differently now? Someone is having trouble with this device, where the template has Switch1 and Switch3. While the user for some reason now has a gpio config with Switch2 and Switch3, both setups would be hit by issues here.
https://templates.blakadder.com/milfra_MFA06.html

arendst, 2023-02-18:
What's the use for ``Switch3`` without havong a relay3?

smartboart, 2023-02-18:
I also have configurations with switch numbers wothout cooespodent Relais.. for example switch 3 is a motion sensor...for these Things i use rules to set a nun configurated  Relais for example 10 to get a Real time mqtt Information about the switch state..so in the mqtt Broker i trigger with motion information of Relais 10...

arendst, 2023-02-18:
I suggest you try latest dev release and report back if it solves your issues.

If so, I'll release v12.4.1 ....

smartboart, 2023-02-18:
Hello ..latest dev fixed it...i have tried with the example one from here in this issue.. Thanks...

arendst, 2023-02-18:
OK. I'll add the change to v12.4.0 and keep it v12.4.0

helmar74, 2023-02-18:
Hi just a question from my side.
I updated ~60 Devices to 2.4.0 which was released on 16.02.2023. Should I recompile and  update again?

sfromis, 2023-02-18:
Are you affected by this issue? If yes, better update.....
If using precompiled builds, better first make sure that the files have been pushed to the servers.

Jason2866, 2023-02-18:
> Hi just a question from my side.
> I updated ~60 Devices to 2.4.0 which was released on 16.02.2023. Should I recompile and update again?

If you have the switch and Relais counting done as recommended (starting from one and continous) there is no need for.

Jason2866, 2023-02-18:
Closing since OP gave feedback it is fixed

closed 2023-02-18 by Jason2866