arendst/Tasmota #17984: Version 12 does not request time from the routers IP address anymore

JsBergbau, 2023-02-18:
### PROBLEM DESCRIPTION
Tasmota 12.3.1 does not try to sync time with the IP-Adress oft the router as version 8 did. From the FAQ https://github.com/tasmota/docs-8.1/blob/master/help/FAQ.md

> If the Tasmota device receives its IP address via DHCP from the router, Tasmota will request its time sync from the router's time server. This is managed by the Arduino core, not Tasmota (https://github.com/arendst/Tasmota/issues/5283#issuecomment-466888846). Therefore, if the NTP server on the router is not configured, or configured improperly, the time on the Tasmota device could be wrong. If the router does not have a time server, this is not the problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ x ] Tasmota binary firmware version number used: 12.3.1
  - [ x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just install version 12, put it into a network with Fritzbox and time won't be set

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota gets the time from the Fritzbox NTP Server

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Even if routers would provide wrong time, this would be corrected after connecting the preconfigured pool.ntp.org NTP servers. However in networks where internet connection is blocked for Tasmotas they will set the time from router.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-02-19:
You should not refer to vastly outdated docs from version 8.1, but instead current docs. https://tasmota.github.io/docs/

The feature for using NTP servers announced via DHCP gave so much trouble with various routers not doing it right, that the feature was deprecated and removed from Tasmota. You need to specifically config NTP servers in Tasmota via the `NtpServer` command, that includes if/when the router is providing a NTP server.

As things are now, Tasmota is working as expected, and current docs does not suggest anything about DHCP-announced NTP servers being supported.

JsBergbau, 2023-02-19:
I think we're talking about different things here. There is an option (042) https://serverfault.com/questions/7565/dhcp-options-for-network-time-server for the DHCP server to specify an NTP-Server. Fritzbox does not use Option 42 to give clients a DHCP server. 

With version 8 tasmota just tried to pull time from router IP, so see if there is a listening NTP server and if so use that time. Thats different from using the NTP servers DHCP offers. Changing this from my point of view very useful behaviour leads to unforseen troubles because tasmota devices have no time anymore after the update.

Jason2866, 2023-02-19:
This feature was removed, since many routers are not able to do it correctly.
As designed. To override the default servers the have to be set manually.

closed 2023-02-19 by Jason2866

arendst, 2023-02-19:
I just tried using my router as NTP address and it just fails (as my router does NOT supply NTP.

If you like you can change line 1258 in file support_wifi.ino to
```
//  snprintf_P(fallback_ntp_server, sizeof(fallback_ntp_server), PSTR("%d.pool.ntp.org"), random(0,3));
  ext_snprintf_P(fallback_ntp_server, sizeof(fallback_ntp_server), PSTR("%_I"), Settings->ipv4_address[1]);
```
and chk if yours does work. Make sure to delete the current defined NtpServers using a command like ``NtpServer1 0``.

Considering the response from my router and past knowledge I suggest you set your NtpServer1 to your routers IP address and enjoy it's functionality. The above change won't make it into future releases.