arendst/Tasmota #17985: E-Paper Display not working anymore in Tasmota 12.4.0

helmar74, 2023-02-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I compiled Tasmota by myself with this options defined in user_config_override.h:
```
#undef USE_DISPLAY
#define USE_DISPLAY                            // Add I2C Display Support (+2k code)

#undef USE_SPI
#define USE_SPI                                  // Hardware SPI using GPIO12(MISO), GPIO13(MOSI) and GPIO14(CLK) in addition to two user selectable GPIOs(CS and DC)

#undef USE_DISPLAY

#define USE_DISPLAY                            // Add SPI Display support for 320x240 and 480x320 TFT
#undef USE_DISPLAY_ILI9341                  // [DisplayModel 4] Enable ILI9341 Tft 480x320 display (+19k code)

#undef USE_DISPLAY_EPAPER_29
#define USE_DISPLAY_EPAPER_29                // [DisplayModel 5] Enable e-paper 2.9 inch display (+19k code)
```

Until Tasmota 12.3.1 the 2,9 inch E-Paper Display where working fine. 
 Here'S the example configuration of one Device:
[image]

Now the Display isn't working anymore. It doesn't show any content, I'm not able to clear display and send out some text

```
57:49.576 SRC: WebConsole from 192.168.1.31
08:57:49.578 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 50, Pld -99, Data 'displaytext [z]; displaytext test; displaytext [d]'
08:57:49.641 SRC: Backlog
08:57:49.643 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 3, Pld -99, Data '[z]'
08:57:49.855 MQT: stat/sonoffWemosMini_2/RESULT = {"DisplayText":"[z]"}
08:57:49.861 SRC: Backlog
08:57:49.862 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 4, Pld -99, Data 'test'
08:57:50.049 MQT: stat/sonoffWemosMini_2/RESULT = {"DisplayText":"test"}
08:57:50.261 SRC: Backlog
08:57:50.263 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 3, Pld -99, Data '[d]'
08:57:50.631 MQT: stat/sonoffWemosMini_2/RESULT = {"DisplayText":"[d]"}
08:57:51.450 DHT: Pin4 cycles (80/80)  138 66 127 66 138 66 138 66 137 66 138 66 138 66 138 186 173 189 137 66 ..
08:57:51.451 DHT: Pin4 read 01A800B962
08:57:55.449 DHT: Pin4 cycles (80/80)  138 66 127 66 138 66 138 66 137 66 138 66 138 66 138 186 173 189 137 66 ..
08:57:55.450 DHT: Pin4 read 01A900BA64
08:57:57.842 WIF: Prüfe Verbindung...
08:57:59.450 DHT: Pin4 cycles (80/80)  138 66 127 66 138 66 138 66 137 66 138 66 138 66 138 186 173 189 137 66 ..
08:57:59.451 DHT: Pin4 read 01A800B962
```
The Displaytext commands are recognized, but the display simply doesn't show anything.
If I revert back to 12.3.1 the display is working again.

Has something changed regarding E-Paper usage from 12.3.1 to 12.4.0 ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos Mini D withe 2,9 inch E-Paper Display
- [X] Tasmota binary firmware version number used: 12.4.0 (released on 18th february)
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used:PlatformIO
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

09:00:54.152 CMD: Backlog Template; Module; GPIO 255
09:00:54.217 MQT: stat/sonoffWemosMini_2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,4704],"FLAG":0,"BASE":18}
09:00:54.552 MQT: stat/sonoffWemosMini_2/RESULT = {"Module":{"18":"Generic"}}
[... 42 lines cut ...]
09:02:11.411 DHT: Pin4 read 01A700B961
09:02:13.606 CMD: displaytext [d]
09:02:13.607 SRC: WebConsole from 192.168.1.31
09:02:13.609 CMD: Grp 0, Cmd 'DISPLAYTEXT', Idx 1, Len 3, Pld -99, Data '[d]'
09:02:13.977 MQT: stat/sonoffWemosMini_2/RESULT = {"DisplayText":"[d]"}
09:02:15.453 DHT: Pin4 cycles (80/80)  138 66 127 66 138 66 138 66 137 66 138 66 138 66 138 186 173 189 137 66 ..
09:02:15.454 DHT: Pin4 read 01A700B961

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upgrade from 12.3.1  to 12.4.0 on a wemos Mini D with connected 2,9 inch E-Paper Display

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Display should work

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2023-02-19:
the driver now supports per default Version 2 of the display.
to compile for V1  comment out
#define EPD_29_V2
in file epd2in19.cpp

or use universal display driver with v1 descriptor

helmar74, 2023-02-19:
Thanl you, how can I :
or use universal display driver with v1 descriptor
?

gemu2015, 2023-02-19:
if you have a file system (devices 4M) you may check in folder "displaydesc" there is a readme.
and compile with USE_UINIVERSAL_DISPLAY

with 1M devices it is easiest to comment out like described above.

helmar74, 2023-02-19:
OK, I'll try it.
Are DisplayText commands working with univesal driver as before?
... and do I have to change configuration of device?
Current config:
[image]

gemu2015, 2023-02-19:
display text is the same, but you need to connect RST and BUSY too, so you need 2 more pins.
and configure
RST to Display RST
BUSY to SSPI_MISO

helmar74, 2023-02-19:
OK, understood.

Is it possible to undefine EPD_29_V2 in user_config_override.h ?

gemu2015, 2023-02-19:
currently not. i will change this with a pr

helmar74, 2023-02-19:
> 

That would be great. Thank you!

gemu2015, 2023-02-19:
pr is done
configure:
#define EPD_29_V1

helmar74, 2023-02-19:
Great thank you, when will it be in current development?
I also tried to comment out 
```
#define EPD_29_V2
in file epd2in19.cpp
```

But this didn't worked...

helmar74, 2023-02-19:
I tried this, but it doesn'T work:
[image]

gemu2015, 2023-02-19:
ok, i have 2 of these old ones but use universal driver. i will compile v1 and test whats going on.

helmar74, 2023-02-19:
Just tried new development version with:
[image]

This doesn't work for me.

```
2:27:20.875 CMD: status 0
12:27:20.880 MQT: stat/sonoffWemosMini_2/STATUS = {"Status":{"Module":18,"DeviceName":"WemosMini-2","FriendlyName":["WemosMini-2"],"Topic":"sonoffWemosMini_2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:27:20.915 MQT: stat/sonoffWemosMini_2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffStandardGruppe","OtaUrl":"http://192.168.1.2:5555/data/firmwares/tasmota.bin","RestartReason":"External System","Uptime":"0T00:01:33","StartupUTC":"2023-02-19T11:25:47","Sleep":200,"CfgHolder":4617,"BootCount":273,"BCResetTime":"2020-03-21T13:09:36","SaveCount":638,"SaveAddress":"F9000"}}
12:27:20.952 MQT: stat/sonoffWemosMini_2/STATUS2 = {"StatusFWR":{"Version":"12.4.0.1(tasmota)","BuildDateTime":"2023.02.19 11:37:19","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"418/699"}}
12:27:20.975 MQT: stat/sonoffWemosMini_2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hsnetioteg24ghz",""],"TelePeriod":900,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00002800000000","00000080","00006000","00004000","00000000"]}}
12:27:21.002 MQT: stat/sonoffWemosMini_2/STATUS4 = {"StatusMEM":{"ProgramSize":611,"Free":392,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8FA0C58F","040489A1","020814C7","002033C0","80007081","00004020","00000000","54000020","00000080"],"Drivers":"1,2,3,4,6,8,9,10,12,13,16,24,26,35,62","Sensors":"1,3,4,5,6,9,10,16,22,35,41"}}
12:27:21.035 MQT: stat/sonoffWemosMini_2/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-2","IPAddress":"192.168.1.124","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:3B:79:93","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:27:21.063 MQT: stat/sonoffWemosMini_2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"WemosMini-2","MqttClient":"WemosMini-2","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":15}}
12:27:21.086 MQT: stat/sonoffWemosMini_2/STATUS7 = {"StatusTIM":{"UTC":"2023-02-19T11:27:21","Local":"2023-02-19T12:27:21","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"07:09","Sunset":"17:35"}}
12:27:21.107 MQT: stat/sonoffWemosMini_2/STATUS10 = {"StatusSNS":{"Time":"2023-02-19T12:27:21","AM2301":{"Temperature":18.5,"Humidity":42.3,"DewPoint":5.4},"TempUnit":"C"}}
12:27:21.122 MQT: stat/sonoffWemosMini_2/STATUS11 = {"StatusSTS":{"Time":"2023-02-19T12:27:21","Uptime":"0T00:01:34","UptimeSec":94,"Vcc":2.983,"Heap":15,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":7,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"hsnetioteg24ghz","BSSId":"F0:B0:14:6F:B5:4F","Channel":1,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

gemu2015, 2023-02-19:
this works on my old v1 epaper, please confirm

[Epaper_29-gemu-1.0.zip](https://github.com/arendst/Tasmota/files/10777199/Epaper_29-gemu-1.0.zip)

helmar74, 2023-02-19:
Yeah that worked. Thank you for the quick fix.

I used this settings in user_config_override.h
`
// 19.02.2023 Version 12.4.0 https://github.com/arendst/Tasmota/issues/17985
// Für mein Epaper muss der Displaydriver V1 verwendet werden daher
// V2 Treiber auskommentieren
// Wird erst in den nächsten Versionen "scharf geschaltet"
#undef EPD_29_V1
#define EPD_29_V1
#undef USE_DISPLAY_EPAPER_29
#define USE_DISPLAY_EPAPER_29                // [DisplayModel 5] Enable e-paper 2.9 inch display (+19k code)
`

and replaced the 3 files which you provided in the zip above!

PR #17987 (2023-02-19): make epd29  v1 configurable
## Description:

**Related issue (if applicable):** fixes #17985

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

 [... cut]
PR #17999 (2023-02-19): fix epaper 29 non-universal display
## Description:

**Related issue (if applicable):** fixes #17985

now really tested with both versions

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIB [... cut]
closed 2023-02-19 by helmar74