arendst/Tasmota #17989: Shelly 1PM don't show Voltage and Frequency

GerdRuetten, 2023-02-19:
### PROBLEM DESCRIPTION
I flashed a Shelly 1PM with Tasmota 12.4.0.
On the main page, the values for voltage and frequency not show (see screenshots above)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Shelly 1PM
- [ x] Tasmota binary firmware version number used: 12.4.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: Tasmotizer
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:56:50.258 CMD: Backlog Template; Module; GPIO 255
12:56:50.285 MQT: stat/tasmota_8659D4/RESULT = {"NAME":"Shelly 1PM","GPIO":[320,0,0,0,192,2720,0,0,0,0,0,224,0,4736],"FLAG":0,"BASE":18}
12:56:50.508 MQT: stat/tasmota_8659D4/RESULT = {"Module":{"0":"Shelly 1PM"}}
12:56:50.763 MQT: stat/tasmota_8659D4/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"192":"Switch_n1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`: no rules used on this device
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:57:47.934 CMD: Status 0
12:57:47.942 MQT: stat/tasmota_8659D4/STATUS = {"Status":{"Module":0,"DeviceName":"Schalter Klimaanlage Garage","FriendlyName":["Schalter Klimaanlage Garage"],"Topic":"tasmota_8659D4","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:57:47.950 MQT: stat/tasmota_8659D4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Power On","Uptime":"4T04:09:28","StartupUTC":"2023-02-15T07:48:19","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2023-01-22T00:31:58","SaveCount":78,"SaveAddress":"FB000"}}
12:57:47.957 MQT: stat/tasmota_8659D4/STATUS2 = {"StatusFWR":{"Version":"12.3.1(tasmota)","BuildDateTime":"2022.12.16 10:33:11","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"433/699"}}
12:57:47.964 MQT: stat/tasmota_8659D4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 7590 RU",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00208009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
12:57:47.977 MQT: stat/tasmota_8659D4/STATUS4 = {"StatusMEM":{"ProgramSize":630,"Free":372,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15605E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
12:57:47.985 MQT: stat/tasmota_8659D4/STATUS5 = {"StatusNET":{"Hostname":"tasmota-8659D4-6612","IPAddress":"192.168.178.157","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.107","DNSServer2":"0.0.0.0","Mac":"E8:9F:6D:86:59:D4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:57:47.992 MQT: stat/tasmota_8659D4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8659D4","MqttUser":"mqtt_user","MqttCount":61,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:57:48.001 MQT: stat/tasmota_8659D4/STATUS7 = {"StatusTIM":{"UTC":"2023-02-19T11:57:47","Local":"2023-02-19T12:57:47","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:51","Sunset":"18:16"}}
12:57:48.007 MQT: stat/tasmota_8659D4/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:57:48.019 MQT: stat/tasmota_8659D4/STATUS10 = {"StatusSNS":{"Time":"2023-02-19T12:57:48","Switch1":"OFF","ANALOG":{"Temperature":40.7},"ENERGY":{"TotalStartTime":"2023-01-22T00:31:58","Total":5.999,"Yesterday":0.113,"Today":0.061,"Power":5},"TempUnit":"C"}}
12:57:48.028 MQT: stat/tasmota_8659D4/STATUS11 = {"StatusSTS":{"Time":"2023-02-19T12:57:48","Uptime":"4T04:09:29","UptimeSec":360569,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":61,"POWER":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 7590 RU","BSSId":"3C:A6:2F:ED:F1:C5","Channel":6,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":14,"Downtime":"0T00:01:48"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Show the values for voltage and frequency and publish this over MQTT

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
Calibrating as described in the manual are done

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-02-19:
Interresting find. I just pulled out my ancient Shelly 1PM out of the devices box and observed the same on v12.1.1.5!!!

[image]

I'll have a look what's going on here....

arendst, 2023-02-19:
Ah I just remembered. The Shelly 1PM was never able to show voltage/current as it lacks the connection to the energy monitoring chip.

Some users implemented a hack to make it work.

So the Shelly 1PM lacking voltage/current is as designed.

GerdRuetten, 2023-02-19:
Dear Theo, 

thank you for this support on sunday!
Respect

Best regards
Gerd

closed 2023-02-19 by GerdRuetten

sfromis, 2023-02-19:
The Shelly Plus 1PM does have voltage/current/frequency and power factor.

arendst, 2023-02-19:
Correct. The Shelly 1PM doesn't.

Jason2866, 2023-02-19:
The hack https://github.com/arendst/Tasmota/issues/5716#issuecomment-1264361612