arendst/Tasmota #18023: Tasmota (Sensors) stopped working BMP280 in v12.4.0

RPocas, 2023-02-22:
### PROBLEM DESCRIPTION
_On I2C BMP280 D2-GPIO4 SDA and D1-GPIO5 SCL stopped working after upgrade from v12.3.1 to v12.4.0, in rollback to previos version just works fine again._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NomeMCU ESP8266 v3
- [X] Tasmota binary firmware version number used: 12.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA site (Sensors version)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-02-22:
Not filling up the Issue template won't help us to help you
You most likely upgraded to `tasmota` instead of `tasmota-sensors`

sfromis, 2023-02-22:
I've got no issues with BME280 sensors after upgrading to 12.4.0

Jason2866, 2023-02-22:
Closing since issue Template not filled and issue can not be reproduced

closed 2023-02-22 by Jason2866