arendst/Tasmota #18024: Loss of energy total data by updating to 12.4.0

wohack, 2023-02-22:
When updating from e.g. 12.0.2 or 12.3.1 to 12.4.0 the energy total value is lost. At low values the value is 0 after the update, at high values it is about 40 times smaller.
[image]
[image]
[image]
[image]

Elektrik1, 2023-02-23:
+1, exactly the same situation.

arendst, 2023-02-24:
Gents I need the requested information when you'all openend this issue. Without it would be guessing what's going on.

wohack, 2023-02-24:
It happens during a standard update. Installed was on most of my devices 12.0.2 DE
http://ota.tasmota.com/tasmota/release-12.0.2/tasmota-DE.bin.gz
on two devices 12.3.1 DE, but one of them with a Tuya MCU.
http://ota.tasmota.com/tasmota/release-12.3.1/tasmota-DE.bin.gz 
I upgraded them to minimal.bin
http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz
and then in the next step to 12.4.0 DE
http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz
Nothing special. I recognized that the data was lost on all devices that have the energy total calculated in the software.
The device with the Tuya MCU still has the correct values because the data is stored in the MCU.
After getting awareness of whats happening I updated two other devices where you can see the screenshots.
Allways from 12.0.2 to 12.4.0 as you can see on the bottom of the screenshots
The energy total values are different after the update. energy today and energy yesterday are unchanged.
0.067 to 0.000 on the one and 16.651 to 0.300 on the other.
After the screenshots I opened the ticket.

arendst, 2023-02-24:
Since v12.4.0 the internal energy totals representation should have been divided by 100 to accomodate larger total values. For a user this should have been "invisible". Tests we did indeed provided the expected results.

What do you see if you set ``EnergyRes 5``? It shows more resolution and might show instead of 0.067 to 0.000, 0.00067

arendst, 2023-02-24:
I think I know where it goes wrong.

When a newer version tasmota is installed it checks if the version number has changed and if so performs settings changes supported by the newer version.

In this case it performs the new energy total calculation in both settings AND RTC storage.

The minimal version is supposed the be not updating settings and that works fine except this time I also made the energy total change in RTC memory which is now executed twice. Once by the minimal version and once by the final version.

As I test without the intermediate minimal version I didn't see this double divide.

I'll do some tests and will change the minimal version not to update both settings AND RTC storage.

arendst, 2023-02-24:
- Fixed in dev branch
- Replaced files http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz and http://ota.tasmota.com/tasmota/release-12.4.0/tasmota-minimal.bin.gzwith fix too.

Jason2866, 2023-02-24:
@arendst Did this bug only affected esp8266 devices?

arendst, 2023-02-24:
Not sure. I expect safeboot not to update the RTC but perhaps it's best I push the fix to master and generate new safeboot/factory images too.

arendst, 2023-02-25:
Done. It's now fixed in latest master release from February 24th for all binaries.

closed 2023-02-25 by Jason2866

wohack, 2023-02-26:
I agree, tested on multiple devices, it's fixed