arendst/Tasmota #18029: Fade issues after upgarding to 12.4.0 from 10.1.0

cooldude204, 2023-02-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After Upgrading to 12.4.0 the fade stoped working when turning off and on any bulbs that I upgraded. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Teckin SB50 , Teckin SB53 
- [ X] Tasmota binary firmware version number used: 12.4.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: OTA
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: "NAME":"Teckin SB53","GPIO":[0,0,0,0,416,419,0,0,417,420,418,0,0,0],"FLAG":0,"BASE":18

```
- [X ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: none

```
- [ X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:17:16.217 MQT: Flood_light_1/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Flood light 1","FriendlyName":["Flood light 1"],"Topic":"Flood_light_1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:17:16.223 MQT: Flood_light_1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"1T18:25:14","StartupUTC":"2023-02-21T04:52:02","Sleep":50,"CfgHolder":4617,"BootCount":61,"BCResetTime":"2020-04-09T20:45:01","SaveCount":4873,"SaveAddress":"F4000"}}
00:17:16.228 MQT: Flood_light_1/stat/STATUS2 = {"StatusFWR":{"Version":"12.4.0(tasmota)","BuildDateTime":"2023-02-18T16:07:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"432/699"}}
00:17:16.232 MQT: Flood_light_1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOTWIFI_2GHz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028029","2805C80001000600003C5A0A002800000000","00000280","00006200","00004000","00000000"]}}
00:17:16.244 MQT: Flood_light_1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14400B","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
00:17:16.252 MQT: Flood_light_1/stat/STATUS5 = {"StatusNET":{"Hostname":"Flood-light-1-1640","IPAddress":"192.168.10.177","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.10.1","DNSServer2":"0.0.0.0","Mac":"68:C6:3A:ED:26:68","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:17:16.258 MQT: Flood_light_1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ED2668","MqttUser":"HASSIO_IOT","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:17:16.263 MQT: Flood_light_1/stat/STATUS7 = {"StatusTIM":{"UTC":"2023-02-22T23:17:16","Local":"2023-02-23T00:17:16","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:44","Sunset":"18:22"}}
00:17:16.268 MQT: Flood_light_1/stat/STATUS10 = {"StatusSNS":{"Time":"2023-02-23T00:17:16"}}
00:17:16.275 MQT: Flood_light_1/stat/STATUS11 = {"StatusSTS":{"Time":"2023-02-23T00:17:16","Uptime":"1T18:25:14","UptimeSec":152714,"Heap":21,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":6,"Color":"0,0,0,0,16","HSBColor":"45,86,0","White":6,"CT":500,"Channel":[0,0,0,0,6],"Scheme":0,"Fade":"ON","Speed":3,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IOTWIFI_2GHz","BSSId":"FA:92:BF:5D:0B:1F","Channel":11,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:15"}}}
00:17:24.398 MQT: Flood_light_1/tele/STATE = {"Time":"2023-02-23T00:17:24","Uptime":"1T18:25:22","UptimeSec":152722,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":6,"Color":"0,0,0,0,16","HSBColor":"45,86,0","White":6,"CT":500,"Channel":[0,0,0,0,6],"Scheme":0,"Fade":"ON","Speed":3,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IOTWIFI_2GHz","BSSId":"FA:92:BF:5D:0B:1F","Channel":11,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:15"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

not sure

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Fade on should fade the lights but isn't 
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I'm using the Tasmota Integration in homeassistant 

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-02-23:
Which command or setup do you have for fading? You have provided no information about.

Jason2866, 2023-03-07:
Closing since no feedback

closed 2023-03-07 by Jason2866