arendst/Tasmota #18034: Xiaomi temperature+humidity sensor with atc441 shows wrong humidity value

odelma, 2023-02-23:
### PROBLEM DESCRIPTION
Xiaomi temperature+humidity sensor with atc441 shows wrong humidity value (perhaps what is actually reported as battery perccentage). Humidity might be showing the battery (as 0-255).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  XIAOMI Mijia Bluetooth Thermometer
- [ ] Tasmota binary firmware version number used: 12.4.0(bluetooth)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:05:14.404 MQT: stat/tasmota_3C6B7C/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
16:05:14.636 MQT: stat/tasmota_3C6B7C/RESULT = {"Module":{"1":"ESP32-DevKit"}}
16:05:14.896 MQT: stat/tasmota_3C6B7C/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
16:06:46.520 MQT: stat/tasmota_3C6B7C/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":294,"Free":217,"Rules":"ON ATC28b25a#temperature DO var1 %value% ENDON   ON ATC28b25a#battery DO var2 %value% ENDON   ON ATC28b25a#rssi DO var3 %value% ENDON   ON ATC28b25a#humidity DO publish domoticz/in {\"idx\":905,\"dtype\":\"Temp + Humidity\",\"nvalue\":0,\"svalue\":\"%var1%;%value%;1\",\"RSSI\":%var3%,\"Battery\":%var2%} ENDON"}}
16:06:46.720 MQT: stat/tasmota_3C6B7C/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
16:06:46.972 MQT: stat/tasmota_3C6B7C/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}} 

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:07:38.365 MQT: stat/tasmota_3C6B7C/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_3C6B7C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:07:38.409 MQT: stat/tasmota_3C6B7C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-bluetooth.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:13:36","StartupUTC":"2023-02-23T14:54:02","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2023-02-23T14:48:57","SaveCount":20}}
16:07:38.449 MQT: stat/tasmota_3C6B7C/STATUS2 = {"StatusFWR":{"Version":"12.4.0(bluetooth)","BuildDateTime":"2023-02-18T16:10:08","Core":"2_0_6","SDK":"4.4.3.20230129","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"438/699"}}
16:07:38.466 MQT: stat/tasmota_3C6B7C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AOmokki",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004002","00000000"]}}
16:07:38.504 MQT: stat/tasmota_3C6B7C/STATUS4 = {"StatusMEM":{"ProgramSize":1635,"Free":1244,"Heap":80,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8F9AC7CF","00148001","000000CF","010013C0","C0000981","40004080","00200000","5400082C","00000080"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,59,62,79,82,85","Sensors":"1,2,3,5,6,52,62,127"}}
16:07:38.557 MQT: stat/tasmota_3C6B7C/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3C6B7C-2940","IPAddress":"192.168.1.147","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"AC:67:B2:3C:6B:7C","IP6Global":"fdf6:7d11:eebd:4a35:ae67:b2ff:fe3c:6b7c","IP6Local":"fe80::ae67:b2ff:fe3c:6b7c","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:07:38.619 MQT: stat/tasmota_3C6B7C/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3C6B7C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:07:38.649 MQT: stat/tasmota_3C6B7C/STATUS7 = {"StatusTIM":{"UTC":"2023-02-23T15:07:38","Local":"2023-02-23T16:07:38","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:44","Sunset":"18:22"}}
16:07:38.669 MQT: stat/tasmota_3C6B7C/STATUS10 = {"StatusSNS":{"Time":"2023-02-23T16:07:38"}}
16:07:38.677 MQT: stat/tasmota_3C6B7C/STATUS11 = {"StatusSTS":{"Time":"2023-02-23T16:07:38","Uptime":"0T00:13:36","UptimeSec":816,"Heap":77,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"Wifi":{"AP":1,"SSId":"AOmokki","BSSId":"50:D4:F7:D2:32:56","Channel":4,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Sensor was autodiscovered and showing incorrect values directly after this for humidity.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expecting so see correct humidity

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._Sensor was autodiscovered and showing the humidity in this way directly.
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-02-24:
atc441 firmware is no more maintained as afr as I know
you should switch to PVVX which is a fork of atc441 with a lot of improvement on battery management

odelma, 2023-02-24:
Thanks, switching to PVVX fork seems to have resolved it. I think the original problem wast that I had experimented with setting the battery level to show on screen and somehow that was "stuck" even after I tried to change the setting. Now, flashing with PVVX atc_v42.bin resolved it. It might have been resolved with just refalshing the original atc1441. But all good now in any case 👍

closed 2023-02-24 by odelma