arendst/Tasmota #18037: INA226 not working on ESP8266 or ESP32

VASTechAD, 2023-02-23:
### PROBLEM DESCRIPTION
I attach INA226 to I2c bus and tasmota detect module as INA219. I self-compiled a firmware with I2cDriver35, disabled from console I2cDriver14 for INA219, activated driver for INA226 with I2cDriver35 1. I try on both esp8266 and esp32.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP-01 and ESP32
- [ ] Tasmota binary firmware version number used: 12.4.0.1
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: WebFlash
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
CMD: Backlog Template
RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
CMD: Module
RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
CMD: GPIO 255`
RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"608":"I2C SCL1"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA1"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:48:27.868 CMD: Status 0
22:48:27.872 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_804EB0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:48:27.882 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:19:26","StartupUTC":"2023-02-23T21:29:01","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2023-02-23T22:16:14","SaveCount":35}}
22:48:27.891 RSL: STATUS2 = {"StatusFWR":{"Version":"12.4.0.1(tasmota32)","BuildDateTime":"2023-02-23T22:19:11","Core":"2_0_7","SDK":"4.4.4.20230218","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"403/699"}}
22:48:27.895 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vasile-NoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
22:48:27.910 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1487,"Free":560,"Heap":153,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DD7","480840D2","20200000","D43C482D","000080B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,54,55,56,58,59,64,66,67,74,85,92,95,98,103,127"}}
22:48:27.925 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-804EB0-3760","IPAddress":"10.10.10.37","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer1":"10.10.10.1","DNSServer2":"0.0.0.0","Mac":"78:21:84:80:4E:B0","IP6Global":"","IP6Local":"fe80::7a21:84ff:fe80:4eb0","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.10.10.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:48:27.933 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_804EB0","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:48:27.938 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-02-23T21:48:27","Local":"2023-02-23T22:48:27","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:44","Sunset":"18:22"}}
22:48:27.947 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-02-23T22:48:27"}}
22:48:27.954 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-02-23T22:48:27","Uptime":"0T00:19:26","UptimeSec":1166,"Heap":156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":43},"Wifi":{"AP":1,"SSId":"Vasile-NoT","BSSId":"7A:45:58:AF:F9:DD","Channel":6,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:02"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Plug INA226 device on I2c bus and see results_

### EXPECTED BEHAVIOUR
_To show correct device in main window, and in MQTT ._

### SCREENSHOTS
_PS Picture attached are from ESP8266 but issue is the same in ESP32_
_I2cDriver14 detects module INA 226 as INA219_
_[image]_
_I2cDriver35 ... nothing_
_[image]_

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-02-23:
Your build includes multiple drivers that use the same address as INA219 (14) and INA226 (35) such as HTU (9)
In order to use a INA226, you MUST disable drivers that you don't need and that can conflict
```
backlog i2cdriver14 0; i2cdriver15 0; i2cdriver; i2cdriver35 1
```

VASTechAD, 2023-02-24:
I`ve compiled a version for ESP8266 with i2cdriver35 only.

```lua
09:23:10.324 CMD: i2cdriver
09:23:10.330 MQT: stat/tasmota_24F95C/RESULT = {"I2CDriver":"35"}
```
```lua
09:26:33.277 CMD: i2cscan
09:26:33.299 MQT: stat/tasmota_24F95C/RESULT = {"I2CScan":"Device(s) found at 0x40"}
```
Same result:
[image]

Jason2866, 2023-02-24:
Please post your `user_config_override.h`.

VASTechAD, 2023-02-24:
@Jason2866 
user_config_override.h is default, no modification there.

VASTechAD, 2023-02-24:
I see a post were someone use build 9.5.0.1 and INA226 works there. Anyone knows where i can download source code for old version of tasmota?

sfromis, 2023-02-24:
Since you have only `{"I2CDriver":"35"}` you did modify your build to get to that, the key is not if it specifically was done in `user_config_override.h` (which is generally the preferred place for such) or not.

Jason2866, 2023-02-24:
Sorry, without providing relevant Infos we asked for, to help you with YOUR problem nobody can help.