arendst/Tasmota #18040: Rules not working on v12.4.0 Peter

enricota66, 2023-02-24:
### PROBLEM DESCRIPTION
Rules not working on v12.4.0 Peter on esp8266 device

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Nodemcu esp8266
- [ ] Tasmota binary firmware version number used: Tasmota firmware 12.4.0 on ESP8266
  - [ ] Self-compiled
- [ ] Flashing tools used: Visual s\Studio/ Platformio
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: (use rule1 only)
[09:46:27] cmnd/NodePT2/rule1 
[09:46:27] stat/NodePT2/RESULT {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":102,"Free":409,"Rules":"on switch3#state do publish stat/NodePT2/MOTION %value% endon on Switch3#State do Power3 %value% endon"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
[09:48:23] cmnd/NodePT2/Status 0
[09:48:23] stat/NodePT2/STATUS {"Status":{"Module":0,"DeviceName":"NodePT2","FriendlyName":["Mobile","StickDue","PirSoggiorno","CamSogg"],"Topic":"NodePT2","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
[09:48:23] stat/NodePT2/STATUS1 {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:20:43","StartupUTC":"2023-02-24T08:27:40","Sleep":100,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2023-02-23T23:24:38","SaveCount":6583,"SaveAddress":"FA000"}}
[09:48:23] stat/NodePT2/STATUS2 {"StatusFWR":{"Version":"12.4.0(sensors)","BuildDateTime":"2023-02-24T09:20:37","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"386/699"}}
[09:48:23] stat/NodePT2/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Enlafra","ASUS"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["8002A009","2805C80000000600003C5A0A002800000000","00008280","00006000","00004000","00000000"]}}
[09:48:23] stat/NodePT2/STATUS4 {"StatusMEM":{"ProgramSize":698,"Free":304,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0000C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103"}}
[09:48:23] stat/NodePT2/STATUS5 {"StatusNET":{"Hostname":"NodePT2-4826","IPAddress":"192.168.0.55","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"3C:61:05:CF:D2:DA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
[09:48:23] stat/NodePT2/STATUS6 {"StatusMQT":{"MqttHost":"192.168.0.30","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CFD2DA","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
[09:48:23] stat/NodePT2/STATUS7 {"StatusTIM":{"UTC":"2023-02-24T08:48:23","Local":"2023-02-24T09:48:23","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"07:02","Sunset":"17:58"}}
[09:48:23] stat/NodePT2/STATUS10 {"StatusSNS":{"Time":"2023-02-24T09:48:23","Switch3":"OFF","AM2301":{"Temperature":17.6,"Humidity":71.1,"DewPoint":12.3},"TempUnit":"C"}}
[09:48:23] stat/NodePT2/STATUS11 {"StatusSTS":{"Time":"2023-02-24T09:48:23","Uptime":"0T00:20:43","UptimeSec":1243,"Heap":14,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Enlafra","BSSId":"38:10:D5:BB:46:7E","Channel":1,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
compiled whit
built using core 2.7.4.9
SetOption36 0
#define USE_RULES in user_config_override.h
------------
An pir sensor is a switch3 on D7 GPIO13 pin
when pir sensor is triggered the rule1 do not commute power3 relay and non publish mqtt command

### EXPECTED BEHAVIOUR
On Tasmota Tasmota-11.01 the same configuration working fine

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I have a second board at home whit Tasmota-11.01 and pir sensore and same rule work fine

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-02-24:
Most probably config error. It was never a valid config to start switch and relay numbering NOT with "1".  Furthermore the counting has to be continous. No double numbers allowed.
Since v12.4. this is more strict.

Change you switch number to "1". Does it work with?

enricota66, 2023-02-24:
" Most probably config error. It was never a valid config to start switch and relay numbering NOT with "1". "
Sorry I'm not sure to understand, I need to set the switch and relay from 3 to 1  for use the tasmota rule? 
Or I need that for other reason ?
Because I never read this on Rule documentation.....
Anyway I can try but this change a lot of setting on my HomeAssitant configuration.

Jason2866, 2023-02-24:
You posted no config you use. Without that only guess work is possible if your config is valid or not.

enricota66, 2023-02-24:
> You posted no config you use. Without that only guess work is possible if your config is valid or not. What you posted does not contain anything
> 
> ```
>   {"NAME":"NodePT2","GPIO":[0,0,1,1,1,1,0,0,0,1,1,0,0,0],"FLAG":0,"BASE":18}
> ```

I replied with an update to the previous comment Thanks again

closed 2023-02-24 by Jason2866

sfromis, 2023-02-24:
Switch1 cannot control Relay1 or Relay3, but default actions of Switch or Button inputs are to control power channels, and power channels are not merely Relay pins, but also PWM, neopixels, lights or displays. Hence, the numbers cannot be used to indicate pairing, just the ordering within each type.

enricota66, 2023-02-24:
@sfromis 
In fact it was what I had understood in the past, at this point I was wrong