arendst/Tasmota #18043: Tasmota (Sensors) stopped working CCS811 in v12.4.0

jlclark440, 2023-02-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP8266 CCS811
- [X] Tasmota binary firmware version number used: v12.4.0 sensors
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmota GUI
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: 12:47:34.872 CMD: Backlog Template; Module; GPIO 255
12:47:34.938 MQT: stat/Nodemcu1/RESULT = {"NAME":"Environment Monitor 1","GPIO":[1312,0,0,0,608,640,0,0,0,1216,0,0,0,0],"FLAG":0,"BASE":18}
12:47:35.174 MQT: stat/Nodemcu1/RESULT = {"Module":{"0":"Environment Monitor 1"}}
12:47:35.430 MQT: stat/Nodemcu1/RESULT = {"GPIO0":{"1312":"DS18x201"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"1216":"AM2301"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  12:48:04.339 CMD: Status 0
12:48:04.348 MQT: stat/Nodemcu1/STATUS = {"Status":{"Module":0,"DeviceName":"Nodemcu1","FriendlyName":["Nodemcu1"],"Topic":"Nodemcu1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:48:04.385 MQT: stat/Nodemcu1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:58","StartupUTC":"2023-02-24T17:45:06","Sleep":50,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2022-08-23T03:25:36","SaveCount":2041,"SaveAddress":"F9000"}}
12:48:04.418 MQT: stat/Nodemcu1/STATUS2 = {"StatusFWR":{"Version":"12.4.0(sensors)","BuildDateTime":"2023-02-24T15:32:29","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"464/699"}}
12:48:04.439 MQT: stat/Nodemcu1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ASUS_E0_2G","ASUS_E1_2G"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028109","2805C80001000600003C5A0A002800000000","00008280","00006000","00004000","00000000"]}}
12:48:04.477 MQT: stat/Nodemcu1/STATUS4 = {"StatusMEM":{"ProgramSize":698,"Free":304,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0000C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103"}}
12:48:04.520 MQT: stat/Nodemcu1/STATUS5 = {"StatusNET":{"Hostname":"nodemcu1","IPAddress":"192.168.0.241","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"48:3F:DA:0D:F3:CE","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:48:04.543 MQT: stat/Nodemcu1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DF3CE","MqttUser":"JLClark1","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:48:04.567 MQT: stat/Nodemcu1/STATUS7 = {"StatusTIM":{"UTC":"2023-02-24T17:48:04","Local":"2023-02-24T12:48:04","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"-05:00","Sunrise":"01:42","Sunset":"12:24"}}
12:48:04.592 MQT: stat/Nodemcu1/STATUS10 = {"StatusSNS":{"Time":"2023-02-24T12:48:04","DS18B20-1":{"Id":"01193A97C92C","Temperature":73.5},"DS18B20-2":{"Id":"01204BE18C75","Temperature":74.1},"DS18B20-3":{"Id":"01204BF1B582","Temperature":75.6},"DS18B20-4":{"Id":"01204C076509","Temperature":76.0},"AM2301":{"Temperature":75.7,"Humidity":48.6,"DewPoint":55.0},"TempUnit":"F"}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here: 12:49:47.083 CMD: i2cscan
12:49:47.086 SRC: WebConsole from 192.168.1.16
12:49:47.089 CMD: Grp 0, Cmd 'I2CSCAN', Idx 1, Len 0, Pld -99, Data ''
12:49:47.110 MQT: stat/Nodemcu1/RESULT = {"I2CScan":"No devices found"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upgrade from 12.3.1 to 12.4.0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
CCS811 sensor not detected after upgrade, reverting to 12.3.1 with out any other changes restores the sensor .

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
i2cscan on 12.3.1 results in expected response; 12:55:48.893 CMD: i2cscan
12:55:48.916 MQT: stat/Nodemcu1/RESULT = {"I2CScan":"Device(s) found at 0x5a"}

**(Please, remember to close the issue when the problem has been addressed)**
[image]
[image]
[image]
[image]

Jason2866, 2023-02-24:
The driver is not in the list?? It is sensor driver number 24
```
12:48:04.477 MQT: stat/Nodemcu1/STATUS4 = {"StatusMEM":{"ProgramSize":698,"Free":304,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0000C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103"}}
```
Just checked the source code, see no reason why it should not be in the sensor firmware

jlclark440, 2023-02-24:
I've since reverted back to 12.3.1 but I will reinstall 12.4 and double check on that tomorrow morning.

sfromis, 2023-02-24:
Jason, the numbers around drivers are a bit confusing, thus:

- The list in `Drivers` is not I2C drivers, but various non-sensor drivers, meaning that CCS811 is not visible there.
- The list in `Sensors` are not `I2CDriver`, but I2C drivers are included, with CCS811 having number 31. Yes.
- The bits in `Features` can also be decoded to reveal that `USE_CCS811` is included in the build. **ok**
- The output of `Status 0` tells nothing about `I2CDriver` configuration, that would need to be a separate command.

jlclark440, 2023-02-24:
I should be able to provide an update possibly later today and tomorrow morning at the latest. Assuming that all previously set values are still the same after the upgrade, I'm not quite sure what has changed. As nothing to the configuration was changed.

sfromis, 2023-02-24:
For me, hooking a CCS811 up to Tasmota 12.4.0 worked right away, both on ESP8266 and ESP32. This test includes the same exact binary version as indicated in the `Status 0` output in the initial post here.

Jason2866, 2023-02-24:
> Jason, the numbers around drivers are a bit confusing, thus:
>
> * The list in `Drivers` is not I2C drivers, but various non-sensor drivers, meaning that CCS811 is not visible there.
> * The list in [...]

Thx sfromis. One more time stumbled over this :-)

jlclark440, 2023-02-24:
For what it's worth, I do have an additional device that exhibits the same behavior that may be useful for testing. I wasn't able to see any correlation between the two.

Jason2866, 2023-02-26:
@jlclark440  Have you done more testing? We can't reproduce the problem.

sfromis, 2023-02-26:
Since CCS811 needs more power than most sensors, like 30-54 mA due to the heater, it may be more susceptible to borderline availability of enough current on 3.3V power input.

jlclark440, 2023-02-26:
Working on it now, have one device running 12.4.0 and the other still on 12.3.1. 

12.4
10:31:39.771 CMD: i2cscan
10:31:39.773 SRC: WebConsole from 192.168.1.16
10:31:39.776 CMD: Grp 0, Cmd 'I2CSCAN', Idx 1, Len 0, Pld -99, Data ''
10:31:39.798 MQT: stat/Nodemcu2/RESULT = {"I2CScan":"No devices found"}

12.3.1
10:31:35.809 CMD: i2cscan
10:31:35.811 SRC: WebConsole from 192.168.1.16
10:31:35.814 CMD: Grp 0, Cmd 'I2CSCAN', Idx 1, Len 0, Pld -99, Data ''
10:31:35.835 MQT: stat/Nodemcu1/RESULT = {"I2CScan":"Device(s) found at 0x5a"}

sfromis, 2023-02-26:
As long as you have `{"I2CScan":"No devices found"}`, my expectation would be that your issue is outside of Tasmota, very plausibly at the "physical level" around hardware, wiring, connections, noise and power. Tasmota can't really do anything if a device does not pop up when scanning the I2C bus for devices.

Jason2866, 2023-02-26:
Please list/describe your hardware and setup. Powers supply, used board and wiring.
A photo of the complete setup would help too.

Jason2866, 2023-03-01:
ping @jlclark440 or can we close?

jlclark440, 2023-03-02:
Closing, as reverting to a previous version does not exhibit this current issue.

closed 2023-03-02 by jlclark440