arendst/Tasmota #18048: Integer overflow for TuyaMCU timers (uint16 vs. int16)

aph3rson, 2023-02-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

For TuyaMCU devices, Tasmota offers support for tracking of timer sensors (Timer1 through Timer4). These are mentioned as handling type 2 (integer) DP Ids.

However, there appears to be a signedness issue somewhere in the handling of these timers. As an example, this DP is returning a value of 36000 (`0x8ca0`), which corresponds correctly to the sensor value as an uint16 (10 hours, in seconds). However, Tasmota appears to be processing it as an int16, and reports the value as -29536 (same bit order, just incorrectly handling the sign bit).

```lua
08:13:21.583 {"TuyaReceived":{"Data":"55AA030700081D02000400008CA060","Cmnd":7,"CmndData":"1D02000400008CA0","DpType2Id29":36000,"29":{"DpId":29,"DpIdType":2,"DpIdData":"00008CA0"}}}
08:13:21.585 TYA: fnId=82 is set for dpId=29
08:13:21.588 MQT: tele/tasmota_D76E47/SENSOR = {"TuyaSNS":{"Timer2":-29536}}
08:13:21.591 TYA: RX value 36000 from dpId 29 
```

It's unclear if these timers are intended to be always-signed, or always-unsigned.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Tuya MCU (Sunbeam Heated Blanket)
- [X] Tasmota binary firmware version number used: 12.4.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: ESP8266 flash jig
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
08:03:14.049 CMD: Backlog Template; Module; GPIO 255
08:03:14.051 SRC: WebConsole from 192.168.0.8
08:03:14.054 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
08:03:14.122 SRC: Backlog
08:03:14.125 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
08:03:14.132 MQT: stat/tasmota_D76E47/RESULT = {"NAME":"Sunbeam Heated Blanket","GPIO":[0,2272,0,2304,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":54}
08:03:14.337 SRC: Backlog
08:03:14.341 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
08:03:14.349 MQT: stat/tasmota_D76E47/RESULT = {"Module":{"0":"Sunbeam Heated Blanket"}}
08:03:14.589 SRC: Backlog
08:03:14.592 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
08:03:14.605 MQT: stat/tasmota_D76E47/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [X] Provide the output of this command: `Status 0`:
```lua
08:04:32.472 CMD: status 0
08:04:32.474 SRC: WebConsole from 192.168.0.8
08:04:32.477 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
08:04:32.485 MQT: stat/tasmota_D76E47/STATUS = {"Status":{"Module":0,"DeviceName":"Sunbeam Heated Blanket","FriendlyName":["Master Power","Master Preheat","SideA Power","SideB Power","SideA Preheat","SideB Preheat"],"Topic":"tasmota_D76E47","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:04:32.492 MQT: stat/tasmota_D76E47/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T03:24:41","StartupUTC":"2023-02-25T03:39:51","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2023-02-23T09:13:35","SaveCount":111,"SaveAddress":"F5000"}}
08:04:32.497 MQT: stat/tasmota_D76E47/STATUS2 = {"StatusFWR":{"Version":"12.4.0(tasmota)","BuildDateTime":"2023-02-18T16:07:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"564/699"}}
08:04:32.503 MQT: stat/tasmota_D76E47/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["__SNIP__",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
08:04:32.515 MQT: stat/tasmota_D76E47/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
08:04:32.524 MQT: stat/tasmota_D76E47/STATUS5 = {"StatusNET":{"Hostname":"tasmota-D76E47-3655","IPAddress":"192.168.0.209","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"205.171.3.65","Mac":"58:BF:25:D7:6E:47","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:04:32.529 MQT: stat/tasmota_D76E47/STATUS6 = {"StatusMQT":{"MqttHost":"__SNIP__","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D76E47","MqttUser":"_tasmota_D76E47","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:04:32.536 MQT: stat/tasmota_D76E47/STATUS7 = {"StatusTIM":{"UTC":"2023-02-25T07:04:32","Local":"2023-02-25T08:04:32","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:40","Sunset":"18:25"}}
08:04:32.544 MQT: stat/tasmota_D76E47/STATUS10 = {"StatusSNS":{"Time":"2023-02-25T08:04:32","TuyaSNS":{"Timer1":-29536,"Timer2":-29536}}}
08:04:32.551 MQT: stat/tasmota_D76E47/STATUS11 = {"StatusSTS":{"Time":"2023-02-25T08:04:32","Uptime":"0T03:24:41","UptimeSec":12281,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","Wifi":{"AP":1,"SSId":"__SNIP__","BSSId":"__SNIP__","Channel":1,"Mode":"11n","RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Locate a TuyaMCU device which utilizes an integer DP that is out-of-range of an int16.
2. Configure said device to output that integer to a timer FnID, e.g. 21 (Timer1).
3. Observe that the value is processed as if it were an int16, and exhibits integer overflow.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Proper processing of the integer in question.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Log outputs are above.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I think this is the first time someone's put Tasmota on one of these - they're not popular devices, but they should be similar to most-other Tuya devices. Note that the Tuya electric blanket docs (device type `dr`) do _not_ appear to be accurate for this device.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-02-25:
What are tuya timer used for?
Is there any use case where a negative time would make sense?

I'll try to have a look in Tuya docs to see if they say something

aph3rson, 2023-02-25:
> What are tuya timer used for?

In this device, it's to indicate how many seconds are left until the device shuts off (sleep mode timer).

> Is there any use case where a negative time would make sense?

Not for this device/one that I'm aware of.

aph3rson, 2023-02-25:
Also, FWIW, I ended up using a workaround here: the DPs for the timer are unbound, but still detected with a rule on a TuyaReceived event:

```
Rule1 
  ON TuyaReceived#DpType2Id28 DO Publish2 stat/%topic%/TIMER1 %value% ENDON 
  ON TuyaReceived#DpType2Id29 DO Publish2 stat/%topic%/TIMER2 %value% ENDON
```

barbudor, 2023-03-01:
I've not been able to find a confirmation in Tuya doc but that seems reasonnable to have a timer unsigned
Added as point 3) in PR https://github.com/arendst/Tasmota/pull/18085

Jason2866, 2023-03-03:
Fixed PR is merged.

closed 2023-03-03 by Jason2866