arendst/Tasmota #18060: SSerialSend5 data in serial response messages is NOT encoded as hex strings

lulol, 2023-02-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
`SSerialSend5` data in serial response messages is NOT encoded as hex strings unless forced with `SerialDelimiter 254`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP12E
- [x] Tasmota binary firmware version number used: 12.4.0.1, 12.4.0, 12.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:32:46.837 CMD: Backlog Template; Module; GPIO 255
22:32:46.857 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:32:47.086 RSL: RESULT = {"Module":{"18":"Generic"}}
22:32:47.293 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"1824":"SerBr Rx"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1792":"SerBr Tx"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:34:13.319 CMD: Status 0
22:34:13.325 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_XXXXXX","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:34:13.330 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:36","StartupUTC":"2023-02-26T21:23:37","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2023-01-26T17:30:34","SaveCount":37,"SaveAddress":"F8000"}}
22:34:13.335 RSL: STATUS2 = {"StatusFWR":{"Version":"12.4.0.1(tasmota)","BuildDateTime":"2023-02-26T19:53:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"340/699"}}
22:34:13.340 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
22:34:13.351 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
22:34:13.356 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-XXXXXX-XXXX","IPAddress":"192.168.x.x","Gateway":"192.168.x.x","Subnetmask":"255.255.x.x","DNSServer1":"192.168.x.x","DNSServer2":"0.0.0.0","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:34:13.360 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_XXXXXX","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:34:13.366 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-02-26T21:34:13","Local":"2023-02-26T22:34:13","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:38","Sunset":"18:27"}}
22:34:13.371 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-02-26T22:34:13"}}
22:34:13.377 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-02-26T22:34:13","Uptime":"0T00:10:36","UptimeSec":636,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"XXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":1,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
See console output below.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
`SerialSend5 dummy` command configuring the data in serial response messages encoded as hex strings as described in the docs and as how it works for [SerialSend5](https://tasmota.github.io/docs/Commands/#serialsend).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The intention is to remotely monitor an internal bidirectional serial communication link of a mobile robot using tasmota's Serial Rx and SerBr Rx (via console or mqtt).
```
00:00:00.001 HDW: ESP8266EX
00:00:00.046 CFG: Loaded from flash at FB, Count 33
00:00:00.159 QPC: Reset
00:00:00.180 Project tasmota - Tasmota Version 12.4.0.1(tasmota)-2_7_4_9(2023-02-26T19:53:33)
00:00:00.770 WIF: Connecting to AP1 XXX Channel 1 BSSId XX:XX:XX:XX:XX:XX in mode 11n as tasmota-XXXXXX-XXXX...
00:00:01.755 WIF: Connected
00:00:02.009 HTP: Web server active on tasmota-XXXXXX-XXXX with IP address 192.168.x.x
22:23:41.030 RSL: INFO1 = {"Info1":{"Module":"Generic","Version":"12.4.0.1(tasmota)","FallbackTopic":"cmnd/DVES_XXXXXX_xx/","GroupTopic":"cmnd/tasmotas/"}}
22:23:41.032 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-XXXXXX-XXXX","IPAddress":"192.168.x.x"}}
22:23:41.033 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":13}}
22:23:45.463 RSL: STATE = {"Time":"2023-02-26T22:23:45","Uptime":"0T00:00:08","UptimeSec":8,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"XXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
22:23:57.132 CMD: Baudrate 115200
22:23:57.135 RSL: RESULT = {"Baudrate":115200}
22:23:58.934 CMD: SBaudrate 115200
22:23:58.939 RSL: RESULT = {"SBaudrate":115200}
22:24:10.420 CMD: SerialSend5 dummy
22:24:10.424 RSL: RESULT = {"SerialSend":"Done"}
22:24:12.931 CMD: SSerialSend5 dummy
22:24:12.935 RSL: RESULT = {"SSerialSend":"Done"}
22:24:33.634 CMD: SerialDelimiter 255
22:24:33.638 RSL: RESULT = {"SerialDelimiter":255}
22:24:40.042 RSL: RESULT = {"SerialReceived":"534144"}
22:24:46.078 RSL: RESULT = {"SSerialReceived":"SAD"}
22:25:03.263 CMD: SerialDelimiter 254
22:25:03.267 RSL: RESULT = {"SerialDelimiter":254}
22:25:07.543 RSL: RESULT = {"SerialReceived":"534144"}
22:25:11.139 RSL: RESULT = {"SSerialReceived":"534144"}
```

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-02-26:
You misunderstood the documentation
SerialSend5 is meant to be used with an HEX payload
To send an ASCII payload, use SerialSend1 or SerialSend2

See example here : https://tasmota.github.io/docs/Rules/#using-dummy-gpio-to-send-serial-codes-to-an-mcu

working as designed

lulol, 2023-02-27:
_Before data will be received, a properly formatted SerialSend<x> or SSerialSend<x> command must be executed._

The `SerialSend5` payload** was just to configure the data type in serial response messages.
To configure the format of the received data it seems to work the same regardless of the data type used (or not being properly formatted).

The issue is in the SSerialReceived data not being formated as HEX following a properly formatted  `SerialSend5 <string>` command.

As I understand it should be the same as with SerialReceived where data in serial response messages is encoded as hex strings following a `SerialSend5 <string>` command or as text following a `SerialSend2 <string>` command.

SerialReceived data follows the `SerialSend5` configuration but `SSerialSend5` has no effect in SSerialReceived data that is in both cases shown the same as text.

Using a properly formatted `SSerialSend5` command makes no difference for the issue.
```
00:27:00.105 CMD: Baudrate 115200
00:27:00.109 RSL: RESULT = {"Baudrate":115200}
00:27:04.082 CMD: SBaudrate 115200
00:27:04.087 RSL: RESULT = {"SBaudrate":115200}
00:27:08.298 CMD: SerialDelimiter 255
00:27:08.302 RSL: RESULT = {"SerialDelimiter":255}
00:27:21.292 CMD: SerialSend5 ff
00:27:21.296 RSL: RESULT = {"SerialSend":"Done"}
00:27:25.875 CMD: SSerialSend5 ff
00:27:25.880 RSL: RESULT = {"SSerialSend":"Done"}
00:27:37.322 RSL: RESULT = {"SerialReceived":"534144"}   <<<<<<< I sent "SAD" to GPIO3
00:27:43.876 RSL: RESULT = {"SSerialReceived":"SAD"}     <<<<<<< I sent "SAD" to GPIO4
00:28:09.298 CMD: SerialSend2 TEXT
00:28:09.303 RSL: RESULT = {"SerialSend":"Done"}
00:28:17.912 RSL: RESULT = {"SerialReceived":"SAD"}
00:28:28.278 CMD: SSerialSend2 TEXT
00:28:28.284 RSL: RESULT = {"SSerialSend":"Done"}
00:28:31.712 RSL: RESULT = {"SSerialReceived":"SAD"}
```
---
 ** Changed the payload to "dummy" trying to be more descriptive for the issue and forgot that wasn't hex. And didn't notice since it doesn't generate any kind of error or warning, which probably is itself another added issue.
```
22:24:10.420 CMD: SerialSend5 dummy
22:24:10.424 RSL: RESULT = {"SerialSend":"Done"}
```
This actual hardware have both TX pins unconnected, the ESP only receives external serial data from both TX and RX lines of the serial link which transfers binary data and the data should be presented as HEX for analysis.

Currently using `SerialDelimiter 254` as workaround.

barbudor, 2023-02-27:
That is clearer, maybe change the title such as "SSerialReceive doesn't display HEX despite using SSerialSend5"
Thanks
I'll look into that later today or tomorrow

lulol, 2023-02-27:
\*shrug\* I haven looked at the SerialDelimiter code or the rationale to use it to set the output format to HEX. For what I can guess has a dual function also acting as a filter.

In my opinion SerialSend5 and SSerialSend5 options should behave coherently.

So if SSerialSend5 won't configure SSerialReceive output format to HEX, neither SerialSend5 should configure SerialReceive output format to HEX.

But `SerialDelimiter 254` to set the output format to HEX for both SerialReceive and SSerialReceive seems ok, although that would limit choosing the output format independently for each stream if needed.

void-spark, 2023-03-26:
Just got hit by this, documentation update would be much appreciated :)

closed 2023-04-25 by github-actions[bot]