arendst/Tasmota #18073: Internal thermostat heating action is activated when is not expected

soyxan, 2023-02-28:
### PROBLEM DESCRIPTION
I am using the Tasmota thermostat implementation (https://tasmota.github.io/docs/Thermostat/) and overall is working ok, but the following screenshots show a strange situation that is occurring very often:

- Blue pen: Current room temperature
- Purple pen: Temperature setpoint
- Orange area: Heating action

As it can be seen in the first screenshot, the heating action is working as expected from 18:00 to 21:00 (the setpoint is 23ºC and the current temperature is below, so the heating action gets activated) but at around 5:30 the heating action is activated again when according to the setpoint is not needed.

In the second screenshot, it is shown a room with constant temperature setpoint for all day (19ºC) and from time to time the heating action is activated when not expected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NSPanel
- [x] Tasmota binary firmware version number used: 12.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled (enabling THERMOSTAT: `#define USE_THERMOSTAT`)
- [x] Flashing tools used: ESPTool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:25:25.711 MQT: stat/tasmota_FA4C08/RESULT = {"NAME":"NSPanel","GPIO":[0,0,0,0,3872,0,0,0,0,0,32,0,0,0,0,225,0,480,224,1,0,0,0,33,0,0,0,0,0,0,0,0,0,0,4736,0],"FLAG":0,"BASE":1}
15:25:25.909 MQT: stat/tasmota_FA4C08/RESULT = {"Module":{"0":"NSPanel"}}
15:25:26.121 MQT: stat/tasmota_FA4C08/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"3872":"Output Lo"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"32":"Button1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"225":"Relay2"},"GPIO20":{"0":"None"},"GPIO21":{"480":"Buzzer"},"GPIO22":{"224":"Relay1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"33":"Button2"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"4736":"ADC Temp1"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
    Rules output here:
15:28:05.036 MQT: stat/tasmota_FA4C08/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":55,"Free":456,"Rules":"ON ANALOG#Temperature1 DO TempMeasuredSet %value% ENDON"}}
15:28:05.238 MQT: stat/tasmota_FA4C08/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:28:05.439 MQT: stat/tasmota_FA4C08/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:28:35.899 MQT: stat/tasmota_FA4C08/STATUS = {"Status":{"Module":0,"DeviceName":"NSPanel_dormitorio","FriendlyName":["NSPanel_dormitorio_switch1","NSPanel_dormitorio_switch2"],"Topic":"tasmota_FA4C08","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:28:35.915 MQT: stat/tasmota_FA4C08/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T06:17:56","StartupUTC":"2022-12-12T08:10:39","Sleep":0,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2022-12-10T13:36:06","SaveCount":443}}
15:28:35.934 MQT: stat/tasmota_FA4C08/STATUS2 = {"StatusFWR":{"Version":"12.2.0(nspanel)","BuildDateTime":"2022-12-12T08:08:18","Core":"2_0_5","SDK":"v4.4.3.5-17-gd772747b2","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"471/699"}}
15:28:35.947 MQT: stat/tasmota_FA4C08/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vodafoneBA2213",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00800080","00006000","00004000","00000000"]}}
15:28:35.973 MQT: stat/tasmota_FA4C08/STATUS4 = {"StatusMEM":{"ProgramSize":1446,"Free":1856,"Heap":128,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400844D2","20200000","D434482D","00000001"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,39,50,52,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,127"}}
15:28:35.997 MQT: stat/tasmota_FA4C08/STATUS5 = {"StatusNET":{"Hostname":"tasmota-FA4C08-3080","IPAddress":"192.168.0.127","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"C0:49:EF:FA:4C:08","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:28:36.010 MQT: stat/tasmota_FA4C08/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FA4C08","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:28:36.026 MQT: stat/tasmota_FA4C08/STATUS7 = {"StatusTIM":{"UTC":"2022-12-12T14:28:36","Local":"2022-12-12T15:28:36","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"16:52"}}
15:28:36.049 MQT: stat/tasmota_FA4C08/STATUS10 = {"StatusSNS":{"Time":"2022-12-12T15:28:36","ANALOG":{"Temperature1":23.9},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":21.00,"CtrDutyCycleRead":0},"TempUnit":"C"}}
15:28:36.069 MQT: stat/tasmota_FA4C08/STATUS11 = {"StatusSTS":{"Time":"2022-12-12T15:28:36","Uptime":"0T06:17:57","UptimeSec":22677,"Heap":126,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"Berry":{"HeapUsed":15,"Objects":230},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"vodafoneBA2213","BSSId":"F8:AA:3F:62:01:A0","Channel":13,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

hallard, 2023-03-01:
I'm using Thermostat day by day and it has a lot of different parameters, but I must admit that sometimes, on my side it has strange issue also such as powering for 1/2 seconds without understanding why.  

Also you need to check that your device does not reboot because in this case in need back calculation on fresh start.

May be another option would be to switch to PID mod to compare.

barbudor, 2023-03-01:
I suggest that you log to a syslog server continuously with syslog 3 to have details
It may explain why tasmota decide to heat when not needed

closed 2023-03-31 by github-actions[bot]