arendst/Tasmota #18078: Tasmota32-webcam - no I2C

uncarvedblock78, 2023-02-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After upgrade, I2C not initialized, missing SDA/SCL in pin assigment config

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): aithinker esp32-cam
- [x] Tasmota binary firmware version number used: 12.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
stat/esp-cam/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}

stat/esp-cam/RESULT = {"Module":{"2":"ESP32-Cam"}}

stat/esp-cam/RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"608":"I2C SCL"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"640":"I2C SDA"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"0":"None"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I2C initialized, sensors connected

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-02-28:
That's no bug, but by design. Due to camera operation taxing the ESP32 pretty hard, it is not recommended with combined setups including camera. If you want to experiment with this, on your own, you'll have to do your own combined build.

uncarvedblock78, 2023-02-28:
also to note, sending command `I2CScan` results with `{"Command":"Unknown"}`

uncarvedblock78, 2023-02-28:
> That's no bug, but by design. Due to camera operation taxing the ESP32 pretty hard, it is not recommended with combined setups including camera. If you want to experiment with this, on your own, you'll have to do your own combined build.

This is an undocumented change then? It was working as expected in 12.3.1...

barbudor, 2023-02-28:
No change
See the configuration tasmoa32-webcam v12.3.1 : https://github.com/arendst/Tasmota/blob/e8edf976dacf1e636a4119268280c983a595c6c3/tasmota/include/tasmota_configurations_ESP32.h#L195-L232

Line 205 : #undef USE_I2C

Rolling back to 11.0.0 it's the same
I didn't checked older release but I would assume that was the case for long

You may have already built your own with 12.3.1 ?

sfromis, 2023-02-28:
>  It was working as expected in 12.3.1...

Yes, it worked **as expected** in 12.3.1 meaning **No I2C in the webcam build**. Just tested that.

uncarvedblock78, 2023-02-28:
It is possible that I built my own 12.3.1 binary and am not remembering. I'll close the issue, thanks.

closed 2023-02-28 by uncarvedblock78