arendst/Tasmota #18096: http response only

s1nbad, 2023-03-03:
Is there a way of an http request can get an http response only and not invoking an mqtt response ?

eg: http://xxx.xxx.xxx.xxx/cm?cmnd=status0';

I get the correct http rersponse - BUT - also the request generates an mqtt response.  When you are polling 200 devices this creates a lot of broadcast traffic. 

Is there any way around this behaviour ?

barbudor, 2023-03-03:
No. As far as MQTT is enabled, everything is published to MQTT. 

If you are using MQTT just stick to MQTT as this is the way Tasmota is meant to be used and the most efficient. Http polling is not a good way of using Tasmota

I'm using TDM which extensively poll Tasmota devices in MQTT. No problems...

sfromis, 2023-03-03:
The amount of traffic from polling merely 200 devices with short messages is **by far** not a lot. Your http usage generates quite a bit more traffic due to this protocol not being as efficient, due to creating/closing sessions for each. As pointed out by barbudor, better avoid using http for polling.

On Discord, I already answered your question about MQTT messages not being optional . You do not get answers you'd like more by asking the same question in a place which is not even meant for asking questions. On Github, that would be in "Discussions" instead of "Issues".

closed 2023-03-03 by Jason2866