arendst/Tasmota #18103: Esp32 ENERGY_MAX_PHASES and Pzem-004t V3

bob-tm, 2023-03-03:
### PROBLEM DESCRIPTION
I connect one Pzem-004t-v3 to ESP32. R8 where lowered.  

At boot  Web UI show 8 phases and then  drops to 3 phases (???). Correct values where only for L1 and each second L1 resets to all zeros. Then correct values for 1 second and again zeros. L2 and L3 where always zeros or null.  

Then I change ENERGY_MAX_PHASES from 8 to 3 in xdrv_03_esp32_energy.ino and recompile firmware. 
Now only L1 left and everything works as expected.

There is some problems with ESP32 and pzem v3

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Robotdyn Esp32
- [x] Tasmota binary firmware version number used: 12.4.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:58:44.807 MQT: stat/tasmota_9B0F50/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
18:58:45.049 MQT: stat/tasmota_9B0F50/RESULT = {"Module":{"1":"ESP32-DevKit"}}
18:58:45.261 MQT: stat/tasmota_9B0F50/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1472":"PZEM0XX Tx"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"1536":"PZEM016 Rx"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"224":"Relay1"},"GPIO26":{"225":"Relay2"},"GPIO27":{"1152":"RFrecv"},"GPIO32":{"227":"Relay4"},"GPIO33":{"226":"Relay3"},"GPIO34":{"32":"Button1"},"GPIO35":{"33":"Button2"},"GPIO36":{"34":"Button3"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"35":"Button4"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

18:59:32.254 CMD: Status 0
18:59:32.263 MQT: stat/tasmota_9B0F50/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmota_9B0F50","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:59:32.281 MQT: stat/tasmota_9B0F50/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:10:49","StartupUTC":"2023-03-03T17:48:43","Sleep":50,"CfgHolder":4617,"BootCount":132,"BCResetTime":"2022-04-12T14:27:52","SaveCount":969}}
18:59:32.294 MQT: stat/tasmota_9B0F50/STATUS2 = {"StatusFWR":{"Version":"12.4.0(tasmota)","BuildDateTime":"2023-03-03T17:46:51","Core":"2_0_6","SDK":"4.4.3.20230129","CpuFrequency":160,"Hardware":"ESP32-D0WD rev.1","CR":"449/699"}}
18:59:32.308 MQT: stat/tasmota_9B0F50/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["/dev/null",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["0000A009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:59:32.331 MQT: stat/tasmota_9B0F50/STATUS4 = {"StatusMEM":{"ProgramSize":1492,"Free":1856,"Heap":149,"StackLowMark":2,"PsrMax":4093,"PsrFree":4093,"ProgramFlashSize":4096,"FlashSize":16384,"FlashChipId":"184020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","9F9AD7CF","0015E001","B7F7BFCF","05DA9BC1","E0360FC7","480840D2","20200000","D43C482D","000080B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,15,16,17,20,21,24,26,27,28,29,34,35,38,50,52,59,60,62,63,66,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,29,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,127"}}
18:59:32.355 MQT: stat/tasmota_9B0F50/STATUS5 = {"StatusNET":{"Hostname":"Controller-BlackHole","IPAddress":"192.168.100.236","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.100.1","DNSServer2":"0.0.0.0","Mac":"34:AB:95:9B:0F:50","IP6Global":"","IP6Local":"fe80::36ab:95ff:fe9b:f50","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.100.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:59:32.375 MQT: stat/tasmota_9B0F50/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9B0F50","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:59:32.393 MQT: stat/tasmota_9B0F50/STATUS7 = {"StatusTIM":{"UTC":"2023-03-03T17:59:32","Local":"2023-03-03T18:59:32","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:28","Sunset":"18:35"}}
18:59:32.413 MQT: stat/tasmota_9B0F50/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:59:32.438 MQT: stat/tasmota_9B0F50/STATUS10 = {"StatusSNS":{"Time":"2023-03-03T18:59:32","ENERGY":{"TotalStartTime":"2022-04-12T14:27:52","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":50,"Voltage":224,"Current":0.000},"INA219":{"Id":40,"Voltage":1,"Current":0.000,"Power":0},"PCF8574-1":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1}}}
18:59:32.462 MQT: stat/tasmota_9B0F50/STATUS11 = {"StatusSTS":{"Time":"2023-03-03T18:59:32","Uptime":"0T00:10:49","UptimeSec":649,"Heap":150,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":44},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"/dev/null","BSSId":"52:FF:20:8E:B3:96","Channel":1,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2023-03-03:
Try to change PZEM_STABILIZE in file xnrg_03_pzem004t.ino from 30 to 80.

bob-tm, 2023-03-04:
Build latest 12.4.0.2. Everything is fine. Tnx!

closed 2023-03-05 by arendst