arendst/Tasmota #18108: MQTT Subscribe to multiple topics event performs twice

13, 2023-03-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 DevKit v1
- [x] Tasmota binary firmware version number used: 12.4.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:31:38.075 MQT: tasmota/stat/tasmota_74EDAC/RESULT = {"NAME":"ESP32-DevKit-MUH","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,160,161,162,32,0,0,0,228,0,224,225,226,0,0,0,0,227,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
17:31:38.315 MQT: tasmota/stat/tasmota_74EDAC/RESULT = {"Module":{"0":"ESP32-DevKit-MUH"}}
17:31:38.572 MQT: tasmota/stat/tasmota_74EDAC/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"160":"Switch1"},"GPIO17":{"161":"Switch2"},"GPIO18":{"162":"Switch3"},"GPIO19":{"32":"Button1"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"228":"Relay5"},"GPIO24":{"0":"None"},"GPIO25":{"224":"Relay1"},"GPIO26":{"225":"Relay2"},"GPIO27":{"226":"Relay3"},"GPIO32":{"227":"Relay4"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
17:32:02.358 MQT: tasmota/stat/tasmota_74EDAC/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":334,"Free":177,"Rules":"on Switch1#state do Publish2 tasmota/sensors/HD/state %value% endon on Switch2#state do Publish2 tasmota/sensors/HDL/state %value% endon on Button1#state do Publish2 tasmota/sensors/HDB/state %value% endon on Switch3#state=1 do publish tasmota/sensors/HDP/state 1 endon on Switch3#state=0 do Publish2 tasmota/sensors/HDP/state 0 endon"}}
17:32:02.584 MQT: tasmota/stat/tasmota_74EDAC/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":423,"Free":88,"Rules":"ON mqtt#connected DO Subscribe LEDG, tasmota/sensors/G ENDON ON mqtt#connected DO Subscribe LEDGDL, tasmota/sensors/GDL ENDON ON Event#LEDG DO Backlog var1 %value%; IF ((var1==1) AND (var2==1)) Power5 1  ELSEIF ((var1==0) AND (var2==0)) Power5 0  ELSE Power5 3  ENDIF ENDON ON Event#LEDGDL DO Backlog var2 %value%; IF ((var1==1) AND (var2==1)) Power5 1  ELSEIF ((var1==0) AND (var2==0)) Power5 0  ELSE Power5 3  ENDIF ENDON"}}
17:32:02.809 MQT: tasmota/stat/tasmota_74EDAC/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:32:19.413 MQT: tasmota/stat/tasmota_74EDAC/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4",""],"Topic":"tasmota_74EDAC","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[2,2,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:32:19.487 MQT: tasmota/stat/tasmota_74EDAC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:54","StartupUTC":"2023-03-04T16:31:25","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2023-03-03T13:56:34","SaveCount":338}}
17:32:19.539 MQT: tasmota/stat/tasmota_74EDAC/STATUS2 = {"StatusFWR":{"Version":"12.4.0.2(tasmota32)","BuildDateTime":"2023-03-03T14:13:28","Core":"2_0_7","SDK":"4.4.4.20230218","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"434/699"}}
17:32:19.578 MQT: tasmota/stat/tasmota_74EDAC/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["muhxnetwork",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C80001000600003C5A0A192800000000","00800080","00006000","00004001","00000000"]}}
17:32:19.628 MQT: tasmota/stat/tasmota_74EDAC/STATUS4 = {"StatusMEM":{"ProgramSize":1480,"Free":1399,"Heap":156,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","9F8AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","480840D2","20200000","D4BC482D","000280B1"],"Drivers":"1,2,3,4,5,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,82,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77"}}
17:32:19.718 MQT: tasmota/stat/tasmota_74EDAC/STATUS5 = {"StatusNET":{"Hostname":"tasmota-74EDAC-3500","IPAddress":"192.168.22.195","Gateway":"192.168.22.6","Subnetmask":"255.255.255.0","DNSServer1":"192.168.22.6","DNSServer2":"0.0.0.0","Mac":"30:AE:A4:74:ED:AC","IP6Global":"","IP6Local":"fe80::32ae:a4ff:fe74:edac","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.22.6","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:32:19.783 MQT: tasmota/stat/tasmota_74EDAC/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.22.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_74EDAC","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:32:19.823 MQT: tasmota/stat/tasmota_74EDAC/STATUS7 = {"StatusTIM":{"UTC":"2023-03-04T16:32:19","Local":"2023-03-04T17:32:19","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:26","Sunset":"18:37"}}
17:32:19.864 MQT: tasmota/stat/tasmota_74EDAC/STATUS10 = {"StatusSNS":{"Time":"2023-03-04T17:32:19","Switch1":"OFF","Switch2":"OFF","Switch3":"ON"}}
17:32:19.894 MQT: tasmota/stat/tasmota_74EDAC/STATUS11 = {"StatusSTS":{"Time":"2023-03-04T17:32:19","Uptime":"0T00:00:54","UptimeSec":54,"Heap":154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"muhxnetwork","BSSId":"1C:61:B4:4F:7D:3E","Channel":11,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
17:33:13.494 RUL: EVENT#LEDG performs "Backlog var1 1; IF ((var1==1) AND (var2==1)) Power5 1  ELSEIF ((var1==0) AND (var2==0)) Power5 0  ELSE Power5 3  ENDIF"
17:33:13.505 SRC: Rule
17:33:13.508 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 110, Pld -99, Data 'var1 1; IF ((var1==1) AND (var2==1)) Power5 1  ELSEIF ((var1==0) AND (var2==0)) Power5 0  ELSE Power5 3  ENDIF'
17:33:13.524 SRC: Backlog
17:33:13.526 CMD: Grp 0, Cmd 'VAR', Idx 1, Len 1, Pld 1, Data '1'
17:33:13.535 MQT: tasmota/stat/tasmota_74EDAC/RESULT = {"Var1":"1"}
17:33:13.822 SRC: Backlog
17:33:13.825 CMD: Grp 0, Cmd 'IF', Idx 1, Len 99, Pld -99, Data '((var1==1) AND (var2==1)) Power5 1  ELSEIF ((var1==0) AND (var2==0)) Power5 0  ELSE Power5 3  ENDIF'
17:33:13.833 MQT: tasmota/stat/tasmota_74EDAC/RESULT = {"If":"Done"}
17:33:14.061 SRC: Backlog
[... cut ...]
ON mqtt#connected DO Subscribe LEDG, tasmota/sensors/G ENDON
ON mqtt#connected DO Subscribe LEDGDL, tasmota/sensors/GDL ENDON
ON Event#LEDG DO Backlog var1 %value%;
IF ((var1==1) AND (var2==1)) Power5 1 
ELSEIF ((var1==0) AND (var2==0)) Power5 0 
ELSE Power5 3 
ENDIF
ENDON
ON Event#LEDGDL DO Backlog var2 %value%;
IF ((var1==1) AND (var2==1)) Power5 1 
ELSEIF ((var1==0) AND (var2==0)) Power5 0 
ELSE Power5 3 
ENDIF
ENDON

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-03-04:
IIRC, the old subscribe feature is "a bit simplistic" in doing the topic matching as "string starts with".

When you are using ESP32, I'd much recommend using the Berry language instead of classic rules, and doing a small test it also appears that the MQTT subscribe feature in Berry correctly distinguishes between similar topics of that type.

barbudor, 2023-03-04:
Problem is identified
The matching is done with a `startsWith()`
This was intended to trigger with publication on subscribed topic as well as subtopic
But obviously the match goes a bit further than expected

I'm currently testing a fix

sfromis, 2023-03-04:
Beware that fixing the code to do a "more correct" topic matching may be a breaking change to users being deliberately "less exact" about using the full expected incoming topic, reasons for that could be a topic structure where they want to ignore variable parts of one topic level.

barbudor, 2023-03-04:
The documentation and the result of the command itself clearly state that
`Subscribe LEDG, tasmota/sensors/G` subscribes to `tasmota/sensors/G/#` and
`Subscribe LEDGDL, tasmota/sensors/GDL` subscribe `tasmota/sensors/GDL/#`
which means the topic and any-subtopic, not any topic starting with
Publishing on `tamota/sensors/GDL/state` should definitively not match `tasmota/sensors/G/#`

Anyone basing it's operation on a bug is likely to face a problem when the bug is fixed

Feel free to comment on https://github.com/arendst/Tasmota/pull/18114

PR #18114 (2023-03-05): Fix subscribe topic match comparison
## Description:

**Related issue (if applicable):** fixes #18108

Currently topic matching is made using:
```
if (sTopic.startsWith(event_item.Topic))
```
where `sTopic` is the topic of the currently received message and `event_item.Topic` is the subscribed topic
With such lazy comparison, a publish on `something/ABC/state` would trigger both the events subscribed on `something/A/#` and `something/ABC/#` which is not the intended result.

The fix consist in matching for either strict equality OR starting with "subscribed topic + `/`"

## Checklist:
  - [X]  [... cut]
closed 2023-03-05 by arendst