arendst/Tasmota #18129: DS3231 internal Temperature sensor data only updated when teleperiod

pyys, 2023-03-07:
### PROBLEM DESCRIPTION
saving DS3231's internal Temperature sensor data to script variable got error

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [*] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [*] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [*] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [*] Device used (e.g., Sonoff Basic): [esp32 relay 2ch module] (https://www.aliexpress.com/item/1005004466280920.html?spm=a2g0o.cart.0.0.53ef38daiELwzf&mp=1)
- [*] Tasmota binary firmware version number used: 12.4.0_____
  - [ ] Pre-compiled
  - [*] Self-compiled
- [*] Flashing tools used: Esp-flasher (based on PyFlasher)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:32:22.016 CMD: Backlog Template; Module; GPIO 255
18:32:22.018 SRC: WebConsole from 192.167.0.2
18:32:22.020 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
18:32:22.041 SRC: Backlog
18:32:22.043 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
18:32:22.053 MQT: stat/CatalonixTimer_2ch_4153A4/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
18:32:22.264 SRC: Backlog
18:32:22.266 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
18:32:22.275 MQT: stat/CatalonixTimer_2ch_4153A4/RESULT = {"Module":{"0":"ESP32-DevKit"}}
18:32:22.513 SRC: Backlog
18:32:22.515 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
18:32:22.529 MQT: stat/CatalonixTimer_2ch_4153A4/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"640":"I2C SDA1"},"GPIO14":{"608":"I2C SCL1"},"GPIO15":{"0":"None"},"GPIO16":{"225":"Relay2"},"GPIO17":{"224":"Relay1"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"288":"Led1"},"GPIO24":{"0":"None"},"GPIO25":{"704":"SPI MOSI1"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"736":"SPI CLK1"},"GPIO33":{"6720":"SDCard CS"},"GPIO34":{"4704":"ADC Input1"},"GPIO35":{"672":"SPI MISO1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
-not using

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:33:43.123 CMD: status 0
18:33:43.125 SRC: WebConsole from 192.167.0.2
18:33:43.127 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
18:33:43.133 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS = {"Status":{"Module":0,"DeviceName":"CatalonixTimer00","FriendlyName":["CatalonixTimer00","CatalonixTimer002"],"Topic":"CatalonixTimer_2ch_4153A4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:33:43.140 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T01:57:31","StartupUTC":"2023-03-07T07:36:12","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2023-03-07T16:33:39","SaveCount":30}}
18:33:43.145 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS2 = {"StatusFWR":{"Version":"12.4.0(TasmoCompiler-esp32generic)","BuildDateTime":"2023-03-02T07:25:30","Core":"2_0_6","SDK":"4.4.3.20230129","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"488/699"}}
18:33:43.150 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CTT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:33:43.161 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS4 = {"StatusMEM":{"ProgramSize":1269,"Free":1856,"Heap":119,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","07C045CE","000C0001","00000003","00000400","C0000881","00000080","00200000","54000008","00000080"],"Drivers":"1,2,4,8,9,10,12,24,26,29,35,38,50,52,56,62","Sensors":"1,2,127"}}
18:33:43.171 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS5 = {"StatusNET":{"Hostname":"CatalonixTimer-2ch-4153A4-5028","IPAddress":"192.167.0.22","Gateway":"192.167.0.1","Subnetmask":"255.255.255.0","DNSServer1":"168.126.63.1","DNSServer2":"168.126.63.2","Mac":"34:94:54:41:53:A4","IP6Global":"","IP6Local":"fe80::3694:54ff:fe41:53a4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:33:43.177 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS6 = {"StatusMQT":{"MqttHost":"ha.catalonix.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4153A4","MqttUser":"catalonix","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:33:43.183 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS7 = {"StatusTIM":{"UTC":"2023-03-07T09:33:43","Local":"2023-03-07T18:33:43","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+09:00","Sunrise":"15:20","Sunset":"02:41"}}
18:33:43.226 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS10 = {"StatusSNS":{"Time":"2023-03-07T18:33:43","ANALOG":{"A1":1127},"DS3231":{"Temperature":26.2},"TempUnit":"C"}}
18:33:43.234 MQT: stat/CatalonixTimer_2ch_4153A4/STATUS11 = {"StatusSTS":{"Time":"2023-03-07T18:33:43","Uptime":"0T01:57:31","UptimeSec":7051,"Heap":121,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":33},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"CTT","BSSId":"58:86:94:5D:BC:4A","Channel":7,"Mode":"11n","RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

compiling parameter

#define PROJECT                "***"
#define DEVICE_NAME          "***"
[... 71 lines cut ...]
I hope DS3231's intenal sensor data update rate is independant with teleperiod.

p.s
when if I use script like this
=========================================
>D 
volt=0
>T
volt=ANALOG#A1
>S
print voltage=%volt%
=========================================

it update 'volt' variable every second.

what is differce with "volt=ANALOG#A1" and "temp=DS3231#Temperature" ?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-03-07:
I can add that fast updates are not happening for rules either. After a cursory look, I noticed that the code in [support_tasmota.ino](https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_support/support_tasmota.ino#L1011-L1024) only seems to call `XdrvRulesProcess` for TelePeriod processing, but not for the other handling there.

barbudor, 2023-03-07:
Script `>T` section is supposed to be executed not only on FUNC_TELEPERIOD_RULES_PROCESS but also on FUNC_EVERY_100_MSECOND

However there are tricks in the 100ms loop
1) Only 1 sensor is checked at each 100ms. There's a round robin through every enabled sensor which could reduced the polling rate of a given sensor
2) The DS3231 driver is not a sensor driver, but a RTC driver. Which means it is not included into that 100ms polling and only in the Teleperiod polling

So that's the main difference between ANALOG (which is a 'sensor' driver) while DS3231 is a standard 'driver'

My main question would be : why do you want to have a temperature measurments faster than teleperiod ?
Temperature sensors are generally slow

One possibility would be a dual-implementation on the DS3231 driver : both a RTC driver and a sensor

pyys, 2023-03-07:
> Script `>T` section is supposed to be executed not only on FUNC_TELEPERIOD_RULES_PROCESS but also on FUNC_EVERY_100_MSECOND
>
> However there are tricks in the 100ms loop
>
>     1. Only 1 sensor is [...]

For my project, teleperiod would be 3600second. (or disabled. we are not sure to user MQTT)
but I need check volt(by analog in, gpio34) and temperature(by ds3231. my project dosen't sesitive accuracy, but need to reduce pin using. we are aready using 9pin, and need 6 more, or more), and relay stat every 300second and write it to sdcard.(for making watchdog log file)

It's okay to me that teleperiod is longer than 60 second.
but I need update volt and temp data when triggered situation.
thats why I hope I could update DS3231's temp data by manual trigger. not by teleperiod.

p.s
I've already tried reduce teleperiod to 30 second, for update DS3231's Temp data. and it satisfy bottom point of my project.
but I hope I could touch better solution.

sfromis, 2023-03-07:
To get "when triggered", issuing the `Status 10` command will publish extra telemetry data, instead of waiting for the next scheduled telemetry period.

pyys, 2023-03-07:
> To get "when triggered", issuing the `Status 10` command will publish extra telemetry data, instead of waiting for the next scheduled telemetry period.
> 
> Or get a "now" telemetry message by issuing the `TelePeiod` command without an argument. Allows rules to be triggered.

thank you for reply. 
I'll try and look that I could use in my code. at tomorrow.
(I'm in GMT+9 , here are AM 01:50 :) )

barbudor, 2023-03-07:
Without using more pins as you already have I2c, you could use a Hdc1080 (temp+Hum) sensor or TMP75 which are real 'sensor' drivers and won't suffer from the same problem.

Jason2866, 2023-03-07:
Moving to discussions, since not an issue/bug.