arendst/Tasmota #18146: SML: '%' as unit of measurement causes memory corruption

TheChatty, 2023-03-09:
### PROBLEM DESCRIPTION
When defining meter metrics with UoM = '%' it causes memory corruption in V12.4.0.2.
This problem did not occur in V12.2.0.5.

The corruption occurs in the displayed values in WebUI. A few metrics defined below the '%' metric show garbage instead of label + value.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP12F dev board
- [x] Tasmota binary firmware version number used: V12.4.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

working:
```
...
2,F70304UUuu@i2:1,StellsignalRk1,prozent,CtrlSig_RK1,0
2,=h<hr/>
...
```

no longer working:
```
...
2,F70304UUuu@i2:1,StellsignalRk1,%,CtrlSig_RK1,0
2,=h<hr/>
...
```

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

TheChatty, 2023-03-09:
@gemu2015 This issue is probably to be addressed by you.

gemu2015, 2023-03-10:
#define SML_REPLACE_VARS is now default which means you must place `%%` to get the `%` char

Jason2866, 2023-03-10:
Closing since answered

closed 2023-03-10 by Jason2866