arendst/Tasmota #18163: Hardware watchdog reset and wifi problems

boaschti, 2023-03-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If I enable AS608 tasmota resets after about one day with Hardware Watchdog or it is disconnected from wifi. In tried a few versions and its alway th same problem. Without AS608 it runs stable.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [ x] Tasmota binary firmware version number used: Project tasmota - Tasmota Version 12.4.0.2(allsensors)-2_7_4_9(2023-03-06T16:21:42)
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:22:25.287 CMD: Backlog Template; Module; GPIO 255
14:22:25.353 MQT: stat/tuerpanel/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:22:25.590 MQT: stat/tuerpanel/RESULT = {"Module":{"18":"Generic"}}
14:22:25.847 MQT: stat/tuerpanel/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"5504":"AS608 Tx"},"GPIO2":{"224":"Relay1"},"GPIO3":{"5536":"AS608 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"194":"Switch_n3"},"GPIO13":{"161":"Switch2"},"GPIO14":{"160":"Switch1"},"GPIO15":{"225":"Relay2"},"GPIO16":{"163":"Switch4"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:23:09.270 CMD: Backlog Rule1; Rule2; Rule3
14:23:09.338 MQT: stat/tuerpanel/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":58,"Free":453,"Rules":"ON switch1#state DO publish cmnd/ring2/power %value% ENDON"}}
14:23:09.579 MQT: stat/tuerpanel/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:23:09.829 MQT: stat/tuerpanel/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:23:43.829 CMD: Status 0
14:23:43.838 MQT: stat/tuerpanel/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["TuerPanel","Tasmota2"],"Topic":"tuerpanel","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,15,15,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:23:43.876 MQT: stat/tuerpanel/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Power On","Uptime":"0T00:07:12","StartupUTC":"2023-03-11T13:16:31","Sleep":50,"CfgHolder":4617,"BootCount":79,"BCResetTime":"2020-10-17T20:06:41","SaveCount":5902,"SaveAddress":"F4000"}}
14:23:43.908 MQT: stat/tuerpanel/STATUS2 = {"StatusFWR":{"Version":"12.4.0.2(allsensors)","BuildDateTime":"2023-03-06T16:21:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"403/699"}}
14:23:43.930 MQT: stat/tuerpanel/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHomeZeig","Servus nachad!"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
14:23:43.978 MQT: stat/tuerpanel/STATUS4 = {"StatusMEM":{"ProgramSize":742,"Free":260,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F908787","04146001","FFFFB7CF","8F7417FB","303E0AD3","1159785A","26204FFF","5408F0F3","00000080"],"Drivers":"1,2,3,4,7,8,9,10,12,14,15,20,21,24,28,29,34,43,47,50,60,62,66","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,24,26,27,29,30,31,32,38,40,41,42,44,45,46,50,51,54,55,56,57,58,59,63,64,65,66,67,70,71,72,73,74,75,76,78,79,81,82,85,86,88,89,90,91,92,93","I2CDriver":"1,2,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,28,29,30,31,32,34,35,36,40,41,42,43,44,45,46,48,49,50,51,52,53,55,56,58,60,61,62,63"}}
14:23:44.002 MQT: stat/tuerpanel/STATUS5 = {"StatusNET":{"Hostname":"tuerpanel","IPAddress":"192.168.178.30","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"50:02:91:FE:55:40","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:23:44.024 MQT: stat/tuerpanel/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.38","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FE5540","MqttUser":"mqttClients","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:23:44.049 MQT: stat/tuerpanel/STATUS7 = {"StatusTIM":{"UTC":"2023-03-11T13:23:44","Local":"2023-03-11T14:23:44","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:12","Sunset":"18:47"}}
14:23:44.072 MQT: stat/tuerpanel/STATUS10 = {"StatusSNS":{"Time":"2023-03-11T14:23:44","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF"}}
14:23:44.085 MQT: stat/tuerpanel/STATUS11 = {"StatusSTS":{"Time":"2023-03-11T14:23:44","Uptime":"0T00:07:13","UptimeSec":433,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":177,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SmartHomeZeig","BSSId":"00:E0:4C:13:96:F9","Channel":11,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
enable AS608 with connected sensor and wait.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Pipitapi, 2023-03-15:
If such a programme were included. It would be nice. I have no contact every 2-3 days! 
https://www.reddit.com/r/tasmota/comments/nvdedp/comment/h14nxac/
https://www.reddit.com/r/tasmota/comments/imgakh/possible_resetting_the_routers_power_if_it_doesnt/

MaceMoneta, 2023-03-23:
I just use a simple rule:

`on Time#Minute|5 do ping8 routerIP endon on Ping#routerIP#Success==0 do restart 1 endon`

Replace the two occurrences of 'routerIP' with your router's IP address. Every five minutes it will ping the router. If it gets no response, it will restart the module. I have this running on over 100 assorted modules.

pawol, 2023-04-07:
the command ping8 is not available in regular build

pawol, 2023-04-07:
https://gregology.net/2020/12/router-rebooter/
There you can download and try tasmota 9.1 compiled with ping8. Works lik a charm as watchdog :)

davlaw, 2023-04-07:
ping8 set me back a sec too, but is tasmota command ping or ping#, where # equals # of packets to send...

barbudor, 2023-04-07:
The command is `Ping` but like many commands can take an index such as with Power1
In the case of Ping, the index is used to specify the number of ping packets to send. If not specified it is equivalent to Ping1 (just like Power == Power1)

arendst, 2023-05-11:
Is this still an issue?

pawol, 2023-05-11:
@arendst
Is it possible to add USE_PING, USE_EXPRESSION, & SUPPORT_IF_STATEMENT  in the precompiled builds ?

boaschti, 2023-05-11:
Yes. I already tried a better LDO for supplying the sensor and ESP.
I can try it with a new ESP. But i have to unmount it from my door for this test, if i find any time for this.
Can i do some additional test to find out what is going on here?

boaschti, 2023-05-11:
Ping and automatically reset is no solution for me. Sometimes if ESP got stuck my door bell on same device also rings. (But not on reeboot!!!) But here I have to double check hardware, if the pulldown and everything else is working.

Noschvie, 2023-05-12:
@boaschti have you tried an ESP32 with the AS608 sensor? For me it seems to be stable with latest version.

boaschti, 2023-05-14:
No. I don't tried it yet.

pcnetzwerke, 2023-05-19:
Hi, I have a similar behaviour. But I use a self compiled 12.4 Version (sensors) with AS608 enabled on a D1 Mini for about a week now.

I do not know if this information helps - but I also need to reset the sensor one time per day, same behaviour as described above.

boaschti, 2023-05-19:
Is there any polling uart sensor? So we can connect this other sensor to verify if the tasmota uart has a problem or the code of AS608?

pcnetzwerke, 2023-05-19:
There are so many different serial sensors (https://tasmota.github.io/docs/Supported-Peripherals), I would be very surprised if this was a genral uart problem in tasmota.

pcnetzwerke, 2023-05-21:
So I checked my syslog server and found out that Heap is permanently decreasing beginning with 22, i.e:

```
messages_2023_05_19:<13>1 2023-05-19T13:29:55.645788+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/STATE = {"Time":"2023-05-19T12:29:55","Uptime":"0T00:00:09","UptimeSec":9,"Heap":22,...
[...]
messages_2023_05_19:<13>1 2023-05-19T18:19:57.414813+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/STATE = {"Time":"2023-05-19T17:19:57","Uptime":"0T04:50:11","UptimeSec":17411,"Heap":6,...
```

I see in the logs that half an hour later I hard restarted my D1 Mini.

I do not know nuch about programming, but could it be the AS608 code does not fully clear the heap?

sfromis, 2023-05-21:
Heap 12 is very low, but mostly sustainable. Lower than that is likely to cause failures. There is indeed a memory leak somewhere, needing to be fixed.

arendst, 2023-05-21:
> but could it be the AS608 code does not fully clear the heap?

To blame the AS608 driver I need more information like the output of command ``status 0``. The OP uses an ``allsensors`` binary which is a non-released binary and is likely using a lot of heap while doing nothing based on his heap of 14 in the initial post.

sfromis, 2023-05-21:
The lastest post (instead of the OP) does show heap sinking from a pretty normal 22 to just 6 in a bit under 6 hours. Smells a lot like a memory leak somewhere. Hard to tell what driver it is, of course, and may be entirely unrelated to the OP.

pcnetzwerke, 2023-05-22:
I am nor sure if I am expected/allowed to address my problem in this thread. But maybe I can help and/or can be helped. Anyway, here is my status 0:

```
07:47:27.343 CMD: status 0
07:47:27.351 MQT: stat/tasmota028/STATUS = {"Status":{"Module":18,"DeviceName":"tasmota028","FriendlyName":["tasmota028"],"Topic":"tasmota028","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
07:47:27.385 MQT: stat/tasmota028/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:20","StartupUTC":"2023-05-22T06:45:07","Sleep":50,"CfgHolder":4617,"BootCount":54,"BCResetTime":"2023-05-13T14:32:14","SaveCount":116,"SaveAddress":"F8000"}}
07:47:27.421 MQT: stat/tasmota028/STATUS2 = {"StatusFWR":{"Version":"12.4.0(sensors)","BuildDateTime":"2023.05.06 21:17:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"433/699"}}
07:47:27.441 MQT: stat/tasmota028/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":2,"LogHost":"192.168.5.251","LogPort":514,"SSId":["mitohnekabel2.4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["04008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
07:47:27.482 MQT: stat/tasmota028/STATUS4 = {"StatusMEM":{"ProgramSize":703,"Free":300,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"166085","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000800","54000020","0000C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,79,98,103"}}
07:47:27.525 MQT: stat/tasmota028/STATUS5 = {"StatusNET":{"Hostname":"tasmota028","IPAddress":"192.168.5.213","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.5.3","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:67:16:D0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
07:47:27.549 MQT: stat/tasmota028/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.3","MqttPort":1883,"MqttClientMask":"tasmota028","MqttClient":"tasmota028","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:47:27.575 MQT: stat/tasmota028/STATUS7 = {"StatusTIM":{"UTC":"2023-05-22T06:47:27","Local":"2023-05-22T07:47:27","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:00","Sunset":"20:33"}}
07:47:27.596 MQT: stat/tasmota028/STATUS10 = {"StatusSNS":{"Time":"2023-05-22T07:47:27"}}
07:47:27.605 MQT: stat/tasmota028/STATUS11 = {"StatusSTS":{"Time":"2023-05-22T07:47:27","Uptime":"0T00:02:20","UptimeSec":140,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"MqttCount":1,"POWER1":"off","Wifi":{"AP":1,"SSId":"mitohnekabel2.4","BSSId":"C0:56:27:7F:EA:77","Channel":6,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

At the moment I restart my D1 Mini every hour to prevent necessity of hard resets. If You need a status 0 under other circumstances/conditions, please tell me.

arendst, 2023-05-22:
To fix possible Hardware Watchdogs try to connect the finger print sensor to GPIO13 (Rx) /15 (Tx) which is using hardware serial.

pcnetzwerke, 2023-05-22:
Thanks for Your help! I changed the wiring and will report.

pcnetzwerke, 2023-05-22:
I am a little confused now.

I wired GPIO13 and GPIO15 and found out, tasmota will not start. So I started over and flashed self-compiled 12.5.0, but without any changes.

I found out that it is possible to attach RX and TX after boot and then soft-restart tasmota. (Is there any other/better GPIOs for RX and TX?).

I defined rule1 because I wanted to find out whether the problem might only affect the d1's network. (maybe the rule continues to steer relay1).

```
22:26:41.083 CMD: status 0
22:26:41.093 MQT: stat/tasmota028/STATUS = {"Status":{"Module":18,"DeviceName":"tasmota028","FriendlyName":["tasmota028"],"Topic":"tasmota028","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:26:41.136 MQT: stat/tasmota028/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:27","StartupUTC":"2023-05-22T20:24:14","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2023-05-22T20:09:55","SaveCount":65,"SaveAddress":"F7000"}}
22:26:41.177 MQT: stat/tasmota028/STATUS2 = {"StatusFWR":{"Version":"12.5.0(sensors)","BuildDateTime":"2023.05.13 13:56:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"432/699"}}
22:26:41.203 MQT: stat/tasmota028/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":2,"LogHost":"192.168.5.251","LogPort":514,"SSId":["mitohnekabel2.4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["04008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
22:26:41.250 MQT: stat/tasmota028/STATUS4 = {"StatusMEM":{"ProgramSize":703,"Free":300,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"166085","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000800","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,79,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
22:26:41.310 MQT: stat/tasmota028/STATUS5 = {"StatusNET":{"Hostname":"tasmota028","IPAddress":"192.168.5.213","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.5.3","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:67:16:D0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:26:41.339 MQT: stat/tasmota028/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6716D0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:26:41.367 MQT: stat/tasmota028/STATUS7 = {"StatusTIM":{"UTC":"2023-05-22T20:26:41","Local":"2023-05-22T22:26:41","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:00","Sunset":"21:33"}}
22:26:41.392 MQT: stat/tasmota028/STATUS10 = {"StatusSNS":{"Time":"2023-05-22T22:26:41"}}
22:26:41.408 MQT: stat/tasmota028/STATUS11 = {"StatusSTS":{"Time":"2023-05-22T22:26:41","Uptime":"0T00:02:27","UptimeSec":147,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER1":"off","Wifi":{"AP":1,"SSId":"mitohnekabel2.4","BSSId":"C0:56:27:7F:EA:77","Channel":6,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

pcnetzwerke, 2023-05-22:
I tried 4 times now with the same result:

1. after about 5 min the d1 disappeares from wifi
2. the fingerprint sensor/rule/relay is still working

arendst, 2023-05-23:
Strike my idea of using 13/15 for serial. I noticed the same result and @sfromis explains why.

I suggest you start using latest dev release as it uses the latest AdafruitFingerprint which might solve some issues.

pcnetzwerke, 2023-05-23:
i striked and wired 12/14 again.

checked out latest development branch (12.5.0.3).

looks pretty. I now have bling-bling and a sensor status in device's tasmota dashboard with last id and confidence.

```
10:44:12.026 CMD: status 0
10:44:12.035 MQT: stat/tasmota028/STATUS = {"Status":{"Module":18,"DeviceName":"tasmota028","FriendlyName":["tasmota028"],"Topic":"tasmota028","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:44:12.077 MQT: stat/tasmota028/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:55","StartupUTC":"2023-05-23T08:37:17","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2023-05-22T20:09:55","SaveCount":87,"SaveAddress":"F4000"}}
10:44:12.117 MQT: stat/tasmota028/STATUS2 = {"StatusFWR":{"Version":"12.5.0.3(sensors)","BuildDateTime":"2023.05.23 07:25:53","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"433/699"}}
10:44:12.146 MQT: stat/tasmota028/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":2,"LogHost":"192.168.5.251","LogPort":514,"SSId":["mitohnekabel2.4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["04008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
10:44:12.193 MQT: stat/tasmota028/STATUS4 = {"StatusMEM":{"ProgramSize":704,"Free":296,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"166085","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000800","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66,68","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,79,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
10:44:12.256 MQT: stat/tasmota028/STATUS5 = {"StatusNET":{"Hostname":"tasmota028","IPAddress":"192.168.5.213","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.5.3","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:67:16:D0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:44:12.285 MQT: stat/tasmota028/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.3","MqttPort":1883,"MqttClientMask":"tasmota028","MqttClient":"tasmota028","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:44:12.313 MQT: stat/tasmota028/STATUS7 = {"StatusTIM":{"UTC":"2023-05-23T08:44:12","Local":"2023-05-23T10:44:12","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"05:59","Sunset":"21:34"}}
10:44:12.340 MQT: stat/tasmota028/STATUS10 = {"StatusSNS":{"Time":"2023-05-23T10:44:12"}}
10:44:12.357 MQT: stat/tasmota028/STATUS11 = {"StatusSTS":{"Time":"2023-05-23T10:44:12","Uptime":"0T00:06:55","UptimeSec":415,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"POWER1":"off","Wifi":{"AP":1,"SSId":"mitohnekabel2.4","BSSId":"C0:56:27:7F:EA:77","Channel":6,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

I will let it run for some days and report.

Thanks a lot so far for Your help!

pcnetzwerke, 2023-05-23:
After about 45 minutes the device disappeared from network. The rule and relay are still working.
I can see in syslog that Heap did NOT decrease at all.

arendst, 2023-05-23:
for further debugging you must setup a syslog server and use command ``syslog 3``. Alternative use seriallogging.

we need log info to get a clue what is going on.

BTW my wemos still works fine since last restart yesterday.

pcnetzwerke, 2023-05-23:
My Heap is 19 at the moment.

It is still a sensors binary with AS608 enabled and a relay on gpio5
```

14:23:24.776 MQT: tele/tasmota028/STATE = {"Time":"2023-05-23T14:23:24","Uptime":"0T02:05:18","UptimeSec":7518,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"off","Wifi":{"AP":1,"SSId":"mitohnekabel2.4","BSSId":"C0:56:27:7F:EA:77","Channel":6,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:24:05.670 CMD: gpio
14:24:05.685 MQT: stat/tasmota028/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"224":"Relay1"},"GPIO12":{"5536":"AS608 Rx"},"GPIO13":{"0":"None"},"GPIO14":{"5504":"AS608 Tx"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

arendst, 2023-05-23:
How do you power the relay and ESP8266 device? What ESP8266 hardware do you use?

Not enough power is a known source of wifi issues.

sfromis, 2023-05-23:
Many D1 Mini clone boards have a paltry (150 mA) voltage regulator for 3.3V, meaning that it does not help having lots of 5V power available.

pcnetzwerke, 2023-05-23:
Hmmm, I understand. So I will try one after another:

1. Exchange 5V Power Supply for D1 Mini, Relay attached to 3.3V GPIO
2. Support 3.3V Power Supply with another DC-DC StepDown.

pcnetzwerke, 2023-05-23:
> How do you detect the device disappears from WiFi network?

this time:
1. no connection in browser
2. no ping
3. no more syslog
4. MQTT: LWT: offline

pcnetzwerke, 2023-05-24:
So I improved the power supply in different ways. Still the device works, but network is instable.

Since yesterday I find lots of hardware watchdog restarts in my syslog.

```
<13>1 2023-05-24T07:26:38.115975+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/STATE = {"Time":"2023-05-24T07:26:41","Uptime":"0T03:35:09","UptimeSec":12909,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER1":"off","Wifi":{"AP":1,"SSId":"mitohnekabel2.4","BSSId":"C0:56:27:7F:EA:77","Channel":6,"Mode":"1
1n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
<13>1 2023-05-24T07:28:11.115283+02:00 tasmota028 ESP-HTP - - - ESP-HTP: Hauptmenü
<13>1 2023-05-24T07:28:14.121659+02:00 tasmota028 ESP-HTP - - - ESP-HTP: Hauptmenü
<13>1 2023-05-24T07:28:49.733653+02:00 tasmota028 ESP-HDW - - - ESP-HDW: ESP8266EX
<13>1 2023-05-24T07:28:49.733653+02:00 tasmota028 ESP-CFG - - - ESP-CFG: aus Flash geladen von F5, Anzahl 110
<13>1 2023-05-24T07:28:49.733672+02:00 tasmota028 ESP-QPC - - - ESP-QPC: Count 1
<13>1 2023-05-24T07:28:49.733672+02:00 tasmota028 ESP-ROT - - - ESP-ROT: Mode 1
<13>1 2023-05-24T07:28:49.735397+02:00 tasmota028 ESP-SRC - - - ESP-SRC: Restart
<13>1 2023-05-24T07:28:49.737467+02:00 tasmota028 ESP-Projekt - - - ESP-Projekt tasmota - tasmota028 Version 12.5.0.3(sensors)-2_7_4_9(2023.05.23 07:25:53)
<13>1 2023-05-24T07:28:49.740381+02:00 tasmota028 ESP-AS6 - - - ESP-AS6: Detected with 4 fingerprint(s) stored
<13>1 2023-05-24T07:28:49.742012+02:00 tasmota028 ESP-WIF - - - ESP-WIF: Verbindungsversuch...
<13>1 2023-05-24T07:28:49.744235+02:00 tasmota028 ESP-WIF - - - ESP-WIF: verbinden mit AP1 mitohnekabel2.4 Channel 6 BSSId C0:56:27:7F:EA:77 in Modus 11n als tasmota028...
<13>1 2023-05-24T07:28:49.745809+02:00 tasmota028 ESP-WIF - - - ESP-WIF: verbunden
<13>1 2023-05-24T07:28:49.748849+02:00 tasmota028 ESP-HTP - - - ESP-HTP: Web-Server aktiv bei tasmota028 mit IP-Adresse 192.168.5.213
<13>1 2023-05-24T07:28:49.750680+02:00 tasmota028 ESP-WIF - - - ESP-WIF: DNS resolved '192.168.5.251' (192.168.5.251) in 0 ms
<13>1 2023-05-24T07:28:51.752501+02:00 tasmota028 ESP-WIF - - - ESP-WIF: DNS resolved '2.pool.ntp.org' (2.255.255.255) in 1001 ms
<13>1 2023-05-24T07:28:51.753704+02:00 tasmota028 ESP-NTP - - - ESP-NTP: No reply from 2.255.255.255
<13>1 2023-05-24T07:28:52.721333+02:00 tasmota028 ESP-WIF - - - ESP-WIF: DNS resolved '2.europe.pool.ntp.org' (213.81.129.98) in 16 ms
<13>1 2023-05-24T07:28:52.721814+02:00 tasmota028 ESP-RTC - - - ESP-RTC: UTC 2023-05-24T05:28:56, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
<13>1 2023-05-24T07:28:52.726982+02:00 tasmota028 ESP-RTC - - - ESP-RTC: Synced by NTP
<13>1 2023-05-24T07:28:52.728054+02:00 tasmota028 ESP-MQT - - - ESP-MQT: Verbindungsversuch...
<13>1 2023-05-24T07:28:52.728446+02:00 tasmota028 ESP-WIF - - - ESP-WIF: DNS resolved '192.168.5.3' (192.168.5.3) in 0 ms
<13>1 2023-05-24T07:28:52.728745+02:00 tasmota028 ESP-MQT - - - ESP-MQT: verbunden
<13>1 2023-05-24T07:28:52.730481+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/LWT = Online (beibehalten)
<13>1 2023-05-24T07:28:52.731078+02:00 tasmota028 ESP-MQT - - - ESP-MQT: cmnd/tasmota028/POWER =
<13>1 2023-05-24T07:28:52.732513+02:00 tasmota028 ESP-MQT - - - ESP-MQT: abonniere cmnd/tasmota028/#
<13>1 2023-05-24T07:28:52.734016+02:00 tasmota028 ESP-MQT - - - ESP-MQT: abonniere cmnd/tasmotas/#
<13>1 2023-05-24T07:28:52.735733+02:00 tasmota028 ESP-MQT - - - ESP-MQT: abonniere cmnd/tasmota028_fb/#
<13>1 2023-05-24T07:28:52.737362+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/INFO1 = {"Info1":{"Module":"Generic","Version":"12.5.0.3(sensors)","FallbackTopic":"cmnd/tasmota028_fb/","GroupTopic":"cmnd/tasmotas/"}}
<13>1 2023-05-24T07:28:52.738891+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota028","IPAddress":"192.168.5.213"}}
<13>1 2023-05-24T07:28:52.740287+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/INFO3 = {"Info3":{"RestartReason":"Hardware Watchdog","BootCount":37}}
<13>1 2023-05-24T07:28:52.741844+02:00 tasmota028 ESP-MQT - - - ESP-MQT: stat/tasmota028/RESULT = {"POWER1":"off"}
<13>1 2023-05-24T07:28:52.743289+02:00 tasmota028 ESP-MQT - - - ESP-MQT: stat/tasmota028/POWER1 = off
<13>1 2023-05-24T07:28:54.755806+02:00 tasmota028 ESP-QPC - - - ESP-QPC: Reset
<13>1 2023-05-24T07:28:56.719980+02:00 tasmota028 ESP-APP - - - ESP-APP: Anzahl Startvorgänge 37
<13>1 2023-05-24T07:28:57.716778+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/STATE = {"Time":"2023-05-24T07:29:01","Uptime":"0T00:00:10","UptimeSec":10,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"MqttCount":1,"POWER1":"off","Wifi":{"AP":1,"SSId":"mitohnekabel2.4","BSSId":"C0:56:27:7F:EA:77","Channel":6,"Mode":"11n"
,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}
<13>1 2023-05-24T07:28:57.719951+02:00 tasmota028 ESP-CFG - - - ESP-CFG: in Flash gespeichert an F4, Anzahl 111, Bytes 4096
<13>1 2023-05-24T12:40:14.067766+02:00 tasmota028 ESP-HDW - - - ESP-HDW: ESP8266EX
<13>1 2023-05-24T12:40:14.067766+02:00 tasmota028 ESP-CFG - - - ESP-CFG: aus Flash geladen von F8, Anzahl 115
<13>1 2023-05-24T12:40:14.067790+02:00 tasmota028 ESP-QPC - - - ESP-QPC: Count 1
<13>1 2023-05-24T12:40:14.068486+02:00 tasmota028 ESP-ROT - - - ESP-ROT: Mode 1
<13>1 2023-05-24T12:40:14.070045+02:00 tasmota028 ESP-SRC - - - ESP-SRC: Restart
[... cut ...]
<13>1 2023-05-23T17:31:37.230889+02:00 tasmota028 ESP-ROT - - - ESP-ROT: Mode 1
<13>1 2023-05-23T22:42:59.727792+02:00 tasmota028 ESP-ROT - - - ESP-ROT: Mode 1
<13>1 2023-05-23T23:20:06.825768+02:00 tasmota028 ESP-ROT - - - ESP-ROT: Mode 1
<13>1 2023-05-24T03:51:32.317424+02:00 tasmota028 ESP-ROT - - - ESP-ROT: Mode 1
<13>1 2023-05-24T07:28:49.733672+02:00 tasmota028 ESP-ROT - - - ESP-ROT: Mode 1
<13>1 2023-05-24T12:40:14.068486+02:00 tasmota028 ESP-ROT - - - ESP-ROT: Mode 1
```
What do they mean?

sfromis, 2023-05-24:
`ROT: Mode 1` is what I see when having a rotary encoder configured.

pcnetzwerke, 2023-05-24:
Ah, thank You.

Something new:

```
18:57:28.048 MQT: tele/tasmota028/INFO1 = {"Info1":{"Module":"Generic","Version":"12.5.0.3(sensors)","FallbackTopic":"cmnd/tasmota028_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:57:28.061 MQT: tele/tasmota028/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota028","IPAddress":"192.168.5.213"}}
18:57:28.080 MQT: tele/tasmota028/INFO3 = {"Info3":{"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["40264cfc","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4026c430","402508d1","401015a0","40250c0d","40250c92","40250d6c","40234d88","4025d140","402a406b","402295f8","40234dca","402a406b","40234e98","402a406b","40234ed2","4025d140","402368dc","4029a1d1","402a4c96","402a406b","40203c8e","402a406b","402373c0","40237580","402440e1","40250b8e","40251161","4025d7d0","402295f8","4026152a","40261588"]},"BootCount":39}}
18:57:28.133 MQT: stat/tasmota028/RESULT = {"POWER1":"off"}
18:57:28.137 MQT: stat/tasmota028/POWER1 = off
18:57:31.232 QPC: Reset
```

Jason2866, 2023-05-26:
To know if the issue is caused by the specific driver AS608, please do a customized build which includes only this driver.

pcnetzwerke, 2023-05-29:
> To know if the issue is caused by the specific driver AS608, please do a customized build which includes only this driver.

Did that and flashed the D1 Mini.

The device disappeares from network about one time a day. After some time more it reappeares after a self initiated reboot.

The last entry in sylog after 18 hours in WiFi is almost one day old:

`<13>1 2023-05-28T13:44:01.010529+02:00 tasmota028 ESP-MQT - - - ESP-MQT: tele/tasmota028/STATE = {"Time":"2023-05-28T13:44:01","Uptime":"0T18:10:23","UptimeSec":65423,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER1":"off","Wifi":{"AP":1,"SSId":"mitohnekabel2.4","BSSId":"C0:56:27:7F:EA:77","Channel":6,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}`

The fingerprint sensor and relais are still working perfectly.

I think I should detach the relais and/or the fingerprint sensor to see how long it remains online without those devices attached?

Jason2866, 2023-05-29:
>The device disappeares from network about one time a day. After some time more it reappeares after a self initiated reboot.

This is the typical behaviour when an esp8266 is not powered well. Not surprising since you are using a Wemos D1 mini. Since nearly no one of this boards have a decent 3.3V LDO on board. Swap the board against a NodeMCU and this issue will be gone.

Jason2866, 2023-05-29:
Closing the issue. The show issue is (now) is not related to the sensor.
Feel free to open when it arrives when not using a Wemos D1 mini.

closed 2023-05-29 by Jason2866

boaschti, 2023-05-30:
I flashed today 12.4.0.2(allsensors) from 2023-03-06T16:21:42 dev branch.
My Wemos D1 mini is modified with a RT9013 500mA 3.3V LDO.
The AS608 is powered with a separate 3.3 LDO.

Result:
No WIFI connection after about 4h.
Restart reason at 20:40 with exception.

And please remember I have NO WIFI and restart problems if I disable AS608. (Not tested with dev fw today)

boaschti, 2023-05-30:
Oh shit. The update failed. I have not double checked the date.

boaschti, 2023-06-01:
There are some Hardware Watchdog restarts with [this](https://github.com/Jason2866/Tasmota-specials/blob/firmware/firmware/tasmota/other/tasmota-allsensors.bin.gz) dev Firmware.
Should I test anything?

`08:02:18.648 MQT: stat/tuerpanel/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["TuerPanel","Tasmota2"],"Topic":"tuerpanel","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,0,15,15,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:02:18.654 MQT: stat/tuerpanel/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T04:38:29","StartupUTC":"2023-06-01T02:23:49","Sleep":50,"CfgHolder":4617,"BootCount":95,"BCResetTime":"2020-10-17T20:06:41","SaveCount":6237,"SaveAddress":"F9000"}}
08:02:18.660 MQT: stat/tuerpanel/STATUS2 = {"StatusFWR":{"Version":"12.5.0.3(allsensors)","BuildDateTime":"2023-05-30T20:53:08","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"408/699"}}
08:02:18.666 MQT: stat/tuerpanel/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHomeZeig","Servus nachad!"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
08:02:18.697 MQT: stat/tuerpanel/STATUS4 = {"StatusMEM":{"ProgramSize":743,"Free":260,"Heap":12,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F909787","04046001","FFFFB7CF","8F7417FB","303E0AD3","1159785A","26204FFF","5408F0F3","00000080"],"Drivers":"1,2,3,4,7,8,9,10,12,14,15,20,24,28,29,34,43,47,50,60,62,63,66","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,24,26,27,29,30,31,32,38,40,41,42,44,45,46,50,51,54,55,56,57,58,59,63,64,65,66,67,70,71,72,73,74,75,76,78,79,81,82,85,86,88,89,90,91,92,93","I2CDriver":"1,2,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,28,29,30,31,32,34,35,36,40,41,42,43,44,45,46,48,49,50,51,52,53,55,56,58,60,61,62,63"}}
08:02:18.708 MQT: stat/tuerpanel/STATUS5 = {"StatusNET":{"Hostname":"tuerpanel","IPAddress":"192.168.178.30","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"50:02:91:FE:55:40","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:02:18.716 MQT: stat/tuerpanel/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.38","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FE5540","MqttUser":"mqttClients","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:02:18.726 MQT: stat/tuerpanel/STATUS7 = {"StatusTIM":{"UTC":"2023-06-01T07:02:18","Local":"2023-06-01T08:02:18","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:44"}}
08:02:18.736 MQT: stat/tuerpanel/STATUS10 = {"StatusSNS":{"Time":"2023-06-01T08:02:18","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF"}}
08:02:18.745 MQT: stat/tuerpanel/STATUS11 = {"StatusSTS":{"Time":"2023-06-01T08:02:18","Uptime":"0T04:38:29","UptimeSec":16709,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":122,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SmartHomeZeig","BSSId":"00:E0:4C:13:96:F9","Channel":4,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:05"}}}
`

arendst, 2023-06-01:
Yep I see them too. I expect it's caused by software serial interrups going wrong unless you use hardware serial, then I'm out of ideas.

boaschti, 2023-06-01:
I use HW Serial.

boaschti, 2023-06-02:
Today a WIFI disconnect again.
Local switches and corresponding relais are working during tasmota is disconnected.

boaschti, 2023-06-04:
Today a Exception with a short pulse at restart on my door bell relais. (D8)