arendst/Tasmota #18184: ZBbridge connection problems - Need to configure port 8888 every day.

ryny24, 2023-03-13:
### PROBLEM DESCRIPTION
I'm using a ZBBridge with Tasmota to Home Assistant (Zigbee2MQTT) for a year now. It works great, when it does. But almost every day or every other day I see my lights are offline. I restart Z2M but it won't connect until I enter this into the console of the ZBBridge

`backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0`

I've updated to the latest Tasmota. I've searched for others having this issue, and mapped a GPIO15 to a relay but that hasn't helped.

Thank you for your help!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic):  Sonoff ZB Bridge Pro
- [ ] Tasmota binary firmware version number used: 12.4.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:39:34.962 CMD: Backlog Template; Module; GPIO 255
16:39:35.027 RSL: RESULT = {"NAME":"Sonoff ZHABridge","GPIO":[320,5088,0,5120,323,322,0,0,0,0,0,0,32,0],"FLAG":0,"BASE":18}
16:39:35.232 RSL: RESULT = {"Module":{"0":"Sonoff ZHABridge"}}
16:39:35.436 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"5088":"TCP Tx"},"GPIO2":{"0":"None"},"GPIO3":{"5120":"TCP Rx"},"GPIO4":{"323":"Led_i4"},"GPIO5":{"322":"Led_i3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:40:29.320 CMD: Status 0
16:40:29.324 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_26FBF3","ButtonTopic":"0","Power":1,"PowerOnState":5,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:40:29.327 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-zbbridge.bin.gz","RestartReason":"Exception","Uptime":"0T09:57:27","StartupUTC":"2023-03-13T05:43:02","Sleep":50,"CfgHolder":4617,"BootCount":136,"BCResetTime":"2022-06-14T19:15:24","SaveCount":308,"SaveAddress":"FD000"}}
16:40:29.329 RSL: STATUS2 = {"StatusFWR":{"Version":"12.4.0.2(zbbridge)","BuildDateTime":"2023-03-09T20:35:43","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"385/699"}}
16:40:29.331 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Sunflower",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000080","00006000","00004000","00000000"]}}
16:40:29.336 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":741,"Free":1048,"Heap":24,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","0F1007CE","04400001","00000003","00000000","00000000","00020080","00200000","04000800","00000080"],"Drivers":"1,2,4,7,9,10,12,20,23,38,41,50,59,62","Sensors":"1","I2CDriver":""}}
16:40:29.339 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-26FBF3-7155","IPAddress":"192.168.240.119","Gateway":"192.168.240.1","Subnetmask":"255.255.254.0","DNSServer1":"192.168.240.1","DNSServer2":"0.0.0.0","Mac":"48:3F:DA:26:FB:F3","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:40:29.341 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_26FBF3","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:40:29.344 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-03-13T15:40:29","Local":"2023-03-13T16:40:29","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"07:08","Sunset":"18:51"}}
16:40:29.346 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-03-13T16:40:29"}}
16:40:29.350 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-03-13T16:40:29","Uptime":"0T09:57:27","UptimeSec":35847,"Vcc":3.447,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Sunflower","BSSId":"2C:E6:CC:0A:E9:08","Channel":1,"Mode":"11n","RSSI":100,"Signal":-22,"LinkCount":1,"Downtime":"0T00:00:04"}}}
16:40:29.356 RSL: STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40268491","401010b8","4027d127","40105593","4027d114","4027d0bb","4027c210","4027c22d","40279ca0","4027600d","4027a997","4027a40a","40283d4f","4028360b","40252778","40000f49","40000f49","40000e19","40105699","4010569f","4010000d","402824e0","40282491","4025e426","4025ed5d","4025e426","4025e426","4029e734","4025ed5d","4029e734","4029e735"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gerritdrost, 2023-03-14:
Two days ago I changed up two things in my setup:
- upgraded Tasmota on my zbbridge from 12.3.1 to 12.4.0
- switched from ZHA to Zigbee2mqtt

Yesterday I first experienced this exact issue. ZHA also used the zbbridge on the same TCP port and had been rock solid for weeks. 

@ryny24 You say you upgraded Tasmota while trying to fiix this. What was the old version of Tasmota you had these problems with? If you also had this issue with Tasmota 12.3.1 or older, the combination of Zigbee2mqtt and Tasmota seems to be the culprit, as those versions worked just fine with ZHA.

ryny24, 2023-03-15:
I don't know the exact version I flashed with, but whatever version was released before 6/14/22.   Since then, I've upgraded maybe 3 times. Each one having the same issue.  Please let me know if you switch back to one (or both) of those, and the problem goes away. I would suggest ZHA first, as I've never had my ZBBridge on ZHA.

ryny24, 2023-03-20:
I couldn't take it anymore. I tried switching the ZBBridge gateway to a [Moes ZHUB Ethernet Gateway](https://www.aliexpress.com/item/2255800279107159.html?gatewayAdapt=4itemAdapt), and so far so good. My devices have been online for a week now. I'm not sure if the MOES is running Tasmota or not, but it works. The flashing process is [here](https://zigbee.blakadder.com/Tuya_TYGWZ-01.html).

gerritdrost, 2023-03-22:
Problem seems to be caused by zigbee2mqtt, not tasmota. I'm also switching to a different Zigbee Gateway that is better suited for use with Zigbee2mqtt. This issue can probably be closed.

ryny24, 2023-03-24:
Z2M working rock solid now with my Moes ZHUB Ethernet Gateway. Will close this issue.

closed 2023-03-24 by ryny24

s-hadinger, 2023-04-14:
Why are you doing that? Why not a rule to start tcp at startup. The only reason tcp would stop is if you explicitly stop it or reboot

mklan, 2023-04-14:
@s-hadinger there is currently a bug in zigbee2mqtt (maybe also home-assistant) that somehow causes the tcp connection to crash. For me it also breaks the startup rule and I need to set the rule again.

s-hadinger, 2023-04-15:
I must be missing some information. TCP connections don't crash. And if Z2M remotely makes Tasmota rules not to work, that is a serious bug that needs to be addressed. Can you provide more information?

gerritdrost, 2023-04-15:
Well it's most definitely happening, at least, in terms of us needing to reconfigure the TCP port to get it working again.

I switched to a different gateway to mitigate my problems, but I'll summarize it once more, it might help others experiencing this problem:
- the problem only occurs when using a Tasmota-flashed Sonoff zbbridge with Z2M (Zigbee2mqtt). The exact same bridge with ZHA works fine. It had been running fine using ZHA for weeks before I switched to Zigbee2mqtt and started encountering this bug.
- Initially, everything is fine. Only after a couple of hours will Zigbee2mqtt lose connection to the gateway.
- The time to failure feels non-deterministic. Sometimes it would happen after 5 hours, sometimes it would work almost a full day.
- the only way to recover the connection is to reconfigure the TCP port in Tasmota. Only restarting Zigbee2mqtt does not recover the connection.
- Downgrading to Z2M 1.30.1 did not fix the issue
- Downgrading to Z2M 1.29.2 fixes the issue. Z2M still crashes on a regular basis, but the error is different, and it reconnects after a restart without reconfiguring the TCP port in Tasmota. 

I never took the effort to use tcpdump to see what actually happens at the TCP level, the underlying cause might be at another level. However, me and others definitely needed to reconfigure the TCP port to get it working again. As this problem only occurs in recent versions of Z2M, and the changelog of Z2M reports several changes to the EZSP code in recent versions, I suspect the bug is in the Z2M codebase, not the Tasmota codebase. That's why I suggested the issue be closed here.

barbudor, 2023-04-15:
You haven't provided any actual and detailed information and logs about Tasmota.

TCPbridge would likely stop only on a restart so that's what you should track
And a rule on wifi#connected should be your solution to automatically re-enable the bridge on restart.

s-hadinger, 2023-04-15:
Ok. I think I get it now. You mean that the TCP connection is still alive but zigbee2mqtt can't use it anymore. Hence you need to restart tcp on Tasmota side to kill the previous connection?

mccasian, 2023-07-26:
Hello,

I am also encountering the same issue with the combination of HA(2023.7.3) + Zigbee2MQTT(v1.32.0, Coordinator Type EZSP V8, revision 6.7.8.0 build 373) + Sonoff ZBBridge(Tasmota 13.0.0).

I have set the following rule to start the TCP on boot:
ON system#boot DO Backlog Delay 20; TCPStart 8888 endon

However, that works until TCP crashes/stops and Z2M service is complaining that it's not able to connect to the bridge on port 8888. Testing it with telnet confirms: port is closed. 
I always have to issue the command "TCPStart 8888" or use the workaround suggested by @mklan.

I can confirm that the issue is happening exactly like @ryny24 described it. 

I am not an expert with Tasmota, but i would like to contribute by providing information, logs, anything - please suggest.

So far i can tell:
- rule based on "ON wifi#connected" does not work. "ON system#boot" works but it crashes at random intervals.
- upgrading coordinator to 6.7.9 does not help, it only bring other stability issues on this type of bridge.
- I tested it with both Tasmota 12.x and 13.0.0
- the cause of TCP "fails" is not device reboot, it must be something else. I suspect WIFI instability, which causes the bridge to lose the IP, so the TCP fails to bind on the IP, therefore crashes - but this is just an assumption, i have no idea how to check this properly.

Next that I will try(and report after):
- downgrade Z2M to 1.29.2 as suggested by @gerritdrost 

However, I dare to say that Sonoff ZBBridge is not the best choice for HA+Z2M - not with the current (buggy) versions. I wish I knew that 2 year ago when I bought multiple bridges to cover all areas.

s-hadinger, 2023-07-26:
Could you please decompose the problem and do some simple tests.

When the problem occurs, can you check with a computer whether the TCP port is open on port 8888. It's unclear whether it's Z2M that can't connect, or if it's Tasmota that lost the TCP server.

You mention TCP to "crash". Do you really see a Tasmota crash? Or is just the TCP port 8888 refusing connections?

s-hadinger, 2023-07-26:
One more thing, what does the Tasmota logs says when you try to connect with TCP?

s-hadinger, 2023-07-26:
Please try again with the latest version in development. #19199 might solve an IP filtering problem since we enabled IPv6.

mccasian, 2023-07-26:
While I can't say for sure, yet, It looks like Tasmota is losing the TCP server. I believe that since the telnet from my computer(or any other device in the same network) to port 8888 will fail.

> One more thing, what does the Tasmota logs says when you try to connect with TCP?

`00:24:49.121 HTP: Main Menu
00:24:50.625 HTP: Consoles
00:24:51.575 HTP: Console
00:24:57.529 CMD: TCPStart 8888
00:24:57.530 SRC: WebConsole from 172.16.0.152
00:24:57.532 CMD: Grp 0, Cmd 'TCPSTART', Idx 1, Len 4, Pld 8888, Data '8888'
00:24:57.534 TCP: Stopping TCP server
00:24:57.536 TCP: Starting TCP server on port 8888
00:24:57.539 MQT: stat/zbBridgeKeller/RESULT = {"TCPStart":"Done"}`

> Please try again with the latest version in development. #19199 might solve an IP filtering problem since we enabled IPv6.

I have downgraded Z2M to 1.29.2 for all bridges, also installed Tasmota 13.0.0.3 which includes the ip filter fix/feature on one(only) of the bridges and I configured a syslog-ng docker container which is now collecting all the logs from my tasmota(level 4 logging - more debug).

As soon as I notice any issue with the current setup, I will post here.

Symbianx, 2023-08-24:
@s-hadinger I'm running tasmota 13.1.0 which should include #19199 but I'm still having the issue described in this thread.

s-hadinger, 2023-08-25:
You didn't provide the logs. Did you get some?

mklan, 2023-10-10:
> Could you please decompose the problem and do some simple tests.
>
> When the problem occurs, can you check with a computer whether the TCP port is open on port 8888. It's unclear whether it's Z2M that [...]

I periodically run `nc -z -w 2 $ZB_HOST 8888 2>&1 >/dev/null` to check if TCP is running. So I assume the issue is at inside Tasmota. This problem still occurs in the latest version.

depuytnl, 2023-10-23:
> I am running the following script using crontab every minute. This will restart tcp if it stops:
> 
> ```shell
> #!/bin/bash
> 
> HOST=192.168.1.102
> WEB_PORT=80
> TCP_PORT=8888
> 
> if (nc -z -w 2 $HOST $WEB_PORT 2>&1 >/dev/null)
> then
>     if !(nc -z -w 2 $HOST $TCP_PORT 2>&1 >/dev/null)
>     then
>         echo "[$(date)] tcp not running. Will start!" >> zbbridge.log
>         curl http://$HOST/cm?cmnd=TCPStart%20$TCP_PORT
>     fi
> else
>         echo "[$(date)] zbbridge not reachable!" >> zbbridge.log
> fi
> ```

Do you run this on the sonoff? If so, can you tell me how you got it to do that?

sfromis, 2023-10-23:
Clearly, that script is not run on a Tasmota device, it would typically be on a Linux PC.

fmnamado, 2023-10-30:
Hello,
I have been having this problem also.
I have Sonoff Zigbee bridge for 1 year without problems, but some months ago this started.
I tried reflashing (via OTA), minimal then the specific sonoff tasmota firmware, same problem.

Updated to 13.1.0 that I am running, same problem.

Some random, but mostly at least once a day, it loses conection to my HA.
Issuing the command in the console solves the problem.
But I loose connection during the night, and I would like to monitor some sensors and trigger an alarm...

s-hadinger, 2023-10-30:
Hi. I'm happy to work on this problem but nobody has provided any meaningful logs. I understand the symptoms, they have been clearly explained. Now we need logs from Tasmota when tcp connection fails

fmnamado, 2023-10-30:
Can you please help me how to obtain them?
I am sorry but I don't know where to look for.
Thank you for your assistance.

mccasian, 2023-10-30:
I will provide them today, but there is nothing relevant on them. I can already say what it's happening, and the fix should most probably be implemented in Tasmota.

I was not able to figure out the source of the problem until today when I had to reboot the router, so the Wi-Fi was gone.
Yep, unstable Wi-Fi means that Tasmota is losing connection to router, so it's loosing the IP on the Wi-Fi interface, therefore the process listening on that specific IP probably crashes. As soon as the connection is re-established, the the interface is available again and the listening can be restarted with the command "TCPStart 8888".

I would suggest to make the process listen on all interfaces(any/*) or create a keep alive mechanism to restart listening whenever the interface has IP again. 

5 minutes ago I set Manual IP on my Tasmota; let's see if this helps.

mccasian, 2023-10-30:
My Home Assistant reports that my Zigbee sensors went unavailable at 3:28:02 but due to 10 min delay in mqtt, the real time when it happened is 03:18:xx(or maybe 3:17:xx).

I don't see anything abnormal but maybe the comm between the two MCUs may reveal something.

! private data was scrambled

[Oct 30 03:14:29 zbbridgehome zbBridgeHome ESP-WIF.log](https://github.com/arendst/Tasmota/files/13208860/Oct.30.03.14.29.zbbridgehome.zbBridgeHome.ESP-WIF.log)

mccasian, 2023-10-30:
Same here, look at 8:42. Home Assistant reported data until 8:52(crash time + 10 min delay from MQTT)
[2023-10-30T07:40:02.3110000 zbbridgehome HOST.log](https://github.com/arendst/Tasmota/files/13208916/2023-10-30T07.40.02.3110000.zbbridgehome.HOST.log)

fmnamado, 2023-12-05:
I tried by this order:

- Restart tasmota
- restart zigbee2mqtt
- restart mosquitto addon in home assisstant
- restart home assistant

none of this worked.

only copy and pasting the rule would work, for a day, and the process of losing conectivity would repeat.
i remembered this wasn't a problem in the past...

Googling, I read somewhere "I am running Tasmota 9.3.1 rock solid" and "I am running Tasmota 9.2.0 and don't have problems."

I downgraded first to 9.3.1, had to run "module 75" in the console before flashing the zigbee firmware, because it said some error about signature.

I lost my zigbee nodes, had to repair them all.

for now it is running rock solid, for some days only, but it is more time than previously...
try to flash the 9.3.1 zbbdrige bin and then the zigbee firmware and tell me what happens.

fmnamado, 2023-12-06:
test post previouly not appeaering

mccasian, 2023-12-31:
I've been involved in some other discussions on the zigbee topics because of the same issue. The conclusion I got from there is that the zigbee coordinator running on the bridge and is pretty stable.
Tasmota main purpose is to provide a "tasmota like solution" between that Zigbee Coordinator flashed on the ZB chip and Zigbee2Mqtt or alternatives.

So tasmota always communicates with the ZB chip on the board, and exposes the info on the port you set(usually 8888). The problem is that tasmota stops listening on port 8888 randomly(apparently).

I am collecting logs from zbbridges on my Syslog server, and i've saw during the time that at the moment when the sensors became unavailable in Home Assistant, Syslog shows either ZbBridge restarts or Wifi disconnects.  Considering this, i see the following:
- when the device restarts, the rule 1 will start the listener on port 8888 ... all good so far.
- when the wifi is unstable, the Wifi adapter loses it's IP(normal), and the listener has no IP to listen(normal if not listening on */0.0.0.0). Here is the moment, when no automation restarts the listener anymore, and the bridge is not listening anymore until manual intervention.

My suggestion here, although i am sorry i can't help due to lack of programming knowledge, is that Tasmota should get a mechanism to restart the listener when the wifi connection is back, or listen on all interfaces.

s-hadinger, 2023-12-31:
The listener is listening at `0.0.0.0` and `::0` so I don't understand why it doesn't pick the new IP. Also I didn't find an easy way to simulate this change of IP. Any idea?

mccasian, 2023-12-31:
not trying to be ironic or something, but you could stop the wifi router for a moment and see what happens after everything is back online.

Also, listening on all interfaces may not be helpful if there is no other interface apart from the WIFI one. There must be at least loopback interface to prevent the listener drop. Is this the case with Tasmota?

s-hadinger, 2023-12-31:
This wouldn't work because my router always gives back the same address if the device was already connected

mccasian, 2023-12-31:
mine too. The idea is to make the listener crash when the WIFI drops, not necessary preventing the Tasmota to get the same IP after reconnecting.