arendst/Tasmota #18193: shutter feature broken

Gifford47, 2023-03-14:
### PROBLEM DESCRIPTION
Some of the shutter commands returning command:error. F.e. shutterclose or shutterpwmrange
shuttermode is also not saved after reboot,

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: _12.4____
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
flash tasmota 12.4
backlog SetOption80 1;Shuttermode 5;PWMfrequency 200;SetOption15 0;ShutterPwmRange1 109, 520;ShutterOpenDuration1 5;ShutterCloseDuration1 5;ShutterMotorDelay1 0.2

reboot and return command shuttermode

### EXPECTED BEHAVIOUR
Shutter commands are executed as normal

### SCREENSHOTS

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-03-16:
are you sure that you did not messed up with "savemode"? runs fine on my side. From which version did you upgrade? From 10 there was a change on shuttermode detection. Anyhow if you set shutternmode 5 manually this issue should not appear. Sending the logfile during boot with loglevel 4 will help to identify the issue. 

FIRST: check your savemode

Looks like your settings are not saved. This would explain why it worked before the reboot but not after the reboot

stefanbode, 2023-03-16:
`shutterrelay1 1` is also missing in your configuration

Gifford47, 2023-03-16:
> are you sure that you did not messed up with "savemode"? runs fine on my side. From which version did you upgrade? From 10 there was a change on shuttermode detection. Anyhow if you set shutternmode [...]

ok, thanks for the hint! now it works!

closed 2023-03-16 by Gifford47