arendst/Tasmota #18198: Infinite loop at end of movement when using shutter with tilt configuration

Gemikro1, 2023-03-15:
### PROBLEM DESCRIPTION
I am using Shelly 2.5 for controlling venetian blinds with tilt configured
```lua
19:53:26.751 CMD: ShutterTiltConfig
19:53:26.756 SHT: TiltConfig 1, min: -90, max 90, runtime 26, close_pos: -10, open_pos: -90
19:53:26.812 MQT: stat/swJalousieWZ-01/RESULT = {"ShutterTiltConfig1":"-90 90 26 -10 -90"}
```
When moving the shutter to 100% open in any way, the up relais starts opening and closing infinitely with about 1 second intervals until another shutter position less than 100% is commanded. The problem does not occur when only moving to e.g. position 99.
At each loop, the message `SHT: Tilt not match 90 -> -10` is issued.
```lua
20:25:32.108 CMD: ShutterPosition 100
20:25:32.125 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"ON"}
20:25:32.136 MQT: stat/swJalousieWZ-01/POWER1 = ON
20:25:32.153 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":99,"Direction":1,"Target":100,"Tilt":90}}
20:25:32.232 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":99,"Direction":1,"Target":100,"Tilt":90}}
20:25:33.216 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:33.658 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:25:33.669 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:25:33.671 SHT: Tilt not match 90 -> -10
20:25:34.217 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"ON"}
20:25:34.226 MQT: stat/swJalousieWZ-01/POWER1 = ON
20:25:34.238 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:34.265 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:35.172 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:35.271 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:25:35.280 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:25:35.282 SHT: Tilt not match 90 -> -10
20:25:35.799 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"ON"}
20:25:35.811 MQT: stat/swJalousieWZ-01/POWER1 = ON
20:25:35.823 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:36.194 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:36.889 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:25:36.901 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:25:36.903 SHT: Tilt not match 90 -> -10
20:25:37.449 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"ON"}
20:25:37.461 MQT: stat/swJalousieWZ-01/POWER1 = ON
20:25:37.473 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:38.191 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:38.541 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:25:38.553 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:25:38.556 SHT: Tilt not match 90 -> -10
20:25:39.068 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"ON"}
20:25:39.080 MQT: stat/swJalousieWZ-01/POWER1 = ON
20:25:39.094 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:39.221 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:40.136 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:25:40.146 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:25:40.149 SHT: Tilt not match 90 -> -10
20:25:40.673 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"ON"}
20:25:40.686 MQT: stat/swJalousieWZ-01/POWER1 = ON
20:25:40.699 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:41.226 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
20:25:41.725 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:25:41.739 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:25:41.742 SHT: Tilt not match 90 -> -10
20:25:42.002 CMD: ShutterPosition 99
20:25:42.269 MQT: stat/swJalousieWZ-01/RESULT = {"POWER2":"ON"}
20:25:42.282 MQT: stat/swJalousieWZ-01/POWER2 = ON
20:25:42.295 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":99,"Tilt":90}}
20:25:42.387 MQT: stat/swJalousieWZ-01/RESULT = {"POWER2":"OFF"}
20:25:42.397 MQT: stat/swJalousieWZ-01/POWER2 = OFF
20:25:42.461 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"ON"}
20:25:42.472 MQT: stat/swJalousieWZ-01/POWER1 = ON
20:25:42.484 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":84}}
20:25:42.499 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:25:42.511 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:25:42.522 MQT: stat/swJalousieWZ-01/SHUTTER1 = 100
```
Apart from the 100% position, movement of the shutter and tilt control is working as expected.

[... 412 lines cut ...]
20:02:45.248 SHT: Shtr1 Time 23(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 810000, nStop 0, Trgt 810000, mVelo 0, Dir 1, Tilt 62, TrgtTilt: 90, Tiltmove: 1
20:02:45.473 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:02:45.484 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:02:45.496 MQT: stat/swJalousieWZ-01/SHUTTER1 = 90
20:02:45.508 MQT: stat/swJalousieWZ-01/RESULT = {"Shutter1":{"Position":90,"Direction":0,"Target":90,"Tilt":90}}
20:02:55.429 CMD: weblog 2

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Send any commands that move the shutter to position 100 as `ShutterPosition open` or `ShutterPosition 100` or move the slider on the web interface to 100%.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Shutter stops at 100% open, no additional actions on the relais.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-03-16:
plls check the "shuttertiltconfig" and send me the output. It looks like this definition has a zero runtime. That might explain the behavior. Am I correct you HAVE venetian blinds and want to use them. Maybe you start with the example of the documentation and check if it works in general.

stefanbode, 2023-03-16:
do you have the same behavior if using "shutterposition 100" or shutteropen / shutterclose. I will check on my bench. There where some changes regarding tilt and the ENDPOSITON. Anyhow I agree the behavior is not intended

stefanbode, 2023-03-16:
The way it should work:
```
      // if position is either 0 or 100 reset the tilt to avoid tilt moving at the end
      if (XdrvMailbox.payload ==   0 && ShutterRealToPercentPosition(Shutter[index].real_position, index)  > 0  ) {Shutter[index].tilt_target_pos = Shutter[index].tilt_config[4];}
      if (XdrvMailbox.payload == 100 && ShutterRealToPercentPosition(Shutter[index].real_position, index)  < 100) {Shutter[index].tilt_target_pos = Shutter[index].tilt_config[3];}
```
if you move it to 0 or 100 the first time the TILT gets deleted so there is no move. IF the shutter is ALREADY at the endpoints and you send a shutterposition or shuttertilt the TILT will be maintained.

stefanbode, 2023-03-16:
I do not understand why it stop at 920000 and 3 seconds later the position is 900000 and start again. This is definitly wrong...

```
20:02:21.975 SHT: Shtr1 Time 125(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 920000, nStop 0, Trgt 920000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
20:02:21.977 SHT: Stop 1 Mode 0 time 125
20:02:21.979 SRC: Shutter
20:02:21.980 SHT: Switched relay 1 by Shutter
20:02:21.992 MQT: stat/swJalousieWZ-01/RESULT = {"POWER1":"OFF"}
20:02:22.005 MQT: stat/swJalousieWZ-01/POWER1 = OFF
20:02:22.007 SHT: Tilt not match 90 -> -10
20:02:22.008 SHT: Pos. payload <> (0), payload -99, idx 1 (0), src 23
20:02:22.211 ADE: Channel1 ACCMODE 0x2D2800, VRMS 6068779, Period 4477, IRMS 1814, WATT -86, VA 129, VAR -51
20:02:22.214 ADE: Channel2 ACCMODE 0x2D2800, VRMS 6069680, Period 4477, IRMS 1773, WATT 0, VA 28, VAR 0
20:02:23.319 SHT: Shtr1 Time 16(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 915000, nStop 0, Trgt 920000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
20:02:23.369 SHT: Shtr1 Time 17(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 916000, nStop 0, Trgt 920000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
20:02:23.419 SHT: Shtr1 Time 18(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 917000, nStop 0, Trgt 920000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
20:02:23.469 SHT: Shtr1 Time 19(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 918000, nStop 0, Trgt 920000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
20:02:23.519 SHT: Shtr1 Time 20(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 919000, nStop 0, Trgt 920000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
20:02:23.569 SHT: Shtr1 Time 21(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 920000, nStop 0, Trgt 920000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
20:02:23.570 SHT: Stop 1 Mode 0 time 21
```

stefanbode, 2023-03-16:
ok, worked on my side with the current release. MAIN difference is that I use shutteropenduration 45 and if I send a shutterposition 100 my target in the detail log in 900000 and not 920000. We have to find what makes this change on your side.
```
10:38:36.447 RSL: RESULT = {"Shutter1":{"Position":93,"Direction":1,"Target":100,"Tilt":90}}
10:38:37.451 RSL: RESULT = {"Shutter1":{"Position":95,"Direction":1,"Target":100,"Tilt":90}}
10:38:38.457 RSL: RESULT = {"Shutter1":{"Position":98,"Direction":1,"Target":100,"Tilt":90}}
10:38:39.438 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
10:38:39.487 RSL: RESULT = {"POWER1":"OFF"}
10:38:39.492 RSL: POWER1 = OFF
10:38:39.494 SHT: Tilt not match 90 -> -10
10:38:40.014 RSL: RESULT = {"POWER2":"ON"}
10:38:40.017 RSL: POWER2 = ON
10:38:40.030 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":100,"Tilt":90}}
10:38:40.413 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":100,"Tilt":42}}
10:38:40.819 RSL: RESULT = {"POWER2":"OFF"}
10:38:40.823 RSL: POWER2 = OFF
10:38:40.835 RSL: SHUTTER1 = 100
10:38:40.840 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":-13}}
10:39:38.442 RSL: SENSOR = {"Time":"2023-03-16T10:39:38","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":-13}}
```
```

10:38:39.319 SHT: Shtr1 Time 114(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 897000, nStop 0, Trgt 900000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
10:38:39.370 SHT: Shtr1 Time 115(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 898000, nStop 0, Trgt 900000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
10:38:39.421 SHT: Shtr1 Time 116(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 899000, nStop 0, Trgt 900000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
10:38:39.428 SHT: Shtr1 Real 899000, Start 810000, Stop 900000, Dir 1, Delay 0, Rtc 5.80 [s], Freq 0, PWM 0, Tilt 90
10:38:39.438 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":1,"Target":100,"Tilt":90}}
10:38:39.471 SHT: Shtr1 Time 117(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir 1, Tilt 90, TrgtTilt: -10, Tiltmove: 0
10:38:39.476 SHT: Stop 1 Mode 0 time 117
10:38:39.480 SRC: Shutter
10:38:39.482 SHT: Switched relay 1 by Shutter
10:38:39.487 RSL: RESULT = {"POWER1":"OFF"}
10:38:39.492 RSL: POWER1 = OFF
10:38:39.494 SHT: Tilt not match 90 -> -10
10:38:39.498 SHT: Pos. payload <> (3), payload -99, idx 1 (0), src 23
10:38:39.699 CFG: Saved to flash at FC, Count 28, Bytes 4096
10:38:40.003 SHT: Delay Start? var1 <99>=<>, max10s?
10:38:40.004 SHT: VenetianDelay: 26, Pos: 90, Dir: -1, Delta: 180, Dur: 26, StartP: 90, TgtP: -10
10:38:40.006 SRC: Shutter
10:38:40.008 SHT: Switched relay 2 by Shutter
10:38:40.014 RSL: RESULT = {"POWER2":"ON"}
10:38:40.017 RSL: POWER2 = ON
10:38:40.021 SHT: Shtr1 Real 900000, Start 900000, Stop 900000, Dir -1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 90
10:38:40.030 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":100,"Tilt":90}}
10:38:40.045 SHT: Shtr1 Time 1(1), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 90, TrgtTilt: -10, Tiltmove: 1
10:38:40.095 SHT: Shtr1 Time 2(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 84, TrgtTilt: -10, Tiltmove: 1
10:38:40.147 SHT: Shtr1 Time 3(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 77, TrgtTilt: -10, Tiltmove: 1
10:38:40.197 SHT: Shtr1 Time 4(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 70, TrgtTilt: -10, Tiltmove: 1
10:38:40.247 SHT: Shtr1 Time 5(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 63, TrgtTilt: -10, Tiltmove: 1
10:38:40.296 SHT: Shtr1 Time 6(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 56, TrgtTilt: -10, Tiltmove: 1
10:38:40.347 SHT: Shtr1 Time 7(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 49, TrgtTilt: -10, Tiltmove: 1
10:38:40.396 SHT: Shtr1 Time 8(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 42, TrgtTilt: -10, Tiltmove: 1
10:38:40.404 SHT: Shtr1 Real 900000, Start 900000, Stop 900000, Dir -1, Delay 0, Rtc 0.40 [s], Freq 0, PWM 0, Tilt 42
10:38:40.413 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":100,"Tilt":42}}
10:38:40.449 SHT: Shtr1 Time 9(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 35, TrgtTilt: -10, Tiltmove: 1
10:38:40.499 SHT: Shtr1 Time 10(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 28, TrgtTilt: -10, Tiltmove: 1
10:38:40.551 SHT: Shtr1 Time 11(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt 21, TrgtTilt: -10, Tiltmove: 1
[... cut ...]
10:38:40.802 SHT: Shtr1 Time 16(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 900000, nStop 0, Trgt 900000, mVelo 0, Dir -1, Tilt -13, TrgtTilt: -10, Tiltmove: 1
10:38:40.807 SHT: Stop 1 Mode 0 time 16
10:38:40.811 SRC: Shutter
10:38:40.814 SHT: Switched relay 2 by Shutter
10:38:40.819 RSL: RESULT = {"POWER2":"OFF"}
10:38:40.823 RSL: POWER2 = OFF
10:38:40.827 SHT: Shtr1 Real 900000, Start 900000, Stop 900000, Dir 0, Delay 0, Rtc 0.80 [s], Freq 0, PWM 0, Tilt -13
10:38:40.835 RSL: SHUTTER1 = 100
10:38:40.840 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":-13}}
10:38:41.717 CFG: Saved to flash at FB, Count 29, Bytes 4096
```

Gemikro1, 2023-03-16:
> do you have the same behavior if using "shutterposition 100" or shutteropen / shutterclose. I will check on my bench. There where some changes regarding tilt and the ENDPOSITON. Anyhow I agree the behavior is not intended

Tested with all commands that set the position to 100 via console / MQTT, all lead to the same result. 
To me it looks like the Tilt is reset to the "open" position from the ShutterTiltConfig which is -10, tries to move from the actual tilt 90 to -10 but due to some reason it resets the aPos2 to 900000 which starts the cycle again. 

I thought that there is no Tilt restore at both endpositions in the current Tasmota version ? Why does the shutter try to move at all ?
If you have anything i can test please send me, i can compile and try it. If you send me the name of the module i can also try to have a look at the code.

Gemikro1, 2023-03-16:
I found something: the effect only happens when `ShutterEnableEndStopTime` is set to 1 which was the case on this Shelly. When i set it back to 0 the shutter behaves correctly at both end positions.

stefanbode, 2023-03-16:
thanks this is valueable information and explain the 9200000. Looks like I have to tweak the calculation in this case. The extra end time was not tested with the tilt. I know where to search....

Gemikro1, 2023-03-16:
by the way: would it be possible to reduce the minimum time step for the Tilt control from the current 50ms to a somewhat lower value or do we reach a limit here already ? The number of steps in Tilt movement is just slightly too coarse in my case.

stefanbode, 2023-03-16:
Submitted a fix for the AdditionalEndStopTime. Regarding your question. 50ms is the absolute minimum TASMOTA allow to hook in. Even this time is not a guarantee and depend on the load in the system. No change possible here. I know that tilt is a bit vague because of this.

Gemikro1, 2023-03-16:
No, problem, thought it would be like this, tilt steps are acceptable for me but could be somewhat finer resolved. And Tasmota is currently the only firmware that can handle Venetian blinds reasonably well anyway. Thanks for the good work here !

closed 2023-03-16 by arendst

stefanbode, 2023-03-31:
@Gemikro1 : There is a new version for ESP32 only. you need compiler option currently to get this enabled. Many new features you might be interested in.
There is a new enhanced ESP32 version you can enable with the following compiler option #define USE_SHUTTER_ESP32. Will be default in future versions for ESP32 after some testing. New features most likely only in ESP32 due to memory limitations in ESP8266. Current feature set:

up to 16 shutters,
up to 32 shutterbuttons,
tilt definition with buttons.
Configuration saved to filesystem.

Gemikro1, 2023-04-01:
@stefanbode : 
Thanks for the info, do you know if this has been tested on the new Shelly Plus 2PM already ? 
I did not yet switch from the Shelly 2.5 to the new Plus version as these seem to have much higher consumption, at least with the default firmware. On my Shelly 2.5 using Tasmota i measure < 0.3 W on average, the datasheet of the Shelly Plus devices states <1.4 W, at >20 devices the difference would already quite be noticeable. 
It would be interesting to see how low consumption can be made using Tasmota instead of the default firmware.

stefanbode, 2023-04-01:
The 2.5 is no ESP32 as I know. Therefore no changes to this devise. Power consumption is out of control of driver development. This is in the core or even in the espressif IDE. Nothing you can easily influence. Anyhow the amount of shutters do not change the power consumption.

Gemikro1, 2023-04-01:
Yes, i know, but the Plus has an ESP32. Seems i simply will by one unit and test. 
At least for the 8266, consumption could be tweaked to be la bit ower than with the default firmware.

stefanbode, 2023-04-01:
I just ordered two. Let's see what crazy stuff we can do with it. I was not aware it is ESP32.

Gemikro1, 2023-04-01:
There seem to be several different versions though: [https://www.esphome-devices.com/devices/Shelly-Plus-2PM](url)

stefanbode, 2023-04-02:
Hmm looks like only the 240Mhz dual core with DH in the name of the chip can be flashed. Let's see what Amazon is sending.

Gemikro1, 2023-04-02:
Are you sure ? At least blakadder provides Tasmota templates for both the v0.1.5 and the v0.1.9 versions. 
https://templates.blakadder.com/shelly_plus_2PM.html

stefanbode, 2023-04-04:
I do not know why. But Have problem with flashing. Connected 4 wires GND,RX,TX,GND then press the button on the device and connect to USB. No way to get it ready for flashing. What is the Pin2 CHIP_PU? Never seen before. Tried to pull this on 3.3V but even here no success. 
It is the DH 240MhZ Chip

stefanbode, 2023-04-04:
And yes I have flashed several other devices with success......

Gemikro1, 2023-04-08:
Did you try before to flash it over WLAN ? 
I usually flash all my devices this way because they are installed already in the boxes near the blinds.
In case of low memory you need to flash a minimal installer first though to be able to then flash the final Tasmota image.

`CHIP_PU` is directly connected to the ESP32, this is the Chip Power Up input and has to be pulled up.

When manually flashing, did you start up the device with `GPIO0` grounded ? You did not mention this pin.
I did not receive my devices yet so, will try when i have them.

Gemikro1, 2023-04-08:
I also found the pinout in the Shelly knowledge base:
 [https://kb.shelly.cloud/knowledge-base/shelly-plus-2pm](url)
There seems to also be a debug serial interface on pin 1.

Programming should work like this: 
Connect GND, +3.3V VCC, Rx, Tx and tie GPIO0 to GND. Then power up and release GPIO0 to set the chip to programming mode. 
Start flashing.
The button should not be required.

Noschvie, 2023-04-08:
The link above seems to be broken
[Shelly pro serial interface](https://kb.shelly.cloud/__attachments/35094529/Plus-addon-interface.png?inst-v=fea39ce4-70c1-4ccc-806d-4330adcf4836)

Gemikro1, 2023-04-09:
Sorry, my mistake, here again: 
https://kb.shelly.cloud/knowledge-base/shelly-plus-2pm

stefanbode, 2023-04-12:
Thx for your support. It now works with ota and wired. The combination was the solution. Also the reset pin shortly to ground is very handy

Gemikro1, 2023-04-12:
Good to hear ! I just received my Shelly Plus 2PM, so i will also try flashing next weekend...

stefanbode, 2023-04-13:
If you want to use shutterbuttons with the plus 2pm you need a different template. Switch-n must be replaced by button-n. Autosetup is working. See discussion on autosetup.