arendst/Tasmota #18208: FritzBox NTP server crashes Tasmota

johanneswen99, 2023-03-16:
Hello everybody,

I have experienced the following weird behavior. My Tasmota devices had been rebooting frequently with Restart Reason _Exception_. During 24h every device would accumulate around 70-80 reboots, i.e. around every 20 min. One of the devices has accumulated >6.000 reboots over a few months. 

SOLUTION: After lots and lots of trying out and ruling out possible sources of error I found that when I disable on my FritzBox router the option ["Make FRITZ!Box available as a time server in the home network"](https://en.avm.de/service/knowledge-base/dok/FRITZ-Box-3272-int/336_Configuring-time-synchronization-NTP-for-FRITZ-Box-and-network-devices/) then the reboots stop and the Tasmota devices run for hours without crashing, which I can read from the _Uptime_ field in the web interface.

The issue seems to persist through multiple Tasmota versions and devices. I have tried with flashing Tasmota 12.3.1 and 9.3.1. I have used a Teckin SP22 Smart Plug (configured as BlitzWolf SHP), and five Gosund SP1 (configured as Gosund SP1 v23). All devices and firmwares behave the same. The FritzBox is 7360 with FRITZ!OS 06.33, quite an old device I use as DSL modem and DHCP server in my network.

Can you check what is the reason for this problem?

arendst, 2023-03-17:
Try latest dev branch.

Jason2866, 2023-03-17:
What exactly! have you entered as NTP server in Tasmota for using the Fritzbox as NTP server?

johanneswen99, 2023-03-17:
Hey, thanks for the replies and recommendations. I will check on the weekend if the error appears with the newest dev branch. 

Regarding the NTP server on the Tasmota I use the firmware defaults on all devices for NtpServer1 to NtpServer3. 

The FritzBox provides, when activated, a time server on it's local domain "fritz.box". Which is funny, because that NTP server is not being used by the Tasmotas, and still crashes them.

Jason2866, 2023-03-17:
What happens if you enter as NTP server in Tasmota `fritz.box`?

johanneswen99, 2023-03-18:
I ran tonight an experiment with my five smart plugs. Three of them flashed with Tasmota 12.3.1, two with 12.4.0.2. All of them using the default NTP servers, only one of the latter using 
```
CMD: ntpserver
RESULT = {"NtpServer1":"fritz.box","NtpServer2":"","NtpServer3":""}
```
Funnily none of them crashed that night. Then I rebooted my FritzBox, warm-rebooted all Tasmotas and waited. After a few minutes **all five of them** crashed and restarted with an _Exception_.

Also I wrote a small NTP client python script [ntp.py](https://github.com/arendst/Tasmota/files/11007836/ntp.py.txt), the output is as follows:

```
C:\Users\Johannes>python ntp.py pool.ntp.org
Using server: pool.ntp.org
Offset:  79.85665464401245
Version: 3
Time:    Sat Mar 18 08:33:58 2023
Leap:    no warning
Delay:   0.0107574462890625
IP:      116.203.151.74

C:\Users\Johannes>python ntp.py ptbtime1.ptb.de
Using server: ptbtime1.ptb.de
Offset:  79.86022710800171
Version: 3
Time:    Sat Mar 18 08:34:09 2023
Leap:    no warning
Delay:   1.52587890625e-05
IP:      80.84.66.0

C:\Users\Johannes>python ntp.py fritz.box
Using server: fritz.box
Offset:  79.7808256149292
Version: 3
Time:    Sat Mar 18 08:34:15 2023
Leap:    unknown (clock unsynchronized)
Delay:   1.0
IP:      0.0.0.0
```

So I guess that old Fritzbox is doing some weird stuff and the Tasmota are reacting to it, even when not configured to use the Fritzbox as time server. 

Also I cannot explain why they didn't crash during the night. My guess would be that the issue is somehow related to initialization of the NTP client on the Tasmotas.

Jason2866, 2023-03-18:
The log shows the NTP server of the Fritzbox is not working. Probably the Fritzbox NTP will always be used when activated. Tasmota will not work when no NTP is reachable.
The not valid IP address from the Fritzbox let Tasmota crash. Solution (as you already did) switch off the non working NTP server in Fritzbox.

johanneswen99, 2023-03-18:
Yes, that sounds reasonable. Let's go with that.

closed 2023-03-18 by johanneswen99