arendst/Tasmota #18222: Weird serial output using clean Tasmota firmware on ESP16E

Gemikro1, 2023-03-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I flashed a fresh built Tasmota image with no modifications on a custom made device equipped with an generic ESP-16e using a serial programmer (erasing all flash contents before). 
The firmware boots and i can see the boot message at 74880 baud:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld
```
followed at 115200 baud by
```
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F6, Count 14
00:00:00.055 SER: Set to 5N1 115200 bit/s
```
followed by only garbage
```
0L0L0ÌÁµÅØ¤ ôÕ±K{FF}0L0L0ÌÁ7ÁÙ©¥D -Í­Ñ!ÔZ4­Ñ!d%6©¹ &.ÍÁ®†Ñhk/-¥-ö·´9J06-l­,´Ìº-ºì©K{FF}0L0L0ÌÕ6ÝÊÒ ô¹W´Ê {1H!Ë#¹Wb l 4³Š%¦2N6I·¦·M£¥.Ô½R±Ì¡Ñhk/-µ84™–­†¹s.µê0L0L°æÝ K&¹®´‰µê0L0L0ÍÁ1¡¦ +"h%f%…D)+ ¹ ³ëÑ¡á¢dÍ®Í¸w K4
¥0‘’V³Å˜Á.æÍkê±Ó²¦²ÍÁ±­L:(Ñ´j0-¹'4½—•#K/¹.µê±Ó²¦²ÍÉ¹µŒÓ Í ô¹W´ê¹ ’²É:d²KÅ7'3'5Í°µŒÓ ô¹W4È…)•$D½°¥83É#ÔÉ$%M9(¹  VK{FF}±Ó²¦2s4m #Éi+´K{FF}±Ó²¦3s°V +4…+-Œ) ô¹W´ê¹.¹-ý±Ó²¦3s3& +4ÑÖ£¹®´ê¹ ’²É:d²
```
What causes the serial interface be set to `5N1` ?
I tried to decode this garbage at all resonable baudrates, even using the strange indicated 5N1 but no success.
I can connect to the WifiManager and bring the device into my network as on any other of my Tasmota devices but serial output stays garbage.
Even an OTA update to the standard precomiled Tasmota firmware image gives the same results.
Is there something broken in the serial interface ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 12.4.0.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Serial Programmer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:18:18.377 CMD: Status0
18:18:18.405 RSL: STATUS0 = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_8BCF39","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"5N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:12:08","StartupUTC":"2023-03-19T17:06:10","Sleep":50,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2023-03-19T17:03:22","SaveCount":44,"SaveAddress":"F9000"},"StatusFWR":{"Version":"12.4.0(tasmota)","BuildDateTime":"2023-02-24T15:32:38","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RALF",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00206000","00044000","00000000"]},"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"tasmotaxxxxxx","IPAddress":"xxxxxxxx","Gateway":"xxxxxxxx","Subnetmask":"255.255.255.0","DNSServer1":"xxxxxxxxx","DNSServer2":"0.0.0.0","Mac":"xxxxxxxxxx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":4.8},"StatusMQT":{"MqttHost":"HoPi","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"xxxxxx","MqttUser":"xxxxx","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-03-19T17:18:18","Local":"2023-03-19T18:18:18","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:55","Sunset":"19:00"},"StatusSNS":{"Time":"2023-03-19T18:18:18"},"StatusSTS":{"Time":"2023-03-19T18:18:18","Uptime":"0T00:12:08","UptimeSec":728,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"RALF","BSSId":"E0:28:6D:73:75:CE","Channel":13,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash Tasmota, boot device and listen on serial ...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Readable output at 115200/8N1 on the serial interface ?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2023-03-19:
`SER: Set to 5N1 115200 bit/s` looks like you did some configuration around using serial, and `5N1` is not useful for getting expected output. Maybe try with a standard precompiled build, instead of what you mixed yourself.....

Gemikro1, 2023-03-19:
That is just the issue, i used the pristine downloaded Tasmota 12.4.0.1, just compiled it without any modifications ...
I _did_ want to add some modifications once the pristine image works on the device but did not change anything yet.

sfromis, 2023-03-19:
Did you try what I suggested, using a precompiled image?

Gemikro1, 2023-03-19:
Yes, uploaded the standard Tasmota via OTA from the server (via the minimal image), same issue here with the serial garbage.