arendst/Tasmota #18229: Haspmota switch widget causes bad json event changed on touch

Silverstar, 2023-03-20:
### Discussed in https://github.com/arendst/Tasmota/discussions/18203

Lanbon L8, flashed with different tasmota32-lvgl versions, 12.3.1, 12.4.0, 12.4.0.3 dev and self compiled dev (to enable PSRAM and tried to fix this bug myself, no dice).
I created a switch in the pages.jsonl and added val_rule:power1#state to it, so the switch follows if the relay is toggled.

When I touch the switch and it toggles its state, an event changed will be triggered, but in the console (weblog 4) the following error occurs:
`BRY: ERROR, bad json: {"hasp":{"p1b1":{"event":"changed","val":,"text":"Relay 1"}}}`
"val" is empty.

simple pages.jsonl:
```
{ "page": 0, "id": 0, "bg_color": "#000000", "text_color": "#FFFFFF" }

{ "page": 1, "id": 0, "bg_color": "#AAAAAA", "text_color": "#000000" }
{ "page": 1, "id": 1, "obj":"switch", "x": 37,  "y": 120, "w": 120, "h": 28, "text":"Relay 1", "val_rule":"POWER1#State", "bg_color":"#1fa3ec"}
```

I researched for the relevant source and found [haspmota.be](https://github.com/arendst/Tasmota/blob/development/tasmota/berry/haspmota_src/haspmota_core/haspmota.be#L345).
I tried to fix it myself, but no matter what I change (and do gen_haspmota.sh afterwards), there is no change in behavior on the device.

As advised in the discussion, I tried to work with a haspmota.be in the filesystem rather than compiling the whole image.
Creating a new haspmota.be on device with own changes in the file and performing `load('haspmota.be')` returns true, but no change in behavior.
`import haspmota` thereafter, no change in behavior.
`haspmota.start()` thereafter, no change in behavior.
No matter if I delete haspmota.bec or not, no matter what I change in file, it is ignored.

Trying the solidify procedure in the header of the file, error.
```
import path 
path.remove("haspmota.bec")
import solidify
var haspmota
load('haspmota.be')
BRY: Exception> 'import_error' - module 'solidify' not found
stack traceback:
	<unknown source>: in function `main`
```
Tried many things for many hours, wanted to fix it myself and push a PR, but now I am at my wits' end.

stefanbode, 2023-06-03:
The main reason is that the be files need a special "solidify" to get active in the firmware. This was solved into the automation just a week ago. Please submit the fixed be file and we will add it to the source. Then with the next version it should work. There is also a manual way to do it depending on the operating system you do the compile. I had this learning in fixing the shutter matter Support in be.

s-hadinger, 2023-06-03:
I will do it. HASPmota solidification is more complex

closed 2023-06-04 by s-hadinger

s-hadinger, 2023-06-04:
Fixed

PR #18791 (2023-06-04): Fixed HASPmota event when value is non-integer (fixes #18229)
## Description:

`val` attribute is now correctly escaped to JSON when non-integer.

**Related issue (if applicable):** fixes #18229

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.9
  - [x] I accept the [CLA](htt [... cut]
commit 844caf51 (2023-06-04): Fixed HASPmota event when value is non-integer (fixes #18229) (#18791)