arendst/Tasmota #18232: Tasmota v12.4.0.3 - 2x PCF8574 Mode2 problem

jpier123, 2023-03-21:
### PROBLEM DESCRIPTION
Dear Tasmota Team,

I'm facing a problem with most recently developped build v 12.4.0.3 where PCF8574 mode2 was implemented.  I have two PCF8574 (adress: 0x21, 0x22) modules added and configured with 4 Inputs and 4 outputs each. First auxliary output/relay starts with No 11.

PCF8574 #1: Power 11-14
PCF8574 #2: Power 15-18

**Mode 2** 

SetOption81 = 1

A= B1-4, Ri1-4 aka Power 11-14
B= B5-8, Ri5-8 aka  Power 15-18

**1st tested (JSON) configuration file** (pcf8574.dat): {"NAME":"PCF8574","GPIO":[32,33,34,35,256,257,258,259,36,260,37,261,38,262,39,263]}

**Symptopms:**

- All AUX relays staring from No 11 are fully functional and can be manually toggled via WEB Gui
- None from 8 Buttion inputs do the job. No Debug console events

Note. There is no other native ESP32 board GPIOs configured as Button in the configuration.

_**For reference - Mode 1 (GUI) configuration:**_ 

Invert Ports: Yes

**PCF8574 0x21**
1. Device 1 Port 0 = Input
2. Device 1 Port 1 = Input
3. Device 1 Port 2 = Input
4. Device 1 Port 3 = Input
5. Device 1 Port 4 = Output
6. Device 1 Port 5 = Output
7. Device 1 Port 6 = Output
8. Device 1 Port 7 = Output

**PCF8574 0x22**
1. Device 2 Port 0 = Input
2. Device 2 Port 1 = Output
3. Device 2 Port 2 = Input
4. Device 2 Port 3 = Output
5. Device 2 Port 4 = Input
6. Device 2 Port 5 = Output
7. Device 2 Port 6 = Input
8. Device 2 Port 7 = Output

Note. Functionality of all listed above Inputs/Outpus was confirmd using Web GUI.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32D DEV v4 
- [x] Tasmota binary firmware version number used: 12.4.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESP Flash
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
[... 211 lines cut ...]
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

**Expected behaviour/configuration:**

A= B11-14, Ri11-14
B= B15-18, Ri15-18

**2nd tested (JSON) configuration file** (pcf8574.dat): {"NAME":"PCF8574","GPIO":[42,43,44,45,256,257,258,259,46,260,47,261,48,262,49,263]}

**Symptopms:**

- All AUX relays staring from No 11 are fully functional and can be manually toggled via WEB Gui
- None from 8 Buttion inputs do the job. No Debug console events

### SCREENSHOTS

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jpier123, 2023-03-24:
Hi,

As requesdted, I have tested v12.4.0.4 dev build with below JSON:

{"NAME":"PCF8574","BASE":10,"GPIO":[202,203,204,205,256,257,258,259,206,260,207,261,208,262,209,263]}

**PCF8574 0x21**

    Device 1 Port 0 = Switch 11 (202)
    Device 1 Port 1 = Switch 12 (203)
    Device 1 Port 2 = Switch 13 (204)
    Device 1 Port 3 = Switch 14 (205)
    Device 1 Port 4 = Relay 11 (256)
    Device 1 Port 5 = Relay 12 (257)
    Device 1 Port 6 = Relay 13 (258)
    Device 1 Port 7 = Relay 14 (259)

**PCF8574 0x22**

    Device 2 Port 0 = Switch 15 (206)
    Device 2 Port 1 = Relay 15 (260)
    Device 2 Port 2 = Switch 16 (207)
    Device 2 Port 3 = Relay 16 (261)
    Device 2 Port 4 = Switch 17 (208)
    Device 2 Port 5 = Relay 17 (262)
    Device 2 Port 6 = Switch 18 (209)
    Device 2 Port 7 = Relay 18 (263)

I decided to use switches configured as "SwitchMode 2" (Set switch to inverted follow mode, 0 = ON, 1 = OFF) for beter relay status verification.

**Testing results:**
Four PCF8574 0x21 configured Relays (no 11-14) are working as expected following switch (no 11-14) status.

Next four PCF8574 0x22 attached relays do not work as expected. Only Relay no 18 (263) [very last pair from JSON string] follows status of the switch (209). Other three switches 15-17 (206, 207,208) do not control any relays. I noticed the same anomaly when testied previous dev version. 

Please also note that all those relays (11-14 & 15-18) still can be fully controlled via the Web GUI, so that part of the JSON  works fine (even if presented as not in the sequence).

jpier123, 2023-03-24:
**Test extention with relays and switched configured accross two deferent PCF8574 expanders.**

Tested also below JSON just to verify if second PCF8574 expander Relays can be controlled by first PCF8574 attached swtches.

{"NAME":"PCF8574","BASE":10,"GPIO":[206,207,208,209,256,257,258,259,202,260,203,261,204,262,205,263]}

The result was the same. 

Switches 11-14 did control relays 11-14 even if configured between two expanders. Unfortuinatly, the same as previously noticed only switch No 18 could control relay 18.

arendst, 2023-03-25:
I did some tests too using a config with two relays configured on the esp8266 and 8 relays configured on two PCF8574 devices:
```
rule3 on file#pcf8574.dat do {"NAME":"PCF8574 A=B3456,Ri1234,B=B7Ri5B8Ri6B9Ri7B10Ri8","BASE":1,"GPIO":[34,35,36,37,256,257,258,259,38,260,39,261,40,262,41,263]} endon
```
This works as expected; every button controls it's corresponding relay.

Then changed the buttons to switches:
```
rule3 on file#pcf8574.dat do {"NAME":"PCF8574 A=S3456,Ri1234,B=S7Ri5S8Ri6S9Ri7S10Ri8","BASE":1,"GPIO":[194,195,196,197,256,257,258,259,198,260,199,261,200,262,201,263]} endon
```
and configured ``SwitchMode2 2`` to ``SwitchMode10 2`` to have the same situation as you. Again this works as expected; every switch controls it's corresponding relay momentary when pushed.

So I cannot reproduce your behaviour. Are you sure you've set all your switch modes to 2? Verify with output of command ``Status``

arendst, 2023-03-25:
BTW latest dev branch now supports command ``SwitchMode0 2`` to set all switchmodes at once. Use ``SwitchMode0`` to show current states.

jpier123, 2023-03-25:
Hi,

I decided factory reset ESP32, remove all GPIO Relays and just deal with virtual ones. And again, first four do the job 100%, next four can be switched ON once after full power cicle. 

And, the same as previously, very last switch no 8 can control it own relay without any issues.  The rest will just stuck ON state. I can toggle them OFF using Web GUI, but after that Switch 5-7 can't be switched again until next power cycle. Normal "Restart" doesn't allow switch them ON again. It must be full ESP32 power cycle.

It is so wired.

If you take closer look, you will find that from some reasons Switch 5-7 are always ON, where all other switches status can be changed. I have verified those three inputs control using a GUI, see attached images.

Updared DAT file:
{"NAME":"PCF8574","BASE":1,"GPIO":[192,193,194,195,256,257,258,259,196,260,197,261,198,262,199,263]}

The console indicates status of the switches with DAT file loaded (idle state):

SENSOR = {"Time":"2023-03-25T13:22:32","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"OFF"

 SwitchMode0 output: 
{"SwitchMode":[2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2,2]}

Template:

{"NAME":"ESP32-JP_SMDv1","GPIO":[1,1,1,1,0,0,1,1,0,0,0,0,0,0,0,0,0,640,608,0,0,1,0,0,0,0,0,0,0,1312,0,0,0,0,0,0],"FLAG":0,"BASE":1}

i2cscan
 RESULT = {"I2CScan":"Device(s) found on bus1 at 0x21 0x22 0x27 0x57 0x5f 0x68"}

restart 1

00:00:00.002 HDW: ESP32-D0WD-V3 
00:00:00.120 UFS: FlashFS mounted with 304 kB free
00:00:00.163 CFG: Loaded from File, Count 110
00:00:00.174 I2C: DS3231 found at 0x68
13:26:58.007 I2C: PCF8574 found at 0x21
13:26:58.007 I2C: PCF8574 found at 0x22
13:26:58.084 TFS: File 'mcp23x.dat' not found
13:26:58.437 BRY: Berry initialized, RAM used=3850 bytes
13:26:58.468 TFS: File '.drvset003' not found
13:26:58.469 CFG: Energy use defaults as file system not ready or file not found
13:26:58.470 I2C: LCD found at 0x27
13:26:58.573 DSP: LCD
13:26:58.611 Project Custom_Tasmota_v12.4.3 - ESP_SMD Version 12.4.0.4(tasmota32)-2_0_7(2023-03-25T12:28:28)
13:26:58.629 RSL: STATE = {"Time":"2023-03-25T13:26:58","Uptime":"0T00:00:00","UptimeSec":0,"Heap":164,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":0,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":38},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF"}
13:26:58.653 RSL: RESULT = {"POWER1":"OFF"}
13:26:58.654 RSL: POWER1 = OFF
13:26:59.650 BRY: Successfully loaded 'autoexec.be'
13:27:02.530 WIF: Connecting to AP1 SS02 Channel 3 BSSId 08:CC:68:AD:C1:90 in mode 11n as Custom-Tasmota-v12.4.3-EB1984-6...
13:27:04.082 WIF: Connected
13:27:04.482 HTP: Web server active on Custom-Tasmota-v12.4.3-EB1984-6 with IP address 192.168.11.54
13:27:06.564 MQT: Attempting connection...
13:27:07.577 MQT: Connect failed to 192.168.11.2:1883, rc -2. Retry in 10 sec
13:27:18.708 MQT: Attempting connection...
13:27:19.720 MQT: Connect failed to 192.168.11.2:1883, rc -2. Retry in 20 sec
13:27:40.628 MQT: Attempting connection...
13:27:41.639 MQT: Connect failed to 192.168.11.2:1883, rc -2. Retry in 30 sec
13:28:06.690 RSL: STATE = {"Time":"2023-03-25T13:28:06","Uptime":"0T00:01:08","UptimeSec":68,"Heap":107,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":57},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","Wifi":{"AP":1,"SSId":"SS02","BSSId":"08:CC:68:AD:C1:90","Channel":3,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:28:06.768 RSL: SENSOR = {"Time":"2023-03-25T13:28:06","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"OFF","DS18B20-1":{"Id":"3C51F649B890","Temperature":19.8},"DS18B20-2":{"Id":"3CA0F649311A","Temperature":21.8},"DS18B20-3":{"Id":"007EED0264FF","Temperature":21.9},"Thermostat0":{"ThermostatModeSet":0,"TempTargetSet":22.50,"CtrDutyCycleRead":0,"ControlMethod":1},"TempUnit":"C"}
13:28:12.553 MQT: Attempting connection...
13:28:13.563 MQT: Connect failed to 192.168.11.2:1883, rc -2. Retry in 40 sec

Web GUI status with no DAT file enabled:

[image]
[image]
[image]

Question: 

From what I can see "BASE":x value does not reflects virtual Switch/Button offset, right?  Example: "BASE":10 = first virtual switch = 11

arendst, 2023-03-25:
> And, the same as previously, very last switch no 8 can control it own relay without any issues. The rest will just stuck ON state. I can toggle them OFF using Web GUI, but after that Switch 5-7 [...]

I think a typical case of lack of pullup hardware. The PCF8574 has very limited pullup capability. In my case it works on all inputs OK. In your case it might fail without adding external pullup resistors. Try adding 10k resistors to 3V3 on the ports that fail and see what happens.

> From what I can see "BASE":x value does not reflects virtual Switch/Button offset, right? Example: "BASE":10 = first virtual switch = 11

Correct. It's a flag either 0 for default behaviour meaning adding buttons and switches to the already available buttons and switches OR 1 (or higher) for the new behaviour where all buttons and switches need to have a unique number on both ESP32 AND PCF8574.

jpier123, 2023-03-25:
It looks like those three switches are latched aftre activated, see below the console status:

14:00:34.488 RSL: SENSOR = {"Time":"2023-03-25T14:00:34","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF",**_"Switch5":"ON","Switch6":"ON","Switch7":"ON"_**,"Switch8":"OFF"

I can reset them only by ESP32 power cycle, see status after:
14:03:05.504 RSL: SENSOR = {"Time":"2023-03-25T14:03:05","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF",**_"Switch5":"OFF","Switch6":"OFF","Switch7":"OFF"_**,"Switch8":"OFF",

Example: 

Switch no 5 momentary press results in the latch:

14:05:52.499 RSL: SENSOR = {"Time":"2023-03-25T14:05:52","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF","Switch5":"ON","Switch6":"OFF","Switch7":"OFF","Switch8":"OFF",

arendst, 2023-03-25:
BTW how did you wire switches 5 to 7?

Normally the input is pulled to Gnd by a button or switch.

jpier123, 2023-03-25:
It is wierd, as all inputs were wired the same way. I have used 100k acting as external pullup resistors, see below:

[image]

jpier123, 2023-03-25:
Possibly those 10k (R6-R9) are simply to big (100 ohm) or not necessery.

arendst, 2023-03-25:
Are C4 to C7 connected to Gnd or 5V?

jpier123, 2023-03-25:
GND

arendst, 2023-03-25:
OK. And if they are 1microFarad and 100k resistor to 5V I think you have a big tau value; 

I normally see much smaller values for the capacitor. The 10k in series with the switch is probably too big to lower the voltage of the input below trigger level.

For a test just short the 10k resistor.

jpier123, 2023-03-25:
I have checked all four circurs and did not find any issues. Without DAT file enabled, all Low/High switch status voltages are the same accros all input pins. Situation does change as soon DAT is enabled. Only last set of switch/relay works as expected.

I can confirm that latch efefct must be DAT file related. Looks like switch input triggers itself (Low state) instead the relay and input stays Low state until PCF chip reset..

Final confirmation was just to use short JSON and work with firts PCF expander, see below:
{"NAME":"PCF8574","BASE":1,"GPIO":[192,193,194,195,256,257,258,259]}

I measured secondary expander input voltages (with button presset and without) at every input pin and did not see latch effect any more.

arendst, 2023-03-25:
As it works fine in my setup I think we can conclude it's related to both DAT file AND your unlucky selection of 100k/1uF/10k RC network.

What happens is that the DAT code needs to reset Interrupt state of inputs. The PCF8574 quasi-bidirectional hardware is playing up here. Resetting the interrupt needs a write 1 to the pin that is used as an input.

From the datasheet:

> Input HIGH: The master needs to write 1 to the register to set the port as an input mode
> if the device is not in the default power-on condition. The master reads the register to
> check the input status. If the external source pulls the port pin up to VDD or drives
> logic 1, then the master will read the value of 1.
> Input LOW: The master needs to write 1 to the register to set the port to input mode if
> the device is not in the default power-on condition. The master reads the register to
> check the input status. If the external source pulls the port pin down to VSS or drives
> logic 0, which sinks the weak 100 uA current source, then the master will read the value
> of 0.

As shown; the code must write a 1 first before being able to read the input correctly. This writing, in addition to your RC network will always result in reading a 1.

jpier123, 2023-03-25:
Looks like switch input triggers itself (enforceing input the Low state) _and_ relewanty relay at the same time. You can restore relay state using Web GIU, but switch state stays ON (0V). It stays Low until PCF chip reset.

arendst, 2023-03-25:
> Without DAT file enabled, all Low/High switch status voltages are the same accros all input pins

Can you show what config you used if not using the DAT file.

Can you show a picture of the PCF8574 and the connections to the buttons and the leds?

jpier123, 2023-03-25:
Above will make perfect sense if:

- ESP32 did not report all releant aux inputs/outputs correctly without DAT file. 
- one switch/relay pair did not work as expected.

The wireing I have does not explain input's pin low state, especially after switch/button restore. There is nothink that could enforce 0V at that point, it must be PCF pin itself.  At that point, the input was nominated to act as output. 

Additional test with JSON switch/relay order change did show the same input behaviour but this time effected were switch inputs 6-8 and not like previously 5-7.

arendst, 2023-03-25:
As long as I cannot reproduce the bus stops here.

jpier123, 2023-03-25:
As requested, please find below no DAT file configuration screenshots:

[image]

[image]

[image]

jpier123, 2023-03-25:
[image]

The same as relays, all above inputs were Web GUI tested and did report Low/High state without any issues.

jpier123, 2023-03-25:
Can you just share your test schematic then?

arendst, 2023-03-25:
[image]

jpier123, 2023-03-25:
OK, I will try expain again, as I can see some enforced theory that does not fit here. 

Electricly, 2nd PFC expander configuration does the job (0V = ON, 5V=OFF). I can get working any from four switch/relay pairs until their IDs were placed at end of the JSON string. 

I have done four tests starting with first switch/relay pair only and then expanded the JSON with next in the line pair. Resuls are listed below:

- 1 pair

1th pair: **196,260** - act as expected

{"NAME":"PCF8574","BASE":1,"GPIO":[192,193,194,195,256,257,258,259,**196,260**]}
Issues: none

- 2 pairs

2nd pair: **197,261** - act as expected
{"NAME":"PCF8574","BASE":1,"GPIO":[192,193,194,195,256,257,258,259,196,260,**197,261**]}
Issues: 
1th pair: **196,260** latch itself. Relay 5 goes ON and stays ON.

- 3 pairs

3rd pair: 198,262 - act as expected
{"NAME":"PCF8574","BASE":1,"GPIO":[192,193,194,195,256,257,258,259,196,260,197,261,**198,262**]}
Issues: 
1th pair: **196,260** latch itself. Relay 5 goes ON and stays ON,
2nd pair: **197,261** latch itself. Relay 6 goes ON and stays ON.

- 4 pairs

4th pair: 199,263 - act as expected
{"NAME":"PCF8574","BASE":1,"GPIO":[192,193,194,195,256,257,258,259,196,260,197,261,198,262,**199,263**]}
Issues: 
1th pair: **196,260** latch itself. Relay 5 goes ON and stays ON,
2nd pair: **197,261** latch itself. Relay 6 goes ON and stays ON,
3rd pair: **198,262** latch itself. Relay 7 goes ON and stays ON.

All above just proves that electricly the wiring is right and all four pairs can function correctly, but not at the same time. Always only very last JSON placed pair works. Other haveing some latching probelms.

jpier123, 2023-03-25:
I'm happy host some remote troubleshooting sesion via Teams,if you want find extra time.

I appreciate your effort and would like help get the project moveing.

Thanks in advance.

J.P.

arendst, 2023-03-26:
> All above just proves that electricly the wiring is right and all four pairs can function correctly, but not at the same time. Always only very last JSON placed pair works. Other haveing some latching probelms.

Indeed nice find. But I still cannot explain why I cannot reproduce it. How is your I2C bus terminated? I noticed you use BMP280 too. Is this on the same I2C bus or on the second bus?

BTW today I changed the BASE 1 mode once again which now needs to have the relay addresses also unique with the main template. So in your case where you have relay1 to 10 in the main template you'll need to change the PCF relay numbers to relay11 to 18.

arendst, 2023-03-26:
If you enable ``seriallog 3`` (and ``SO90 1`` to disable non-JSON message) and press a button of a set that works once what does the log look like? Expect something like:

```
00:00:00.001 HDW: ESP8266EX
00:00:00.047-034 CFG: Loaded from flash at FB, Count 353
00:00:00.131-034 QPC: Reset
00:00:00.136-034 I2C: PCF8574 found at 0x20
00:00:00.137-034 I2C: PCF8574 found at 0x21
00:00:00.140-034 PCF: Base 1, Template 'PCF8574'
00:00:00.148-034 SWT: Add vSwitch2, State 1
00:00:00.149-034 SWT: Add vSwitch4, State 1
00:00:00.150-034 SWT: Add vSwitch5, State 1
00:00:00.151-034 SWT: Add vSwitch6, State 1
00:00:00.154-034 SWT: Add vSwitch7, State 1
00:00:00.158-034 SWT: Add vSwitch8, State 1
00:00:00.161-034 SWT: Add vSwitch9, State 1
00:00:00.165-034 SWT: Add vSwitch10, State 1
00:00:00.171-034 SRC: Restart
00:00:00.177-034 Project tasmota - Wemos4 Version 12.4.0.4(theodev)-2_7_4_9(2023-03-26T12:11:25)
00:00:00.188-034 I2C: BMP280 found at 0x76
00:00:00.497-034 WIF: Attempting connection...
00:00:00.833-034 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as wemos4...
00:00:01.806-034 WIF: Connected
00:00:02.008-031 HTP: Web server active on wemos4 with IP address 192.168.2.223
00:00:02.415-031 WIF: DNS resolved 'loghost' (192.168.2.24) in 8 ms
00:00:02.438-031 WIF: DNS resolved 'ntpserver1' (192.168.2.27) in 4 ms
00:00:02.452-031 RTC: UTC 2023-03-26T10:24:34, DST 2023-03-26T02:00:00, STD 2023-10-29T03:00:00
12:24:34.000-031 RTC: Synced by NTP
12:24:35.012-031 MQT: Attempting connection...
12:24:35.019-031 WIF: DNS resolved 'mqttbroker' (192.168.2.24) in 5 ms
12:24:35.029-031 MQT: Connected
12:24:35.032-028 MQT: tele/wemos4/LWT = Online (retained)
12:24:35.034-028 MQT: Subscribe to cmnd/wemos4/#
12:24:35.037-029 MQT: Subscribe to cmnd/tasmotas/#
12:24:35.041-029 MQT: Subscribe to cmnd/theotest1/#
12:24:35.043-028 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
12:24:35.052-029 MQT: tele/wemos4/INFO1 = {"Info1":{"Module":"Generic","Version":"12.4.0.4(theodev)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}}
12:24:35.064-029 MQT: tele/wemos4/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}}
12:24:35.074-027 MQT: tele/wemos4/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":46}}
12:24:35.083-027 MQT: stat/wemos4/RESULT = {"POWER1":"OFF"}
12:24:35.089-027 MQT: stat/wemos4/RESULT = {"POWER2":"OFF"}
12:24:35.094-027 MQT: stat/wemos4/RESULT = {"POWER3":"OFF"}
12:24:35.100-027 MQT: stat/wemos4/RESULT = {"POWER4":"OFF"}
12:24:35.106-027 MQT: stat/wemos4/RESULT = {"POWER5":"OFF"}
12:24:35.109-027 MQT: stat/wemos4/RESULT = {"POWER6":"OFF"}
12:24:35.116-028 MQT: stat/wemos4/RESULT = {"POWER7":"OFF"}
12:24:35.122-028 MQT: stat/wemos4/RESULT = {"POWER8":"OFF"}
12:24:35.127-028 MQT: stat/wemos4/RESULT = {"POWER9":"OFF"}
12:24:35.132-028 MQT: stat/wemos4/RESULT = {"POWER10":"OFF"}
12:24:39.400-026 MQT: tele/wemos4/STATE = {"Time":"2023-03-26T12:24:39","Uptime":"0T00:00:09","UptimeSec":9,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","POWER10":"OFF","Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:24:39.442-028 MQT: tele/wemos4/SENSOR = {"Time":"2023-03-26T12:24:39","Switch1":"OFF","Switch2":"OFF","Switch4":"OFF","Switch5":"OFF","Switch6":"OFF","Switch7":"OFF","Switch8":"OFF","Switch9":"OFF","Switch10":"OFF","BMP280":{"Temperature":22.3,"Pressure":1005.0},"PressureUnit":"hPa","TempUnit":"C"}
12:24:40.366-030 APP: Boot Count 46
12:24:40.614-030 CFG: Saved to flash at FA, Count 354, Bytes 4096
12:24:56.571-029 SRC: Switch
12:24:56.580-026 MQT: stat/wemos4/RESULT = {"POWER7":"ON"}
12:24:57.656-029 CFG: Saved to flash at F9, Count 355, Bytes 4096
12:24:57.663-029 SRC: Switch
12:24:57.671-026 MQT: stat/wemos4/RESULT = {"POWER7":"OFF"}
12:24:58.617-029 CFG: Saved to flash at F8, Count 356, Bytes 4096
```

arendst, 2023-03-26:
You might want to test the attached replacement for ``xdrv_28_pcf8574_v2.ino``.

[xdrv_28_pcf8574_v2.zip](https://github.com/arendst/Tasmota/files/11071350/xdrv_28_pcf8574_v2.zip)

It might solve your issue.

arendst, 2023-03-26:
I think I see what goes wrong. The zip solves part of it. I need to rewrite the code though.....

jpier123, 2023-03-26:
Hi,

You did it!!

All virtual switches are working as expected. 

Innitially tested SON .dat file: 
{"NAME":"PCF8574","BASE":1,"GPIO":[192,193,194,195,256,257,258,259,196,260,197,261,198,262,199,263]}

Also sucesfully tested previous 10+ relay configuration version and latest "BASE" aproach.

A = Sn11-14, Ri11-14, B = Sn15Ri15 Sn16Ri16 Sn17Ri17 Sn18Ri18
{"NAME":"PCF8574","BASE":1,"GPIO":[202,203,204,205,266,267,268,269,206,270,207,271,208,272,209,273]}

Configuration template: {"NAME":"ESP32-JP_SMDv1","GPIO":[1,1,1,1,230,227,1,1,480,1056,234,231,229,228,226,225,0,640,608,224,0,1,416,233,0,0,0,0,167,1312,352,353,32,0,0,33],"FLAG":0,"BASE":1}

I managed to switch ON/OFF all PCF based relays 11-14 & 15-18 with no porblem.

Now will just need check Button mode instead Switch type. 

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
Once again, Thanks for you time and efford. I owe you, so please contact me in the future if you need any assistance with some testing.

Best Regards,

JP

arendst, 2023-03-26:
Glad to hear. With the zip there is an issue at restart; you'll notice leds/relays flashing which were ON or OFF before the restart. I need to fix that.

jpier123, 2023-03-26:
You are right, last four PFC Relays flash at the restart. 

Once again. I'm happy to help, so let me know if any more beta testing would be required.

Buttons 11-14 & 15-18 sucesfully tested with 10+ relay configuration version.

A = Bn11-14, Ri11-14, B = Bn15Ri15 Bn16Ri16 Bn17Ri17 Bn18Ri18
{"NAME":"PCF8574","BASE":1,"GPIO":[74,75,76,77,266,267,268,269,78,270,79,271,80,272,81,273]}

arendst, 2023-03-26:
Pls try this

[xdrv_28_pcf8574_v2.zip](https://github.com/arendst/Tasmota/files/11071576/xdrv_28_pcf8574_v2.zip)

It should fix the blinks at restart.

jpier123, 2023-03-26:
It looks good... the flash is gone.

Tested also last saved state mode (PowerOnState 3) - OK

arendst, 2023-03-26:
Thx. Will make some more tweaks and merge in dev branch soon.

Jason2866, 2023-03-27:
Closing, since fixed.