arendst/Tasmota #18234: MQTT Power State of Channel not Updating with SetOption68 enabled

moesfeld, 2023-03-21:
### PROBLEM DESCRIPTION
When using a multichannel LED-Controller (i.e. LS5P-TAS) with setoption68 ON the Powerstate in MQTT unter stat ist not updated when switching on the channel by setting its value.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): athom_LS5P-TAS
- [x] Tasmota binary firmware version number used: 12.4.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"LS_5PIN_TAS","GPIO":[32,0,576,0,416,0,0,0,417,419,418,0,0,0],"FLAG":0,"BASE":18}
{"Module":{"0":"LS_5PIN_TAS"}}
{"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"419":"PWM4"},"GPIO14":{"418":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
Before switching on via Channel
{"Status":{"Module":0,"DeviceName":"kueche-leds","FriendlyName":["kueche-leds","","",""],"Topic":"leds","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:21:57","StartupUTC":"2023-03-21T14:14:26","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2022-11-18T14:56:21","SaveCount":440,"SaveAddress":"FB000"},"StatusFWR":{"Version":"12.4.0.3(tasmota)","BuildDateTime":"2023-03-21T08:40:21","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285H16","CR":"369/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 6660 Cable 2.4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00040080","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6","I2CDriver":"7"},"StatusNET":{"Hostname":"leds-7138","IPAddress":"192.168.178.64","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"4C:EB:D6:0B:7B:E2","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"coreelec","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B7BE2","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-03-21T14:36:23","Local":"2023-03-21T15:36:23","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:51","Sunset":"19:03"},"StatusSNS":{"Time":"2023-03-21T15:36:23"},"StatusSTS":{"Time":"2023-03-21T15:36:23","Uptime":"0T00:21:57","UptimeSec":1317,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Channel1":14,"POWER2":"OFF","Channel2":26,"POWER3":"OFF","Channel3":51,"POWER4":"OFF","Channel4":55,"Color":"2443828C","Wifi":{"AP":1,"SSId":"FRITZ!Box 6660 Cable 2.4","BSSId":"1C:ED:6F:9F:59:0A","Channel":7,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. 4 PWM Channels configured (i.e. https://templates.blakadder.com/athom_LS5P-TAS.html)
2. 3. setoption68 = ON
3. MQTT enabled and connect to broker
4. All Channels are switched off in the web-Gui
5. Move the first Channel slider to i.e. 50%
-> MQTT stat/RESULT shows POWER1:ON, MQTT stat/POWER1 = OFF

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
MQTT stat/POWER1 = ON

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-03-21:
The tree view of MQTT explorer is not very useful in that case vs a time oriented log
Could you please do the same but provide the console log?
Web log is fine but please use text copy/paste and not screen shots.
Thanks

moesfeld, 2023-03-21:
@barbudor Hi, weblog output is in the inital Post:
`16:50:12.388 CMD: Grp 0, Cmd 'CHANNEL', Idx 1, Len 2, Pld 67, Data '67'
16:50:12.395 MQT: kueche/leds/stat/RESULT = {"POWER1":"ON","Channel1":67,"POWER2":"OFF","Channel2":26,"POWER3":"OFF","Channel3":51,"POWER4":"OFF","Channel4":55,"Color":"AB43828C"}
16:50:13.942 CFG: Saved to flash at F5, Count 446, Bytes 4096`

Not much to see there. Channel 1 is set to 67 and the Result reflects that. But kueche/leds/stat/POWER1 stays OFF. In the webgui it is ON. I can work around it in Openhab/ioBroker etc... by using the RESULT Jsonpath. But I still think it's a Bug.

barbudor, 2023-03-21:
I suggest to use `SetOption90 1` to disable the non-JSON POWER messages

moesfeld, 2023-03-22:
Okay, but to what end? As i said i could work around it with the JSON POWER messages. I just think this should be considered a bug as the non Json Power messages should still reflect the current state. Internal state seems correct to me.

barbudor, 2023-03-22:
The new power state has been correctly updated
What you are looking at are old messages
This is why I sais that MQTT explorer is not a good tool as you don't see the time-line
Those POWER messages will not be resent as the updated state as already been provided in a json
I suggest to remove the non json and focus only on json messages

Jason2866, 2023-03-23:
Closing, since answered

closed 2023-03-23 by Jason2866