arendst/Tasmota #18263: Using Alexa to turn on a light at a low brightness causes a bright flash if the last brightness was high

Taudris, 2023-03-26:
### PROBLEM DESCRIPTION
If you set a Tasmota light to a high brightness, turn it off, then use Alexa to turn it back on at a low brightness, it will flash at the high brightness before switching to the low brightness.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): [ElectroDragon ESP32 LED strip control board](https://www.electrodragon.com/product/esp-led-strip-board/)
- [X] Tasmota binary firmware version number used: 12.4.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: ESPFlasher
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

08:38:21.988 RSL: RESULT = {"NAME":"ElectroDragon ESP32-C3 RGB","GPIO":[0,0,0,0,0,0,0,0,416,0,0,229,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
08:38:22.187 RSL: RESULT = {"Module":{"0":"ElectroDragon ESP32-C3 RGB"}}
08:38:22.399 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"416":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

08:39:24.449 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Kitchen","FriendlyName":["Kitchen"],"Topic":"tasmota_414E58","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:39:24.453 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"7T01:04:23","StartupUTC":"2023-03-19T06:35:01","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2023-03-19T06:32:37","SaveCount":142}}
08:39:24.457 RSL: STATUS2 = {"StatusFWR":{"Version":"12.4.0(tasmota)","BuildDateTime":"2023-02-24T15:35:41","Core":"2_0_6","SDK":"4.4.3.20230129","CpuFrequency":160,"Hardware":"ESP32-C3 rev.3","CR":"458/699"}}
08:39:24.462 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GoopGoop Tolerated",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00108001","2805C80001000600003C5A0A192800000000","00000088","00006200","00004000","00000000"]}}
08:39:24.471 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1543,"Free":1856,"Heap":185,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","080840D2","20200000","D43C482D","000080B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,127"}}
08:39:24.476 RSL: STATUS5 = {"StatusNET":{"Hostname":"kitchen.led","IPAddress":"192.168.12.65","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.12.1","DNSServer2":"0.0.0.0","Mac":"A0:76:4E:41:4E:58","IP6Global":"","IP6Local":"fe80::a276:4eff:fe41:4e58","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:39:24.480 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-03-26T07:39:24","Local":"2023-03-26T08:39:24","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:40","Sunset":"19:10"}}
08:39:24.486 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-03-26T08:39:24"}}
08:39:24.491 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-03-26T08:39:24","Uptime":"7T01:04:23","UptimeSec":608663,"Heap":183,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":40},"POWER":"ON","Dimmer":12,"Fade":"ON","Speed":2,"LedTable":"ON","Wifi":{"AP":1,"SSId":"GoopGoop Tolerated","BSSId":"F6:92:BF:A5:69:98","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

08:26:02.836 RSL: RESULT = {"POWER":"ON"}
08:26:02.838 RSL: POWER = ON
08:26:02.840 RSL: RESULT = {"POWER":"ON","Dimmer":59}
08:26:12.959 RSL: RESULT = {"POWER":"OFF"}
08:26:12.960 RSL: POWER = OFF
08:26:19.497 CMD: weblog 4
08:26:19.498 SRC: WebConsole from 192.168.11.11
08:26:19.500 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 1, Pld 4, Data '4'
08:26:19.502 RSL: RESULT = {"WebLog":4}
08:26:20.315 CFG: Saved, Count 139, Bytes 4096
08:26:22.228 HTP: Hue API (/414e58/lights/68478337) from 192.168.12.26
08:26:22.229 /lights path=/lights/68478337
08:26:22.233 HTP: Hue Result ({"state":{"on":false,"bri":150,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Kitchen","modelid":"ElectroDragon ","manufacturername":"Tasmota","uniqueid":"a0:76:4e:41:4e:58:00:11-01"})
08:26:22.278 HTP: Hue API (/414e58/lights/68478337) from 192.168.12.26
08:26:22.279 /lights path=/lights/68478337
08:26:22.283 HTP: Hue Result ({"state":{"on":false,"bri":150,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Kitchen","modelid":"ElectroDragon ","manufacturername":"Tasmota","uniqueid":"a0:76:4e:41:4e:58:00:11-01"})
08:26:22.328 HTP: Hue API (/414e58/lights/68478337/state) from 192.168.12.26
08:26:22.329 HTP: Hue POST args ({"on":true})
08:26:22.332 SRC: Hue
08:26:22.336 RSL: RESULT = {"POWER":"ON"}
08:26:22.338 RSL: POWER = ON
08:26:22.339 HTP: Hue Result ([{"success":{"/lights/68478337/state/on":true}}])
08:26:22.357 HTP: Hue API (/414e58/lights/68478337) from 192.168.12.26
[... 21 lines cut ...]
08:26:28.606 HTP: Hue Result ({"state":{"on":true,"bri":31,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Kitchen","modelid":"ElectroDragon ","manufacturername":"Tasmota","uniqueid":"a0:76:4e:41:4e:58:00:11-01"})
08:26:28.617 HTP: Hue API (/414e58/lights/68478337) from 192.168.12.26
08:26:28.619 /lights path=/lights/68478337
08:26:28.626 HTP: Hue Result ({"state":{"on":true,"bri":31,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Kitchen","modelid":"ElectroDragon ","manufacturername":"Tasmota","uniqueid":"a0:76:4e:41:4e:58:00:11-01"})
08:26:29.856 CMD: weblog 2
08:26:29.862 RSL: RESULT = {"WebLog":2}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Set a Tasmota light to a high brightness.
2. Turn off the light. (Note that it is still configured to operate at a high brightness.)
3. Use Alexa to turn on the light.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The light turns on at the requested brightness without any unpleasant flashing.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Taudris, 2023-03-26:
I was able to write a simple rule to work around this issue:

`Backlog Rule1 ON Power1#state=0 DO Channel1 0 ENDON;Rule1 1`

Still, it seems like a workaround for this behavior should be implemented in Tasmota, perhaps behind a `SetOption` toggle.

Jason2866, 2023-03-26:
Not a bug, it is a feature. Restore brightness which was used before switched off.

s-hadinger, 2023-03-26:
That's unfortunately the exact order which Alexa sends the commands:
- first turn ON at 08:26:22.329
- then change brightness at 08:26:22.378

On esp8266 it would be too complicated to defer Alexa commands until the next command arrives. Sorry

Jason2866, 2023-03-26:
Closing, since answered

closed 2023-03-26 by Jason2866