arendst/Tasmota #18311: Unknown command if IRHVAC is continuously send/repeated (ESP32-C3)

ferbulous, 2023-04-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

If I try to send IR command continuously,  IR signal doesn't get picked up by the target device
I would need to wait at least 1-2 seconds before sending the next command.

And sometimes sending IR command  doesn't work on the first attempt and I have to send the same command again 

This doesn't happen on an esp8266 remote.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions) - Discussed [here](https://github.com/arendst/Tasmota/discussions/18256)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): IR remote
- [ ] Tasmota binary firmware version number used: [Tasmota 12.4.0.4]
  - [ ] Pre-compiled
  - [x] Self-compiled - Yes (IR-all enabled) for ESP32-C3-05
- [x] Flashing tools used: UART 
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:02:20.006 MQT: stat/remote_livingroom/RESULT = {"NAME":"Tuya Remote","GPIO":[1,1,1,1088,1056,1,32,544,1,1,1,0,0,0,0,0,0,0,1,1,1,1],"FLAG":0,"BASE":1}
08:02:20.235 MQT: stat/remote_livingroom/RESULT = {"Module":{"0":"Tuya Remote"}}
08:02:20.442 MQT: stat/remote_livingroom/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"1088":"IRrecv"},"GPIO4":{"1056":"IRsend1"},"GPIO5":{"0":"None"},"GPIO6":{"32":"Button1"},"GPIO7":{"544":"LedLink"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:04:08.624 MQT: stat/remote_livingroom/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"remote_livingroom","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:04:08.629 MQT: stat/remote_livingroom/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Vbat power on reset","Uptime":"0T02:46:06","StartupUTC":"2023-04-01T04:18:02","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2023-03-28T11:08:56","SaveCount":26}}
08:04:08.633 MQT: stat/remote_livingroom/STATUS2 = {"StatusFWR":{"Version":"12.4.0.4(tasmota32)","BuildDateTime":"2023-03-28T22:38:39","Core":"2_0_7","SDK":"4.4.4.20230218","CpuFrequency":160,"Hardware":"ESP32-C3 rev.3","CR":"417/699"}}
08:04:08.637 MQT: stat/remote_livingroom/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Suliana",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
08:04:08.649 MQT: stat/remote_livingroom/STATUS4 = {"StatusMEM":{"ProgramSize":1717,"Free":1856,"Heap":181,"StackLowMark":5,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DOUT","Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","080840D2","20200000","D4BC482D","000280B1"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,60,62,63,66,67,86,87,88","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77"}}
08:04:08.657 MQT: stat/remote_livingroom/STATUS5 = {"StatusNET":{"Hostname":"remote-livingroom-6820","IPAddress":"192.168.1.75","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"7C:DF:A1:63:5A:A4","IP6Global":"","IP6Local":"fe80::7edf:a1ff:fe63:5aa4","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:04:08.661 MQT: stat/remote_livingroom/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_635AA4","MqttUser":"tasmota2","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:04:08.666 MQT: stat/remote_livingroom/STATUS7 = {"StatusTIM":{"UTC":"2023-04-01T07:04:08","Local":"2023-04-01T08:04:08","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:28","Sunset":"19:19"}}
08:04:08.672 MQT: stat/remote_livingroom/STATUS10 = {"StatusSNS":{"Time":"2023-04-01T08:04:08"}}
08:04:08.677 MQT: stat/remote_livingroom/STATUS11 = {"StatusSTS":{"Time":"2023-04-01T08:04:08","Uptime":"0T02:46:06","UptimeSec":9966,"Heap":180,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":4,"Objects":43},"Wifi":{"AP":1,"SSId":"Suliana","BSSId":"A0:AB:1B:84:8C:D0","Channel":6,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

IR send:
07:50:59.414 MQT: stat/remote_livingroom/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_88","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":19,"FanSpeed":"Low","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
07:51:05.420 MQT: stat/remote_livingroom/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_88","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":19,"FanSpeed":"Low","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
07:51:08.625 MQT: stat/remote_livingroom/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_88","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":19,"FanSpeed":"Low","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
07:51:11.523 MQT: stat/remote_livingroom/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_88","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":19,"FanSpeed":"Low","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
07:51:19.783 MQT: stat/remote_livingroom/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_88","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":20,"FanSpeed":"Low","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
07:51:21.731 MQT: stat/remote_livingroom/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_88","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":21,"FanSpeed":"Low","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
07:51:23.180 MQT: stat/remote_livingroom/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_88","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":22,"FanSpeed":"Low","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
07:51:24.977 MQT: stat/remote_livingroom/RESULT = {"IRHVAC":{"Vendor":"MITSUBISHI_HEAVY_88","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":23,"FanSpeed":"Low","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep" [... cut ...]
1) Use a second IR remote to capture IRHVAC signal
2) Increase/decrease temperature for the AC continously (18 C to 27 C)

### EXPECTED BEHAVIOUR
Some commands not received by target (unknown as indicated in the logs by ir receiver).
No issues doing the same test on esp8266 remote.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-04-01:
There could be multiple reasons for unreliable reception of IR signal
What is the receiving hardware ?
Is it the same circuitry between the ESP8266 and the ESP32 boards ?
If not are you sure the circuitry used on ESP32 have the proper sensitivity ?

Probably looking at the signal on the GPIO with an oscilloscope and comparing between ESP8266 and ESP32 could give some clues on a possible hardware issue.

ferbulous, 2023-04-01:
Sorry, i don't have oscilloscope 
Receiving hardware is the same device, I have 2 units for each one.

H/w tested:

[image]
[image]
[image]

Esp32 (Came with beken chip, I swapped with esp32-c3-05)
1) [Tuya IR remote S-08](https://www.aliexpress.com/item/1005005019779055.html?spm=a2g0o.productlist.main.23.591a5f5djD2HMB&algo_pvid=a3f95d87-05a4-45e6-81a8-1f2612665aa0&algo_exp_id=a3f95d87-05a4-45e6-81a8-1f2612665aa0-11&pdp_ext_f=%7B%22sku_id%22%3A%2212000031360481320%22%7D&pdp_npi=3%40dis%21MYR%2182.23%2126.16%21%21%21%21%21%40211bf31616803764840003328d06f1%2112000031360481320%21sea%21MY%21176259664&curPageLogUid=boKkmXKJ93d5)
2) [Aubess IR Remote](https://www.aliexpress.com/item/1005005249938898.html?spm=a2g0o.productlist.main.23.14933bd9tOm5oA&algo_pvid=59a1d0bd-dd6c-4878-ab16-58867579970e&algo_exp_id=59a1d0bd-dd6c-4878-ab16-58867579970e-11&pdp_ext_f=%7B%22sku_id%22%3A%2212000032360415064%22%7D&pdp_npi=3%40dis%21MYR%2151.47%2119.54%21%21%21%21%21%40211bf55216803764240228912d06c7%2112000032360415064%21sea%21MY%21176259664&curPageLogUid=Big5mZcAEj1G)

Esp8266:
[ESP-01M ](https://www.aliexpress.com/item/1005004486068078.html?aff_fcid=75f525b540934e44a2dd121f4a42bad5-1680376289036-07245-_DE4asUT&tt=CPS_NORMAL&aff_fsk=_DE4asUT&aff_platform=shareComponent-detail&sk=_DE4asUT&aff_trace_key=75f525b540934e44a2dd121f4a42bad5-1680376289036-07245-_DE4asUT&terminal_id=267f3eb90364441b95f083eecef09552&afSmartRedirect=y)

As for the same circuitry, i will swap esp8266 with one of tuya IR remote and redo the test.

Jason2866, 2023-04-17:
I had a similar issue. It is not related to the used MCU. This IR Blaster are all weak in sending.
Just many IR diodes does not make a good transmitter.
Solved the issue using a old IR Remote for sending. The IR diodes are sending with way more power since the are feed with 9v and more peak current.