arendst/Tasmota #18318: ESP8266 PRO MQTT TLS kernel crash

zokl, 2023-04-02:
### PROBLEM DESCRIPTION

After enabling MQTT TLS options tasmota crash immediately after a boot. System without MQTT TLS enabled works fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): WeMos D1 Mini Pro
- [X] Tasmota binary firmware version number used: 12.4.0.4(tasmota-4M), 2_7_4_9/2.2.2-dev(38a443e 
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: VSD + esptool
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```
19:12:09.813 RSL: RESULT = {"NAME":"Well Watter Level Meter","GPIO":[0,0,256,0,7008,7040,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
19:12:10.036 RSL: RESULT = {"Module":{"0":"Well Watter Level Meter"}}
19:12:10.249 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"256":"Relay_i1"},"GPIO3":{"0":"None"},"GPIO4":{"7008":"TFmini+ TX"},"GPIO5":{"7040":"TFmini+ RX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

19:13:29.498 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Kocouri studna","FriendlyName":["Led"],"Topic":"Studna_Kocurovi","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:13:29.502 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-4M.bin.gz","RestartReason":"External System","Uptime":"0T00:02:08","StartupUTC":"2023-04-02T18:11:21","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2023-03-29T21:28:26","SaveCount":85,"SaveAddress":"3FB000"}}
19:13:29.505 RSL: STATUS2 = {"StatusFWR":{"Version":"12.4.0.4(tasmota-4M)","BuildDateTime":"2023-04-02T19:53:24","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"409/699"}}
19:13:29.508 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Dum300-1","PAKOC"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000000"]}}
19:13:29.518 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":784,"Free":1260,"Heap":12,"ProgramFlashSize":4096,"FlashSize":16384,"FlashChipId":"18401C","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","879AC7CF","043E8001","000000CF","010013C0","C000F981","00004084","00201000","54000022","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,38,45,50,62","Sensors":"1,2,3,4,5,6,86","I2CDriver":"7"}}
19:13:29.522 RSL: STATUS5 = {"StatusNET":{"Hostname":"Studna-Kocurovi-4282","IPAddress":"10.30.1.134","Gateway":"10.30.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.30.1.1","DNSServer2":"80.250.1.161","Mac":"2C:F4:32:76:D0:BA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:13:29.526 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"iot.sidlisterez.eu","MqttPort":3883,"MqttClientMask":"Studna_Kocurovi","MqttClient":"Studna_Kocurovi","MqttUser":"iaq102","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:13:29.532 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-04-02T18:13:29","Local":"2023-04-02T19:13:29","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:26","Sunset":"19:21"}}
19:13:29.541 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-04-02T19:13:29","TFminiPlus":{"Distance":"19.0","SignalStrength":"2229","ChipTemperature":24},"WellLevelStatus":{"Level":"1300.00","Volume":"10210.17","WellRadius":"50.00","WellHeight":"1344.00","SensorHeight":"1320.00"},"TempUnit":"C"}}
19:13:29.547 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-04-02T19:13:29","Uptime":"0T00:02:08","UptimeSec":128,"Heap":12,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":405,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Dum300-1","BSSId":"B8:69:F4:E7:A5:DC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Enable MQTT TLS on ESP8266EX

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld
[... 140 lines cut ...]
~ld

00:00:00.001 HDW: ESP8266EX
00:00:00.010 UFS: FlashFS mounted with 1984 kB free
00:00:00.063 CFG: Loaded from File, Count 61
18:36:01.007 CMD: SetOption103 0
18:36:01.011 RSL: RESULT = {"SetOption103":"OFF"}
18:36:01.046 RSL: INFO1 = {"Info1":{"Module":"Well Watter Level Meter","Version":"12.4.0.3(tasmota-4M)","FallbackTopic":"cmnd/Studna_Kocurovi_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:36:01.053 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"Studna-Kocurovi-4282","IPAddress":"10.30.1.134"}}
18:36:01.072 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["4027dbb1","00000000","00000000"],"EXCVADDR":"00000d7a","DEPC":"00000000","CallChain":["4027d4e0","402b0cc0","4026e55d","4026dc26","4026db5c","4026e688","40293af3","40100aa0","40293b42","40297673","40293c26","4010284f","40292bb0","40285fa7","40106017","40100aa0","40102714","401059af","40100aa0","40102714","401059c7","40102714","401032ab","4010348c","4010348c","4010394e","4000050c","40103670","4025176d","40256cc6","402b11fd"]},"BootCount":21}}
18:36:01.111 RSL: RESULT = {"POWER":"OFF"}
18:36:01.114 RSL: POWER = OFF
18:36:01.519 APP: Restarting
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-04-02:
I'm using TLS on ESP8266 without any problems on a few devices (Nous A1T, mini-D1) with cloud MQTT broker flespi.io

> Self-compiled

Please provide details of your build including if any any custom PlatormIO environment
As well as details about your TLS configuration

zokl, 2023-04-02:
I use TLS on many other ESP8266 devices too, but the problem is only on WeMos D1 mini PRO with 16 MB of memory.

zokl, 2023-04-02:
My user_config_override.h. Server USE CA and RSA Is 2048.

```
#ifndef USE_EXPRESSION
#define USE_EXPRESSION         // Add support for expression evaluation in rules (+3k2 code, +64 bytes mem)  
#endif

#ifndef SUPPORT_IF_STATEMENT
#define SUPPORT_IF_STATEMENT   // Add support for IF statement in rules (+4k2 code, -332 bytes mem)  
#endif

#ifndef USE_MQTT_TLS 
#define USE_MQTT_TLS                             // Use TLS for MQTT connection (+34.5k code, +7.0k mem and +4.8k additional during connection handshake)
#define USE_MQTT_TLS_CA_CERT                   // Force full CA validation instead of fingerprints, slower, but simpler to use.  (+2.2k code, +1.9k mem during connection handshake)
// #define USE_4K_RSA
                                                   // This includes the LetsEncrypt CA in tasmota_ca.ino for verifying server certificates
// #define USE_MQTT_TLS_FORCE_EC_CIPHER           // Force Elliptic Curve cipher (higher security) required by some servers (automatically enabled with USE_MQTT_AWS_IOT) (+11.4k code, +0.4k mem)
#endif

#ifndef USE_TFMINIPLUS 
#define USE_TFMINIPLUS 
#endif

#ifndef USE_SCRIPT_JSON_EXPORT
#define USE_SCRIPT_JSON_EXPORT
#endif

#ifndef USE_SCRIPT_WEB_DISPLAY
#define USE_SCRIPT_WEB_DISPLAY
#endif
```

Jason2866, 2023-04-02:
This does nothing
```
#ifndef USE_TFMINIPLUS 
#define USE_TFMINIPLUS 
#endif
```
since Tasmota uses nowhere such a Macro. Have you made changes in code?

zokl, 2023-04-02:
> This does nothing
> 
> ```
> #ifndef USE_TFMINIPLUS 
> #define USE_TFMINIPLUS 
> #endif
> ```
> 
> since Tasmota uses nowhere such a Macro. Have you made changes in code?

Sources are without changes. Only macros in config.

Jason2866, 2023-04-02:
Ahh i missed the PR for supporting `TFmini Plus Time-Of-Flight (TOF)`

zokl, 2023-04-02:
> Ahh i missed the PR for supporting `TFmini Plus Time-Of-Flight (TOF)`

Without TLS everything works:

[image]

```
20:45:39.405 Well Level Status: height = 1090.00 cm, volume = 8560.83 l
20:45:43.423 RSL: STATE = {"Time":"2023-04-02T20:45:43","Uptime":"0T00:21:04","UptimeSec":1264,"Heap":12,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Dum300-1","BSSId":"B8:69:F4:E7:A5:DC","Channel":1,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:45:43.452 RSL: SENSOR = {"Time":"2023-04-02T20:45:43","TFminiPlus":{"Distance":"230.0","SignalStrength":"1589","ChipTemperature":27},"WellLevelStatus":{"Level":"1090.00","Volume":"8560.83","WellRadius":"50.00","WellHeight":"1344.00","SensorHeight":"1320.00"},"TempUnit":"C"}
20:47:03.222 MQT: Attempting connection...
20:47:07.256 MQT: Connect failed to iot.sidlisterez.eu:3883, rc -4. Retry in 120 sec
```

arendst, 2023-04-02:
Heap is way too low. Unable to support TLS in this config.

Jason2866, 2023-04-02:
Yes, since you have enough free heap now ;-)

closed 2023-04-02 by Jason2866

zokl, 2023-04-02:
> Yes, since you have enough free heap now ;-)

Does there exist any way how to increase the heap?

Jason2866, 2023-04-02:
Disable not needed driver in your `user_config_override.h`

zokl, 2023-04-02:
Thank you; your recommendation works. It will be nice to have some tools for memory manipulation.

sfromis, 2023-04-02:
To increase the heap, get ESP32