arendst/Tasmota #18319: Incoming KNX messages stop being processed

realtmf, 2023-04-02:
### PROBLEM DESCRIPTION
My Athom plugs occasionally exhibit a problem where incoming KNX messages stop being received or processed. The problem is intermittent, i.e. it goes away when restarting the plugs but eventually reappears until rebooted. Various plugs suffer from the problem (all devices are connected to the same Unifi 2.4GHz SSID, wireless multicast is enabled).

KNX configuration for one of the plugs exhibiting the problem:
Data to Send to Group Addresses:
Output 1 -> 1 / 1 / 144 	
Voltage -> 5 / 2 / 144 	
Current -> 5 / 3 / 144 	
Power -> 5 / 1 / 144 	
Energy Today -> 5 / 4 / 144 	
Energy Total -> 5 / 5 / 144

Group Addresses to Receive Data from:
1 / 0 / 144 -> Output 1

Please note that this has been going on for a while, also with earlier firmware versions.
Your kind assistance is most appreciated!

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Athom tech EU plug V1
- [X] Tasmota binary firmware version number used: 12.4.0(knx)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA upgrade from factory tasmota firmware
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:16:36.598 CMD: Backlog Template; Module; GPIO 255
19:16:36.600 SRC: WebConsole from 10.9.1.226
19:16:36.604 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
19:16:36.676 SRC: Backlog
19:16:36.679 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
19:16:36.684 RSL: RESULT = {"NAME":"Athom Power Monitoring Plug","GPIO":[0,0,0,32,2720,2656,0,0,2624,544,224,0,0,1],"FLAG":0,"BASE":18}
19:16:36.893 SRC: Backlog
19:16:36.896 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
19:16:36.899 RSL: RESULT = {"Module":{"0":"Athom Power Monitoring Plug"}}
19:16:37.143 SRC: Backlog
19:16:37.146 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
19:16:37.154 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"544":"LedLink"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
n/a

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:17:40.586 CMD: Status 0
19:17:40.589 SRC: WebConsole from 10.9.1.226
19:17:40.591 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
19:17:40.597 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"athom04","FriendlyName":["Athom04"],"Topic":"tasmota_024C24","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:17:40.630 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-knx.bin","RestartReason":"Software/System restart","Uptime":"17T07:01:52","StartupUTC":"2023-03-16T11:15:48","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2022-06-01T22:30:41","SaveCount":1545,"SaveAddress":"F6000"}}
19:17:40.663 RSL: STATUS2 = {"StatusFWR":{"Version":"12.4.0(knx)","BuildDateTime":"2023-02-24T15:32:46","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285H16","CR":"365/699"}}
19:17:40.682 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":4,"LogHost":"10.9.2.70","LogPort":514,"SSId":["thoekske",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["02008001","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
19:17:40.714 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":628,"Free":372,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","9F9AC587","04268011","000000CF","010013C0","8000F181","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,27,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
19:17:40.747 RSL: STATUS5 = {"StatusNET":{"Hostname":"athom04","IPAddress":"10.9.2.144","Gateway":"10.9.2.254","Subnetmask":"255.255.255.0","DNSServer1":"10.9.2.1","DNSServer2":"0.0.0.0","Mac":"4C:EB:D6:02:4C:24","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:17:40.770 RSL: STATUS7 = {"StatusTIM":{"UTC":"2023-04-02T18:17:40","Local":"2023-04-02T19:17:40","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:26","Sunset":"19:21"}}
19:17:40.787 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:17:40.804 RSL: STATUS10 = {"StatusSNS":{"Time":"2023-04-02T19:17:40","ENERGY":{"TotalStartTime":"2022-06-01T22:30:41","Total":14.415,"Yesterday":0.017,"Today":0.007,"Power":9,"ApparentPower":15,"ReactivePower":12,"Factor":0.61,"Voltage":269,"Current":0.057}}}
19:17:40.825 RSL: STATUS11 = {"StatusSTS":{"Time":"2023-04-02T19:17:40","Uptime":"17T07:01:52","UptimeSec":1494112,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"thoekske","BSSId":"18:E8:29:B6:6D:8E","Channel":13,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":36,"Downtime":"0T00:06:52"}}}

[... 36 lines cut ...]
19:23:41.860 KNX: Current sent to 5.3.144
19:23:41.864 KNX: Power sent to 5.1.144
19:23:41.868 KNX: Energy Today sent to 5.4.144
19:23:41.871 KNX: Energy Total sent to 5.5.144
19:23:41.876 RSL: SENSOR = {"Time":"2023-04-02T19:23:41","ENERGY":{"TotalStartTime":"2022-06-01T22:30:41","Total":14.416,"Yesterday":0.017,"Today":0.008,"Period":1,"Power":9,"ApparentPower":15,"ReactivePower":12,"Factor":0.60,"Voltage":268,"Current":0.057}}
19:23:42.200 HLW: EPC 5, CFlen 183536 usec
19:23:43.201 HLW: EPC 6, CFlen 189967 usec
...

So, KNX messages are still sent, but incoming messages for group address 1 / 0 / 144 are not received/processed. Note that both Wireshark as well as "Multicast Tester" on android see the KNX request messages on this group address (see screenshot below).

```

### TO REPRODUCE
Unable to force the problem to occur

### EXPECTED BEHAVIOUR
Reliably identify and process incoming KNX messages

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT
Enabling "Communication Enhancement" in the KNX settings does not seem to affect the issue (i.e. problem exists with or without this setting).

Noschvie, 2023-06-15:
Same here with 12.5. and a Sonoff 4CH Pro. How can I help to debug ?

realtmf, 2023-06-15:
@Noschvie Are you by any chance using Ubiquity network gear? (I am wondering if it has to do with wifi sleep mode, but I have no idea how to reproduce...)

Noschvie, 2023-06-16:
@realtmf sorry, no Ubiquity network gear available.
Which "earlier firmware versions" have been working?

realtmf, 2023-06-16:
@Noschvie I am not aware of FW versions where the problem didn't exhibit itself. (That being said, the problem may lie outside of Tasmota firmware - how to know?...)

Noschvie, 2023-06-16:
> Please note that this has been going on for a while, also with earlier firmware versions.

What does this mean?

Will setup and use a "Shelly Plus 1" for tests...

Noschvie, 2023-06-18:
Solve my issue with WIFI multicast enabled. This setting seems got lost during WLAN reconfigurations...
Tested with Tasmota 12.5.0 (self compiled) and ESP8266, works fine.
Sorry for confusion...

Jason2866, 2023-06-26:
@realtmf  Please try with latest version and pleaseenhance your wifi.
With such a bad `RSSI":48,"Signal":-76` wifi signal a reliable function is not possible.

Jason2866, 2023-07-02:
Closing since no feedback from OP. KNX does work for others.

closed 2023-07-02 by Jason2866