arendst/Tasmota #18327: Programmatically get access to other drivers/sensors

Heizelmann, 2023-04-04:
### Discussed in https://github.com/arendst/Tasmota/discussions/18306

<div type='discussions-op-text'>

<sup>Originally posted by **Heizelmann** March 31, 2023</sup>
E.g. from my own driver modul I would like to access the temperature value of a ds18x20.</div>

sfromis, 2023-04-04:
Don't create so many threads about the same topic, already covered and not relevant as an "issue" to be fixed.

closed 2023-04-04 by arendst

Heizelmann, 2023-04-04:
sorry, but I am not so familiar with github discussions.  I didn't know how to reopen a closed discussion. Is this place only for issues? I thought I can place also question for help here.

sfromis, 2023-04-04:
As discussions are not closed, there is no concept of reopening. They are the place (along with Discord) for questions, help and so on. As already suggested, issues are only for things to be fixed.