arendst/Tasmota #18337: ai-thinker ESP-32 cam problem with standard clock frequency (wcclock) 20 MHz

wohack, 2023-04-05:
I have been using ESP-32 cams for quite some time. I bought the modules in 2021/2022.
Now I needed another one, so I bought a 3 pack from AZ-Delivery. The cams were not usable, constantly freezing and frame rates of 0.2 frames per second. After a restart 1 sometimes the OV2640. could not be initialised. Therefore I thought the cams are broken.
So I returned the modules and bought a 2 pack diymore ESP32 CAM. I already bought this cam module 2 times (4 pcs in 2021/2022) and with it there were and are no problems with WCClock 20.

The new delivery showed the same problems as the cams from AZ delivery (freezing, poor frame rates). 
I next swapped the OV2640 camera module with one of my older cams. That's not the problem.
After that I experimented with WCResolution and finally with WCClock. When I changed WCClock to 17 (MHz) the camera ran smoother. With WCClock 15 (MHz) the camera runs stable and without performance problems for 2 weeks now.

For me it looks like that at the moment ESP32 cams are on the market via different vendors, which do not cope with 20 MHz.

My suggestion is therefore to reduce the default value (WCClock) from 20 to 15 as this increases compatibility.
For the tests I used the current released version 12.4.0 from the download page on my old and new modules. Software seems not to be the problem.

Jason2866, 2023-04-05:
The standard clock mode is 20Mhz. Modules not supporting are not in specs.
Feel free to add a note in Tasmota documentation for getting cheap build modules going.

closed 2023-04-05 by Jason2866

btsimonh, 2024-01-07:
@wohack - I also had this problem.
Check the V2 driver.
And also try setting the freq off from the standard by 1-2 Mhz.
I found that for one of my boards, the wifi is compromised by using the default.  If you tune is slightly off, the wifi performance improved drasticaly. I guess it;s down to board layout.