arendst/Tasmota #18355: Is Backlog within Json command supported?

caenrigen, 2023-04-06:
Hey folks!

First of all thank you for the very nice framework! 😄

I think I am running into an unsupported feature or some kind of bug. Any help is appreciated.

I am in the Tasmota console. This works well:

```
01:18:19.495 CMD: Rule1 ON Clock#Timer=1 DO Backlog Color #FF000000D0; Wakeup 100 ENDON
01:18:19.501 RUL: Stored uncompressed, would compress from 63 to 57 (-10%)
01:18:19.505 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":63,"Free":448,"Rules":"ON Clock#Timer=1 DO Backlog Color #FF000000D0; Wakeup 100 ENDON"}}
```

but this does not work well and does something strange:

```
01:19:30.444 CMD: Json {"Rule1": "ON Clock#Timer=1 DO Backlog Color #FF000000D0; Wakeup 100 ENDON"}
01:19:30.630 RUL: Stored uncompressed, would compress from 45 to 39 (-14%)
01:19:30.634 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":45,"Free":466,"Rules":"ON Clock#Timer=1 DO Backlog Color #FF000000D0"}}
01:19:30.643 RSL: RESULT = {"Wakeup":"Started"}
01:19:30.692 RSL: RESULT = {"Wakeup":"Done","POWER":"ON","Dimmer":100,"Color":"FF000000FF","HSBColor":"0,100,100","White":100,"CT":500,"Channel":[100,0,0,0,100],"Scheme":1,"Fade":"OFF","Speed":1,"LedTable":"ON"}
01:19:59.303 CMD: Rule1
01:19:59.309 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":45,"Free":466,"Rules":"ON Clock#Timer=1 DO Backlog Color #FF000000D0"}}
```

Notice how the `"Rules"` are missing everything that was supposed to come after `;`. I am guessing this is somehow a problem with using `Backlog` within a `Json` command. 

Is this a bug?

---

PS I am running `Tasmota 12.0.2` on a smart lamp that came with it pre-installed.

barbudor, 2023-04-07:
Why are you using json command instead of the straight command?

caenrigen, 2023-04-07:
> Why are you using json command instead of the straight command?

I am sending a series of configurations to the device at once (sending them individually is a bit slow and more painful).

I know that I should be able to use Backlog command to achieve the same goal (did not try yet, I assume it works, does a Backlog within a Backlog work?), though Json is more convenient for my system. Besides the fact that I already have a bunch of "tools" developed that rely on the Json command (I am using Apple Shortcuts)

barbudor, 2023-04-07:
Ok
So indeed there may be a problem with `;` inside a Json command.
Needs some investigation
May be Theo @arendst will have a quick solution as he signed the original implementation
Otherwise, I'll try to have a look this week end

caenrigen, 2023-04-07:
> Ok
> So indeed there may be a problem with `;` inside a Json command.
> Needs some investigation
> May be Theo @arendst will have a quick solution as he signed the original implementation
> Otherwise, I'll try to have a look this week end

Thank you :)

> How are you sending the Json command?
> Console? MQTT? Http?

My full workflow is intended to use Http, but the behaviour seems the same when using the Console

arendst, 2023-04-07:
It's not so much if backlog is supported within a json command as well using a rule command within a backlog command.

The json command is a wrapper around backlog commands. Backlog commands are terminated with ";" so executing a rule within a backlog command won't work.

You would see the same issue if you executed your rule within a backlog command.

Alas, not supported.

sfromis, 2023-04-07:
One ugly workaround could be to first ensure that some variable contains the semicolon, obviously not via the JSON command.
`var9 ;`
(could also be a mem variable to not have to set it every time)

And then:
`Json {"Rule1": "ON Clock#Timer=1 DO Backlog Color #FF000000D0%var9% Wakeup 100 ENDON"}`

Not really recommended, but not impossible either :grimacing:

arendst, 2023-04-07:
Let me think about a solution where the json command only contains one command. 

This wouldn't need to be serviced by backlog but could be fed back to the command server.....