arendst/Tasmota #18359: seriallog suspend

selinawy, 2023-04-07:
### PROBLEM DESCRIPTION
After gpio0 is configured as relay 1 and gpio1 as relay 2, the monitoring serial port log stops somewhere without output.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _ESP32-WROOM-32 dev board____
- [X] Tasmota binary firmware version number used: _12.4.0(tasmota)____
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: _OTA____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
 
11:48:18.405 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
11:48:18.624 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
11:48:18.831 RSL: RESULT = {"GPIO0":{"224":"Relay1"},"GPIO1":{"225":"Relay2"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}

````

### SCREENSHOTS

After gpio0 is configured as relay 1 and gpio1 as relay 2, the monitoring serial port log stops somewhere without output.

[image]

[image]

arendst, 2023-04-07:
Why would this happen you think.

Perhaps GPIO1 is the Tx pin connected to your serial interface that you now have configured as a relay.....

selinawy, 2023-04-07:
@arendst  thanks!!!

closed 2023-04-07 by Jason2866