arendst/Tasmota #18368: Shelly 2.5 Wrong Power Output

Wilhelm1Daniel, 2023-04-08:
### PROBLEM DESCRIPTION
I bought a Shelly 2.5 and flashed it with Tasmota 12.4.0.
To calibrate the device I take a 60W bulb and tried the points like described in github-docs.
When I switch on the light on Channel 1 the power is 66W, if I use Channel 2 the power is only 8W.
Is this a software bug ? Or is the device broken. I can't check it because I only have one of this device.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 2.5
- [ ] Tasmota binary firmware version number used: Tasmota-DE 12.4.0
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ WebFlashingTool] Flashing tools used: _____
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
`
18:21:47.712 MQT: stat/EG/BU/SteckdoseAussen/RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}
18:21:47.923 MQT: stat/EG/BU/SteckdoseAussen/RESULT = {"Module":{"0":"Shelly 2.5"}}
18:21:48.180 MQT: stat/EG/BU/SteckdoseAussen/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"193":"Switch_n2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"192":"Switch_n1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ1"},"GPIO17":{"4736":"ADC Temp"}}`

```
- [NONE ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
`
18:22:33.367 MQT: stat/EG/BU/SteckdoseAussen/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"EG/BU/SteckdoseAussen","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:22:33.373 MQT: stat/EG/BU/SteckdoseAussen/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:59:45","StartupUTC":"2023-04-08T15:22:48","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2023-03-24T20:56:17","SaveCount":199,"SaveAddress":"F6000"}}
18:22:33.379 MQT: stat/EG/BU/SteckdoseAussen/STATUS2 = {"StatusFWR":{"Version":"12.4.0(tasmota)","BuildDateTime":"2023.02.24 15:37:29","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"413/699"}}
18:22:33.383 MQT: stat/EG/BU/SteckdoseAussen/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Unser Netz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:22:33.395 MQT: stat/EG/BU/SteckdoseAussen/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
18:22:33.414 MQT: stat/EG/BU/SteckdoseAussen/STATUS5 = {"StatusNET":{"Hostname":"EG-BU-SteckdoseAussen-2330","IPAddress":"192.168.178.186","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"2C:F4:32:69:09:1A","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:22:33.419 MQT: stat/EG/BU/SteckdoseAussen/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.69","MqttPort":1883,"MqttClientMask":"BUSteckdoseAussen","MqttClient":"BUSteckdoseAussen","MqttUser":"Daniel","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:22:33.425 MQT: stat/EG/BU/SteckdoseAussen/STATUS7 = {"StatusTIM":{"UTC":"2023-04-08T16:22:33","Local":"2023-04-08T18:22:33","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"07:13","Sunset":"20:30"}}
18:22:33.431 MQT: stat/EG/BU/SteckdoseAussen/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:22:33.442 MQT: stat/EG/BU/SteckdoseAussen/STATUS10 = {"StatusSNS":{"Time":"2023-04-08T18:22:33","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":43.3},"ENERGY":{"TotalStartTime":"2023-03-24T20:56:17","Total":0.047,"Yesterday":0.000,"Today":0.010,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":229,"Current":[0.000,0.000]},"TempUnit":"C"}}
18:22:33.448 MQT: stat/EG/BU/SteckdoseAussen/STATUS11 = {"StatusSTS":{"Time":"2023-04-08T18:22:33","Uptime":"0T00:59:45","UptimeSec":3585,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Unser Netz","BSSId":"F0:B0:14:8F:DA:22","Channel":6,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":2,"Downtime":"0T00:00:05"}}}`

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-04-08:
The bulb you used is a classic (Edison) one?  None of the 2 channels shows correct values.
11A for a 60W bulb ;-)

Wilhelm1Daniel, 2023-04-08:
> The bulb you used is a classic (Edison) one? None of the 2 channels shows correct values. 11A for a 60W bulb ;-)

Yes it's a normal one with 60W

I try all of this commands...

`FrequencySet 50.000
PowerSet 60.00
VoltageSet 240.0
CurrentSet 250.0`

but nothing happens...

Jason2866, 2023-04-08:
Calibrate output two using command `VoltageSet2`,` CurrentSet2` and `PowerSet2`

Wilhelm1Daniel, 2023-04-08:
I'm an idiot... that was to easy ;-)
Thanks a lot...

closed 2023-04-08 by Wilhelm1Daniel