arendst/Tasmota #18400: Ser2Net from SmartMeter bad data

ressof, 2023-04-13:
### PROBLEM DESCRIPTION
I get bad data and different data from a Electric Smart Meter that sends data by serial link to a Wemos D1 mini on GPIO3. I have the tasmota act as an Ser2Net device and also meassure temperature with a DHT22 and transfer that by MQTT to domoticz.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D1 mini pro
- [ ] Tasmota binary firmware version number used: 12.4.0 & 12.3.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ressof, 2023-04-13:
I think I resolved the issue by adding this to user_config_override
#define TCP_BRIDGE_BUF_SIZE    1024  // size of the buffer, above 132 required for efficient XMODEM

The buffer is 256 as default.

arendst, 2023-04-15:
Fixed by OP

closed 2023-04-15 by arendst