arendst/Tasmota #18427: [TasmotaClient] teleperiod message is cutted off

Gifford47, 2023-04-16:
### PROBLEM DESCRIPTION
i have a esp32 and a 3v3 arduino pro mini attached to it. on the arduino side i read all gpios and want to send the states via teleperiod function to mqtt. the string from arduino is with 152 chars a bit long. after sending the string from arduino to the esp, the message is cutted off and only 100 chars left. i tried to expand the read and write buffers but nothing changed. so i assume there`s a limit on tasmota side.

The code on the arduino (tasmotaclient) side is:
```
void user_FUNC_JSON(void)
{
  char myjson[240];
  // analog pins as int: A0=14,A1=15,A2=16,A3=17,A4=18,A5=19,A6=20,A7=21
  //send analog pins first
  sprintf(myjson,"{\"1\":%u,\"2\":%u,\"3\":%u,\"4\":%u,\"5\":%u,\"6\":%u,\"7\":%u,\"8\":%u,"
  "\"9\":%d,\"10\":%d,\"11\":%d,\"12\":%d,\"13\":%d,\"14\":%d,\"15\":%d,\"16\":%d,\"17\":%d,\"18\":%d,\"19\":%d,\"20\":%d}", \
  analogRead(A0), analogRead(A1), analogRead(A2), analogRead(A3), analogRead(A4), analogRead(A5), analogRead(A6), analogRead(A7),\
  digitalRead(2), digitalRead(3), digitalRead(4), digitalRead(5), digitalRead(6), digitalRead(7), digitalRead(8), \
  digitalRead(9), digitalRead(10), digitalRead(11), digitalRead(12), digitalRead(13));
  client.sendJSON(myjson);
}
```

the Log on the esp side is:
```
11:42:19.624 CMD: teleperiod
11:42:19.628 MQT: stat/tasmota_AAAAAA/RESULT = {"TelePeriod":600}
11:42:20.078 MQT: tele/tasmota_AAAAAA/STATE = {"Time":"2023-04-16T11:42:20","Uptime":"0T00:01:15","UptimeSec":75,"Heap":175,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER":"OFF"}
11:42:20.123 MQT: tele/tasmota_AAAAAA/SENSOR = {"Time":"2023-04-16T11:42:20","TasmotaClient":{"1":0,"2":0,"3":0,"4":0,"5":0,"6":0,"7":0,"8":0,"9":0,"10":0,"11":0,"12":0,"13":0,"14":0,"15":0,"1}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): esp32devkit
- [X] Tasmota binary firmware version number used: 12.4.0.2
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
11:44:46.865 CMD: Backlog Template; Module; GPIO 255
11:44:46.881 MQT: stat/tasmota_AAAAAA/RESULT = {"NAME":"ESP32-RAIL","GPIO":[1,1,608,1,4160,4192,0,0,5536,1,1,640,4224,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
11:44:47.114 MQT: stat/tasmota_AAAAAA/RESULT = {"Module":{"0":"ESP32-RAIL"}}
11:44:47.372 MQT: stat/tasmota_AAAAAA/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"608":"I2C SCL1"},"GPIO3":{"0":"None"},"GPIO4":{"4160":"Client TX"},"GPIO5":{"4192":"Client RX"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"5536":"ETH POWER"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"640":"I2C SDA1"},"GPIO16":{"4224":"Client RST"},"GPIO17":{"0":"None"},"GPIO18":{"5600":"ETH MDIO"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"5568":"ETH MDC"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
No Rules used

```
- [X] Provide the output of this command: `Status 0`:
```lua
11:46:05.957 CMD: Status 0
11:46:05.963 MQT: stat/tasmota_AAAAAA/STATUS = {"Status":{"Module":0,"DeviceName":"esp32rail","FriendlyName":["esp32rail"],"Topic":"tasmota_AAAAAA","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:46:05.999 MQT: stat/tasmota_AAAAAA/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"0","RestartReason":"Software reset CPU","Uptime":"0T00:05:00","StartupUTC":"2023-04-16T10:41:05","Sleep":50,"CfgHolder":4617,"BootCount":95,"BCResetTime":"2022-06-02T20:06:36","SaveCount":214}}
11:46:06.035 MQT: stat/tasmota_AAAAAA/STATUS2 = {"StatusFWR":{"Version":"12.4.0.2(MyOwn)","BuildDateTime":"2023-03-26T20:28:55","Core":"2_0_7","SDK":"4.4.4.20230218","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"353/699"}}
11:46:06.051 MQT: stat/tasmota_AAAAAA/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["nodes_land",""],"TelePeriod":600,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600006E5A0A192800000000","00000080","00004000","00004000","00000000"]}}
11:46:06.095 MQT: stat/tasmota_AAAAAA/STATUS4 = {"StatusMEM":{"ProgramSize":1392,"Free":1856,"Heap":170,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8FBAC7CF","001493A1","000000CF","010013C0","C0010981","400040A0","00200000","5400002C","00000080"],"Drivers":"1,2,3,4,5,7,8,9,10,12,13,16,20,21,24,26,27,29,31,35,38,50,52,62,82","Sensors":"1,2,3,5,6,127","I2CDriver":"3,4,5,6,7,47"}}
11:46:06.149 MQT: stat/tasmota_AAAAAA/STATUS5 = {"StatusNET":{"Hostname":"tasmota-D286E8-1768","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:D2:86:E8","IP6Global":"","IP6Local":"","Ethernet":{"Hostname":"tasmota-D286E8-1768-eth","IPAddress":"192.168.1.120","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:D2:86:EB","IP6Global":"","IP6Local":"fe80::cac9:a3ff:fed2:86eb"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:46:06.202 MQT: stat/tasmota_AAAAAA/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D286E8","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:46:06.233 MQT: stat/tasmota_AAAAAA/STATUS7 = {"StatusTIM":{"UTC":"2023-04-16T10:46:06","Local":"2023-04-16T11:46:06","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:57","Sunset":"19:42"}}
11:46:06.282 MQT: stat/tasmota_AAAAAA/STATUS10 = {"StatusSNS":{"Time":"2023-04-16T11:46:06","TasmotaClient":{"1":0,"2":0,"3":0,"4":0,"5":0,"6":0,"7":0,"8":0,"9":0,"10":0,"11":0,"12":0,"13":0,"14":0,"15":0,"1}}
11:46:06.302 MQT: stat/tasmota_AAAAAA/STATUS11 = {"StatusSTS":{"Time":"2023-04-16T11:46:06","Uptime":"0T00:05:01","UptimeSec":301,"Heap":172,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER":"OFF"}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
11:46:45.299 CMD: teleperiod
11:46:45.301 SRC: WebConsole from 192.168.1.165
11:46:45.304 CMD: Grp 0, Cmd 'TELEPERIOD', Idx 1, Len 0, Pld -99, Data ''
11:46:45.308 MQT: stat/tasmota_AAAAAA/RESULT = {"TelePeriod":600}
11:46:46.051 BRY: GC from 4083 to 3798 bytes, objects freed 1/43 (in 1 ms) - slots from 48/122 to 45/122
11:46:46.056 MQT: tele/tasmota_AAAAAA/STATE = {"Time":"2023-04-16T11:46:46","Uptime":"0T00:05:41","UptimeSec":341,"Heap":175,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":43},"POWER":"OFF"}
11:46:46.102 MQT: tele/tasmota_AAAAAA/SENSOR = {"Time":"2023-04-16T11:46:46","TasmotaClient":{"1":0,"2":0,"3":0,"4":0,"5":0,"6":0,"7":0,"8":0,"9":0,"10":0,"11":0,"12":0,"13":0,"14":0,"15":0,"1}

```

### TO REPRODUCE
- attach a pro mini with tasmotaclient lib
- try to send a very long teleperiod message with ```user_FUNC_JSON``` 

### EXPECTED BEHAVIOUR
The whole teleperiod message is send to tasmota and further to mqtt broker.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2023-04-16:
Yes
It's here :
https://github.com/arendst/Tasmota/blob/d95d56f3e476a2e3fdb79f3c52d67937fba4b4f4/tasmota/tasmota_xdrv_driver/xdrv_31_tasmota_client.ino#L440-L450

However increasing a buffer in the stack would not be a good idea...

arendst, 2023-04-16:
Let me do some heap research and provide a fix as 100 is indeed a bit small.

arendst, 2023-04-16:
Fixed in latest dev release. Increased to 250 characaters. Larger would need a partial rewrite of TasmotaClient to handle indexes over 8-bit.

Gifford47, 2023-04-16:
Wow, that's a fast answer and solution 😀 thanks for your support @arendst @barbudor !!

Jason2866, 2023-04-16:
Closing since fixed

closed 2023-04-16 by Jason2866