arendst/Tasmota #18428: Wrong sensor detected on the i2C bus --> reported as I2c VEML6070 UV sensor

lboue, 2023-04-16:
### PROBLEM DESCRIPTION
A wrong sensor is detected on the i2C bus. 
It is reported as 

1. VEML6070 UV sensor --> KO
2. BMP280 --> OK

Here is the display:
```
BMP280 Temperature	28.0 °C
BMP280 Pressure	1017.3 hPa
VEML6070 UV Level	65535 inc
VEML6070 UV Index	99.00 OoR
VEML6070 UV Power	2.475 W/m²
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

lboue, 2023-04-16:
I add to compile a custom build and add the following to user_config_override.h:

```
#ifndef USE_AHT2x
#define USE_AHT2x       // [I2cDriver43] Enable AHT20/AM2301B instead of AHT1x humidity and temperature sensor (I2C address 0x38) (+0k8 code)
#endif
```

Now it works fine:

```
BMP280 Temperature	27.1 °C
BMP280 Pressure	1016.7 hPa
AHT2X Temperature	25.2 °C
AHT2X Humidity	47.1 %
AHT2X Dew point	13.1 °C

```

sfromis, 2023-04-16:
In addition to building to make the needed driver available, It is not unexpected that a [sensor driver](https://tasmota.github.io/docs/I2CDEVICES/) may react to the wrong sensor when they use the same I2C address. The general "cure" for such is to disable to unwanted sensor driver, using the I2CDriver command, like `I2CDriver12 0` to disable the VEML6070 driver.

Thus, there is no issue to be fixed.

closed 2023-04-16 by Jason2866

MrPedalkin, 2023-09-14:
Hello. 
I'm trying to connect the AHT20+BMP280 sensor to Tasmata 13.1.0 "Sensors". The "status 0" command does not show the number 43 in the list of available drivers. As a result, the BMP280 sensor works, but the AHT 20 does not work. When I experimented, I disabled all the drivers, but it didn't help.

01:58:42.527 MQT: stat/sonoff_th10/STATUS = {"Status":{"Module":4,"DeviceName":"Sonoff","FriendlyName":["Sonoff"],"Topic":"sonoff_th10","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
01:58:42.532 MQT: stat/sonoff_th10/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"https://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:12:57","StartupUTC":"2023-09-14T00:45:45","Sleep":50,"CfgHolder":4617,"BootCount":4190,"BCResetTime":"2023-08-09T05:51:37","SaveCount":4369,"SaveAddress":"FB000"}}
01:58:42.536 MQT: stat/sonoff_th10/STATUS2 = {"StatusFWR":{"Version":"13.1.0(sensors)","BuildDateTime":"2023-08-15T07:59:54","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"355/699"}}
01:58:42.541 MQT: stat/sonoff_th10/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["New",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A64002800000000","00000080","00006000","00004000","00000000"]}}
01:58:42.559 MQT: stat/sonoff_th10/STATUS4 = {"StatusMEM":{"ProgramSize":700,"Free":300,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1420C2","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66,68","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"!7,!8,!9,!10,!11,!12,!13,!14,!15,!17,!18,!20,!24,!29,!31,!36,!41,!42,!44,!46,!48,!69,!76"}}
01:58:42.568 MQT: stat/sonoff_th10/STATUS5 = {"StatusNET":{"Hostname":"sonoff-th10-2216","IPAddress":"192.168.254.251","Gateway":"192.168.254.254","Subnetmask":"255.255.255.0","DNSServer1":"172.16.1.1","DNSServer2":"0.0.0.0","Mac":"CC:50:E3:52:E8:A8","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
01:58:42.573 MQT: stat/sonoff_th10/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.19.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_52E8A8","MqttUser":"AndyMQTT","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
01:58:42.579 MQT: stat/sonoff_th10/STATUS7 = {"StatusTIM":{"UTC":"2023-09-14T00:58:42","Local":"2023-09-14T01:58:42","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"06:25","Sunset":"19:06"}}
01:58:42.586 MQT: stat/sonoff_th10/STATUS10 = {"StatusSNS":{"Time":"2023-09-14T01:58:42"}}
01:58:42.592 MQT: stat/sonoff_th10/STATUS11 = {"StatusSTS":{"Time":"2023-09-14T01:58:42","Uptime":"0T00:12:57","UptimeSec":777,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"New","BSSId":"4E:5E:0C:52:E7:D7","Channel":3,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:03"}}}

HofeBY, 2023-09-14:
@MrPedalkin
"i2cdriver" don´t show driver 43, so you have to compile your own version of "tasmota.bin" with included driver "AHT20"

daniboube, 2023-10-13:
Hi, we are having the same problem in our project. We want to use the AHT2X sensor on a ESP32C3 chip (Tasmota version 13.1.0), but there is no way to make it works. Tasmota does not detect the AHT2X even though the #define USE_AHT2x line is defined and the VEML6070 sensor is disabled. Is this a bug or we need to do or add something else? This used to work in the previous Tamsota 12 version.

P.D: I2CScan command detects a sensor in the 0x38 address

barbudor, 2023-10-13:
I2cdriver

sfromis, 2023-10-13:
Right (same number for AHT1x too). Hence your build does not include the driver.

daniboube, 2023-10-13:
I have disabled the sensors that conflict with the AHT"X and the TSL2591 (the other sensor i want to use) with the following lines in the user_config_override file:

#undef I2CDRIVERS_0_31
#undef I2CDRIVERS_32_63
#undef I2CDRIVERS_64_95

#define I2CDRIVERS_0_31  0x7FFFEFF7
#define I2CDRIVERS_32_63 0xFFBFFF3F
#define I2CDRIVERS_64_95 0xFFFFFFFD

And then, add this line 
#define USE_AHT2X

This worked in tasmota 12. What I am missing?

barbudor, 2023-10-13:
Changing that in your binary build only applies if your flashing the firmware on a black target (or perform a full flash erase)
On upgrading, this is not taken into account so check the result of the i2cdriver command and disable what you don't need

daniboube, 2023-10-13:
Yes, I'm doing a full erase before flashing the factory binary. I do not know what else can I try...

barbudor, 2023-10-13:
Ah I missed that you did that and that 43 is not there
So indeed bad build or missed upgrade

sfromis, 2023-10-13:
Instead of `#define USE_AHT2X` you should `#define USE_AHT2x`

daniboube, 2023-10-13:
Yep, that seems to be the problem, my bad...
Thank you guys!

JRzen, 2024-02-06:
Found this comment: https://github.com/arendst/Tasmota/discussions/20163#discussioncomment-7922915
And that change has worked with my AHT20.