arendst/Tasmota #18442: ShutterStop not triggering relais in garage mode (ShutterMode 3)

Venwa-IT, 2023-04-17:
### PROBLEM DESCRIPTION
I am using the ShutterMode3.
This means one pulse for start and one pulse for stop (or power off if door has reached end position).

The relais is triggered fine if I change the position of the blinds with ShutterPosition XX. (Changing of direction is also done perfectly) The relais also work as expected if I only use ShutterOpen and ShutterClose.

The Problem is ShutterStop. This should trigger the relais if it is still moving to stop the door. Instead it only stops the internal counter and is not triggering the relais.

If a button is configured to handle the door (ShutterButton1 1 toggle 0), the door should stop if the button is pressed while the door is moving. Same problem as ShutterStop. The relais is not triggered - it only stops the internal counter.

Also:
ShutterToggle, ShutterStopClose, ShutterStopOpen, ShutterStopXX not working properly while door is moving

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NodeMCU (ESP8266EX)
- [X] Tasmota binary firmware version number used: 12.4.0.5
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA-Update
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

22:25:30.103 MQT: stat/tasmota_DEEF42/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:25:30.333 MQT: stat/tasmota_DEEF42/RESULT = {"Module":{"0":"Generic"}}
22:25:30.588 MQT: stat/tasmota_DEEF42/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"256":"Relay_i1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"161":"Switch2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

22:26:19.421 MQT: stat/tasmota_DEEF42/STATUS = {"Status":{"Module":0,"DeviceName":"Testmota","FriendlyName":["Testmota"],"Topic":"tasmota_DEEF42","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,2,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:26:19.427 MQT: stat/tasmota_DEEF42/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:14:58","StartupUTC":"2023-04-17T20:11:21","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2023-04-17T17:40:22","SaveCount":111,"SaveAddress":"FB000"}}
22:26:19.435 MQT: stat/tasmota_DEEF42/STATUS2 = {"StatusFWR":{"Version":"12.4.0.5(tasmota)","BuildDateTime":"2023-04-16T21:17:14","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"342/699"}}
22:26:19.442 MQT: stat/tasmota_DEEF42/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Level4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C80005000600003C5A0A192800000000","40000080","00006000","00004001","00000000"]}}
22:26:19.456 MQT: stat/tasmota_DEEF42/STATUS4 = {"StatusMEM":{"ProgramSize":633,"Free":368,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164054","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
22:26:19.465 MQT: stat/tasmota_DEEF42/STATUS5 = {"StatusNET":{"Hostname":"tasmota-DEEF42-3906","IPAddress":"192.168.110.194","Gateway":"192.168.110.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.110.254","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:DE:EF:42","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:26:19.471 MQT: stat/tasmota_DEEF42/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.110.55","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DEEF42","MqttUser":"any","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:26:19.478 MQT: stat/tasmota_DEEF42/STATUS7 = {"StatusTIM":{"UTC":"2023-04-17T20:26:19","Local":"2023-04-17T22:26:19","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:55","Sunset":"20:43"}}
22:26:19.484 MQT: stat/tasmota_DEEF42/STATUS10 = {"StatusSNS":{"Time":"2023-04-17T22:26:19","Switch2":"OFF","Shutter1":{"Position":63,"Direction":0,"Target":63,"Tilt":0}}}
22:26:19.490 MQT: stat/tasmota_DEEF42/STATUS11 = {"StatusSTS":{"Time":"2023-04-17T22:26:19","Uptime":"0T00:14:58","UptimeSec":898,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level4","BSSId":"DC:39:6F:73:CE:DC","Channel":11,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}
22:26:19.494 MQT: stat/tasmota_DEEF42/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"1000","Calib":[300,500,700,900,1000],"Mode":"3"}}}
22:26:29.447 MQT: tele/tasmota_DEEF42/STATE = {"Time":"2023-04-17T22:26:29","Uptime":"0T00:15:08","UptimeSec":908,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level4","BSSId":"DC:39:6F:73:CE:DC","Channel":11,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:03"}}
22:26:29.454 MQT: tele/tasmota_DEEF42/SENSOR = {"Time":"2023-04-17T22:26:29","Switch2":"OFF","Shutter1":{"Position":63,"Direction":0,"Target":63,"Tilt":0}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Config:

SetOption80 1
restart 1
ShutterMode 3
ShutterRelay1 1
Pulsetime 5
restart 1

Cmd: ShutterOpen
and while moving: ShutterStop

Relais is not triggered to stop the door

### EXPECTED BEHAVIOUR
Relais should be pulsed to stop door

### SCREENSHOTS
22:45:06.551 CMD: ShutterClose

Reails is pulsed once

22:45:06.561 MQT: stat/tasmota_DEEF42/RESULT = {"POWER":"ON"}
22:45:06.564 MQT: stat/tasmota_DEEF42/POWER = ON
22:45:06.570 MQT: stat/tasmota_DEEF42/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0,"Tilt":0}}
22:45:07.154 MQT: stat/tasmota_DEEF42/RESULT = {"POWER":"OFF"}
22:45:07.159 MQT: stat/tasmota_DEEF42/POWER = OFF
22:45:07.455 MQT: stat/tasmota_DEEF42/RESULT = {"Shutter1":{"Position":92,"Direction":-1,"Target":0,"Tilt":0}}
22:45:08.459 MQT: stat/tasmota_DEEF42/RESULT = {"Shutter1":{"Position":82,"Direction":-1,"Target":0,"Tilt":0}}
22:45:09.412 MQT: stat/tasmota_DEEF42/RESULT = {"Shutter1":{"Position":72,"Direction":-1,"Target":0,"Tilt":0}}
22:45:10.417 MQT: stat/tasmota_DEEF42/RESULT = {"Shutter1":{"Position":62,"Direction":-1,"Target":0,"Tilt":0}}
22:45:11.420 MQT: stat/tasmota_DEEF42/RESULT = {"Shutter1":{"Position":52,"Direction":-1,"Target":0,"Tilt":0}}
22:45:12.421 MQT: stat/tasmota_DEEF42/RESULT = {"Shutter1":{"Position":42,"Direction":-1,"Target":0,"Tilt":0}}
22:45:12.572 CMD: ShutterStop

Relais is not pulsed (so the door keeps moving, although the internal position is 41

22:45:12.581 MQT: stat/tasmota_DEEF42/SHUTTER1 = 41
22:45:12.585 MQT: stat/tasmota_DEEF42/RESULT = {"Shutter1":{"Position":41,"Direction":0,"Target":41,"Tilt":0}}

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-04-27:
Thanks for bringing this up. Will check where the mess is generated.

stefanbode, 2023-04-27:
Hmm works on my side. 07:58:30.590 relay is triggered once. AutoStop after pulsetime: 07:58:31.195
07:58:33.873 triggered again after stop. 

```
07:58:30.539 CMD: shutteropen
07:58:30.540 SRC: Serial
07:58:30.541 CMD: Grp 0, Cmd 'SHUTTEROPEN', Idx 1, Len 0, Pld -99, Data ''
07:58:30.542 SHT: Pos. payload <> (0), payload 100, idx 1 (0), src 7
07:58:30.553 SHT: lastsource 7:, real 0, target 200000, tiltreal: 0, tilttarget: 0, payload 100
07:58:30.563 SHT: Delay Start? var1 <99>=<>, max10s?
07:58:30.564 SRC: Shutter
07:58:30.564 SHT: FUNC_SET_POWER Relaymask 0 SwitchedRelay:1 by Shutter, payload 100, powermask 1
07:58:30.576 SHT: FUNC_SET_DEVICE_POWER end. powermask 1, current rel: 1
07:58:30.590 RSL: RESULT = {"POWER1":"ON"}
07:58:30.591 RSL: POWER1 = ON
07:58:30.592 SHT: Shtr1 Real 0, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
07:58:30.603 RSL: RESULT = {"Shutter1":{"Position":0,"Direction":1,"Target":100,"Tilt":0}}
07:58:30.618 SHT: Shtr1 Time 1(1), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 0, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:30.667 SHT: Shtr1 Time 2(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 1000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:30.716 SHT: Shtr1 Time 3(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 2000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:30.766 SHT: Shtr1 Time 4(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 3000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:30.816 SHT: Shtr1 Time 5(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 4000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:30.866 SHT: Shtr1 Time 6(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 5000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:30.916 SHT: Shtr1 Time 7(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 6000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:30.966 SHT: Shtr1 Time 8(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 7000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:30.986 SHT: Shtr1 Real 7000, Start 0, Stop 200000, Dir 1, Delay 0, Rtc 0.40 [s], Freq 0, PWM 0, Tilt 0
07:58:30.987 RSL: RESULT = {"Shutter1":{"Position":4,"Direction":1,"Target":100,"Tilt":0}}
07:58:31.017 SHT: Shtr1 Time 9(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 8000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.066 SHT: Shtr1 Time 10(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 9000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.117 SHT: Shtr1 Time 11(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 10000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.166 SHT: Shtr1 Time 12(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 11000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.180 SRC: PulseTimer
07:58:31.180 SHT: FUNC_SET_POWER Relaymask 1 SwitchedRelay:1 by PulseTime, payload 23, powermask 0
07:58:31.192 SHT: FUNC_SET_DEVICE_POWER end. powermask 0, current rel: 0
07:58:31.195 RSL: RESULT = {"POWER1":"OFF"}
07:58:31.196 RSL: POWER1 = OFF
07:58:31.218 SHT: Shtr1 Time 13(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 12000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.266 SHT: Shtr1 Time 14(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 13000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.317 SHT: Shtr1 Time 15(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 14000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.366 SHT: Shtr1 Time 16(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 15000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.416 SHT: Shtr1 Time 17(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 16000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.466 SHT: Shtr1 Time 18(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 17000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.516 SHT: Shtr1 Time 19(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 18000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.566 SHT: Shtr1 Time 20(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 19000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.616 SHT: Shtr1 Time 21(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 20000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.666 SHT: Shtr1 Time 22(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 21000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.716 SHT: Shtr1 Time 23(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 22000, nStop 0, Trgt 200000, mVelo 0, Dir 1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
07:58:31.730 WIF: Checking connection...
[... cut ...]
07:58:33.887 RSL: RESULT = {"Shutter1":{"Position":33,"Direction":0,"Target":32,"Tilt":0}}
07:58:34.458 SRC: PulseTimer
07:58:34.458 SHT: FUNC_SET_POWER Relaymask 1 SwitchedRelay:1 by PulseTime, payload 23, powermask 0
07:58:34.459 SHT: FUNC_SET_DEVICE_POWER end. powermask 0, current rel: 0
07:58:34.473 RSL: RESULT = {"POWER1":"OFF"}
07:58:34.474 RSL: POWER1 = OFF
07:58:34.479 SHUTTER: About to save settings to file /.drvset027
07:58:51.707 WIF: Checking connection...
07:59:11.712 WIF: Checking connection...
```

stefanbode, 2023-04-27:
Works also in direction -1.

```
08:03:41.380 SHT: Shtr1 Time 44(-1), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 22000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
08:03:41.430 SHT: Shtr1 Time 45(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 21000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
08:03:41.480 SHT: Shtr1 Time 46(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 20000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
08:03:41.530 SHT: Shtr1 Time 47(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 19000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
08:03:41.580 SHT: Shtr1 Time 48(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 18000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
08:03:41.605 CMD: shutterstop
08:03:41.606 SRC: Serial
08:03:41.607 CMD: Grp 0, Cmd 'SHUTTERSTOP', Idx 1, Len 0, Pld -99, Data ''
08:03:41.609 SHT: Stop 1: dir: -1
08:03:41.609 SHT: Shtr1 Time 49(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 17000, nStop 0, Trgt 17000, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
08:03:41.630 SHT: Stop 1 Mode 1 time 49
08:03:41.630 SRC: Shutter
08:03:41.631 SHT: FUNC_SET_POWER Relaymask 0 SwitchedRelay:1 by Shutter, payload -99, powermask 1
08:03:41.642 SHT: FUNC_SET_DEVICE_POWER end. powermask 1, current rel: 1
08:03:41.656 RSL: RESULT = {"POWER1":"ON"}
08:03:41.657 RSL: POWER1 = ON
08:03:41.659 SHT: Shtr1 Real 17000, Start 17000, Stop 17000, Dir 0, Delay 0, Rtc 2.45 [s], Freq 0, PWM 0, Tilt 0
08:03:41.670 RSL: SHUTTER1 = 9
08:03:41.671 RSL: RESULT = {"Shutter1":{"Position":9,"Direction":0,"Target":9,"Tilt":0}}
08:03:42.239 SRC: PulseTimer
08:03:42.239 SHT: FUNC_SET_POWER Relaymask 1 SwitchedRelay:1 by PulseTime, payload 23, powermask 0
08:03:42.240 SHT: FUNC_SET_DEVICE_POWER end. powermask 0, current rel: 0
08:03:42.254 RSL: RESULT = {"POWER1":"OFF"}
08:03:42.255 RSL: POWER1 = OFF
08:03:42.442 SHUTTER: About to save settings to file /.drvset027
08:03:51.710 WIF: Checking connection...
```

stefanbode, 2023-04-27:
Maybe a logfile with loglevel 4 will give some more hints whats wrong with your configuration

Venwa-IT, 2023-04-29:
I think the problem is only present if the command is issued via the WebConsole.

12:52:35.726 CMD: ShutterClose
12:52:35.728 SRC: WebConsole from 192.168.110.175
12:52:35.731 CMD: Grp 0, Cmd 'SHUTTERCLOSE', Idx 1, Len 0, Pld -99, Data ''
12:52:35.735 SHT: Pos. payload <> (0), payload 0, idx 1 (0), src 7
12:52:35.737 SHT: lastsource 7:, real 200000, target 0, tiltreal: 0, tilttarget: 0, payload 0
12:52:35.739 SHT: Delay Start? var1 <99>=<>, max10s?
12:52:35.740 SRC: Shutter
12:52:35.743 SHT: FUNC_SET_POWER Relaymask 0 SwitchedRelay:1 by Shutter, payload 0, powermask 1
12:52:35.744 SHT: FUNC_SET_POWER end. powermask 1
12:52:35.748 RSL: RESULT = {"POWER1":"ON"}
12:52:35.750 RSL: POWER1 = ON
12:52:35.753 SHT: Shtr1 Real 200000, Start 200000, Stop 0, Dir -1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0, Tilt 0
12:52:35.756 RSL: RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0,"Tilt":0}}
...
12:52:39.558 SHT: Shtr1 Time 77(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 124000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0

>>>>>

12:52:39.613 CMD: ShutterStop
12:52:39.615 SRC: WebConsole from 192.168.110.175
12:52:39.618 CMD: Grp 0, Cmd 'SHUTTERSTOP', Idx 1, Len 0, Pld -99, Data ''
12:52:39.621 SHT: Stop 1: dir: -1
12:52:39.623 SHT: Shtr1 Time 78(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 123000, nStop 0, Trgt 123000, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
12:52:39.625 SHT: Stop 1 Mode 1 time 78
12:52:39.628 SHT: Shtr1 Real 123000, Start 123000, Stop 123000, Dir 0, Delay 0, Rtc 3.90 [s], Freq 0, PWM 0, Tilt 0
12:52:39.631 RSL: SHUTTER1 = 62
12:52:39.634 RSL: RESULT = {"Shutter1":{"Position":62,"Direction":0,"Target":62,"Tilt":0}}
12:52:40.634 CFG: Saved to flash at FA, Count 89, Bytes 4096

After ShutterStop via WebConsole is no pulse of the relais.

If the same commands are issued via MTQ everything works as designed. If ShutterStop is issued via a Rule it also works fine.

Turns out, this is no problem for me. I wanted to test the commands in the console so I could later use them in rules.

I am not closing this, because it is still kinda a bug, right? (Or should I?)

stefanbode, 2023-04-30:
No let it open. I will Check for the webconsole issue. It is quite unusual but I do not want to exclude

PR #18572 (2023-05-03): BugFix Garage + new shutterbuttons options
## Description:
- Removed logging
- shutterbuttons support incremental change
- garage mode, fix #18539 and fix #18442
**Related issue (if applicable):** fixes #<Tasmota issue number goes here>

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ES [... cut]
commit 9097f508 (2023-05-03): BugFix Garage + new shutterbuttons options (#18572)

* Fix GarageMode movements

* BugFix Garage + new shutterbuttons options

- Removed logging
- shutterbuttons support incremental change
- garage mode, fix #18539 and fix #18442
closed 2023-05-03 by arendst