arendst/Tasmota #18446: SDM630 with WeMoS D1 Pro can't provide Import Energy (on the WWW dashboard)

panciunio, 2023-04-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
SDM630 Power Meter connected to WeMoS D1 Pro doesn't provide Import Active on each phase.   

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____ WeMoS D1 Pro
- [X] Tasmota binary firmware version number used: _____ 12.5.0.1
  - [X] Pre-compiled
  - [X] Self-compiled
- [X Flashing tools used: _____ OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`: 

```lua
  Configuration output here:
13:05:33.622 CMD: Backlog Template; Module; GPIO 255`
13:05:33.662 MQT: stat/SDM630/RESULT = {"NAME":"WEMOS D1 PRO","GPIO":[0,0,0,0,1984,2016,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
13:05:33.875 MQT: stat/SDM630/RESULT = {"Module":{"0":"WEMOS D1 PRO"}}
13:05:34.081 MQT: stat/SDM630/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"1984":"SDM630 Tx"},"GPIO5":{"2016":"SDM630 Rx"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:06:53.459 CMD: Backlog Rule1; Rule2; Rule3
13:06:53.496 MQT: stat/SDM630/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:06:53.724 MQT: stat/SDM630/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:06:53.925 MQT: stat/SDM630/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:07:20.362 CMD: status 0
13:07:20.371 MQT: stat/SDM630/STATUS = {"Status":{"Module":0,"DeviceName":"SDM630","FriendlyName":["SDM630"],"Topic":"SDM630","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
13:07:20.405 MQT: stat/SDM630/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T00:00:45","StartupUTC":"2023-04-18T12:06:35","Sleep":50,"CfgHolder":4617,"BootCount":495,"BCResetTime":"2021-06-06T16:00:29","SaveCount":532,"SaveAddress":"FB000"}}
13:07:20.441 MQT: stat/SDM630/STATUS2 = {"StatusFWR":{"Version":"12.5.0(sensors)","BuildDateTime":"2023-04-17T08:03:55","Boot":4,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"362/699"}}
13:07:20.460 MQT: stat/SDM630/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Kameleon2_MiWiFi2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["40028009","2805C80001000600003C5A0A002800000000","00000288","00006000","00004000","00000000"]}}
13:07:20.502 MQT: stat/SDM630/STATUS4 = {"StatusMEM":{"ProgramSize":699,"Free":304,"Heap":14,"ProgramFlashSize":1024,"FlashSize":16384,"FlashChipId":"1840EF","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
13:07:20.553 MQT: stat/SDM630/STATUS5 = {"StatusNET":{"Hostname":"SDM630","IPAddress":"192.168.200.10","Gateway":"192.168.200.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.200.2","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:A6:18:1E","Webserver":2,"HTTP_API":1,"WifiConfig":0,"WifiPower":17.0}}
13:07:20.577 MQT: stat/SDM630/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.200.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A6181E","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:07:20.600 MQT: stat/SDM630/STATUS7 = {"StatusTIM":{"UTC":"2023-04-18T12:07:20","Local":"2023-04-18T13:07:20","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:53","Sunset":"19:45"}}
13:07:20.618 MQT: stat/SDM630/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:07:20.643 MQT: stat/SDM630/STATUS10 = {"StatusSNS":{"Time":"2023-04-18T13:07:20","ENERGY":{"TotalStartTime":"2021-06-06T16:00:29","Total":85.485,"Yesterday":12.352,"Today":0.106,"TodaySumImport":0.000,"TodaySumExport":0.000,"ExportActive":[2468.866,2143.309,1667.846],"Power":[166,-475,-219],"ApparentPower":[505,545,369],"ReactivePower":[-436,-225,-251],"Factor":[0.33,-0.91,-0.68],"Frequency":50,"Voltage":[235,238,236],"Current":[2.150,2.295,1.566]}}}
13:07:20.681 MQT: stat/SDM630/STATUS11 = {"StatusSTS":{"Time":"2023-04-18T13:07:20","Uptime":"0T00:00:45","UptimeSec":45,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Kameleon2_MiWiFi2","BSSId":"18:E8:29:A1:00:07","Channel":1,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:09"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:07:55.492 CMD: weblog 4
13:07:55.498 MQT: stat/SDM630/RESULT = {"WebLog":4}
13:07:55.531 DMP: 01 04 04 46 BF C7 88 8D 7E
13:07:55.533 NRG: EnergyTotal[0] 7377.0825 kWh
13:07:55.535 NRG: EnergyTotal[1] 2926.3408 kWh
13:07:55.537 NRG: EnergyTotal[2] 7964.3130 kWh
13:07:55.755 DMP: 01 04 04 43 6B 70 8B FA 7B
13:07:56.007 DMP: 01 04 04 43 6D EC B8 33 6F
13:07:56.261 DMP: 01 04 04 43 6B D8 40 C4 2C
13:07:56.513 DMP: 01 04 04 40 04 AA 8B 91 42
13:07:56.766 DMP: 01 04 04 40 1A 8A B3 E9 56
13:07:57.018 DMP: 01 04 04 3F D5 54 6F 98 84
13:07:57.270 DMP: 01 04 04 42 EC 9C CE C6 9D
13:07:57.522 DMP: 01 04 04 C3 FC FE 91 86 3C
[... 65 lines cut ...]

        case 20:
          Energy->import_active[1] = value;
          break;

        case 21:
          Energy->import_active[2] = value;
          break;
```

look's good as well...

Compiled, flashed... result? the same...
[image]

No Import Active visible...

I tried to dig deeper... found 'configurations.md' file into 'energy_modbus_config' branch. Very interesting file which contains JSON structure for each data from energy measurement device, I founf SDM630 section and discovered that ImportActive variable is hidden into "User" section:

`SDM630             {"Name":"SDM630","Baud":9600,"Config":"8N1","Address":1,"Function":4,"Voltage":[0,2,4],"Current":[6,8,10],"Power":[12,14,16],"ApparentPower":[18,20,22],"ReactivePower":[24,26,28],"Factor":[30,32,34],"Frequency":70,"Total":342,"ExportActive":[352,354,356],**"User":{"R":[346,348,350],"J":"ImportActive","G":"Import Active","U":"kWh","D":24}**}`

The question is... how to check data from table "User" under Tasmota console and how to provide these information to MQTT if these are available?

Best Regards,
panciunio

arendst, 2023-04-18:
SDM630 ImportActive =Energy Total.

If you want more registers you will need to start using the new user configurable modbus driver. See https://github.com/arendst/Tasmota/pull/18413#issuecomment-1510131402

This is also more future proof where dedicated modbus drivers will be removed someday.

panciunio, 2023-04-18:
Hi.  Thanks for quick reply,

Yes.. I agree but I want to see Import Active for each phase... Thanks for tip I will write my own modbus .json dedicated to SDM630 but I thought, the section:

`SDM630             {"Name":"SDM630","Baud":9600,"Config":"8N1","Address":1,"Function":4,"Voltage":[0,2,4],"Current":[6,8,10],"Power":[12,14,16],"ApparentPower":[18,20,22],"ReactivePower":[24,26,28],"Factor":[30,32,34],"Frequency":70,"Total":342,"ExportActive":[352,354,356],"User":{"R":[346,348,350],"J":"ImportActive","G":"Import Active","U":"kWh","D":24}}`

from configurations.md file means the same... but I'm not sure how to inform compiler that should use this particular section.

arendst, 2023-04-18:
Read the docs: ``SetOption129 0/1``

When you use new modbus driver you do not need to tell the compiler any config data. It's entered at runtime. Again, see link above.

panciunio, 2023-04-18:
Okay, because I'm using small 8266 device I decided to use 'rule' method. So in case I added Rule3 as follow:
 
```
on file#modbus do
	{"Name":"SDM630","Baud":9600,"Config":"8N1","Address":1,"Function":4,"Voltage":[0,2,4],"Current":[6,8,10],"Power":[12,14,16],"ApparentPower":[18,20,22],"ReactivePower":[24,26,28],"Factor":[30,32,34],"Frequency":70,"Total":342,"ExportActive":[346,348,350],"User":{"R":[346,348,350],"J":"ImportActive","G":"Import Active","U":"kWh","D":24}}
endon
```
This is exactly the same configuration which is into 'configurations.md' file for SDM630
There is part reflecting to ImportActive for each phase and as I understood correctly - should be visible on dashboard ("G": "Import Active").  I'm sorry, I'm trying to follow documentation and do what is required but i still have lack of knowledge... Could you please explain more with more details what I'm doing wrong?

closed 2023-04-19 by panciunio

jeroenst, 2024-12-01:
Same issue for Shelly 3EM missing Import Active, so not suitable for measuring a phases containing solar or ess.