arendst/Tasmota #18455: Deepsleep problem

reimbe, 2023-04-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Deepsleep is entered every second time only, or after logging time is expired.
Deepsleeptime is set 30 seconds, logging to 45 seconds (that short only for debugging purposes only) 
When deepsleeptime is set initially deepsleep is entered. After wakeup, the device remains active until logging interval is expired. Then it enters deepsleep again. 
This behaviour does not exist in Ver 11.0.0.0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini, Wemos D1 mini
- [x] Tasmota binary firmware version number used: 12.5.0.1
  - [ ] Pre-compiled
  - [X] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:30:37.214 MQT: stat/tasmota_EF27D2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:30:37.425 MQT: stat/tasmota_EF27D2/RESULT = {"Module":{"18":"Generic"}}
09:30:37.631 MQT: stat/tasmota_EF27D2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
none

```
- [x] Provide the output of this command: `Status 0`:
```lua
STATUS 0 output here:
09:31:57.440 MQT: stat/tasmota_EF27D2/STATUS0 = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_EF27D2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:00:23","StartupUTC":"2023-04-19T07:31:34","Sleep":50,"CfgHolder":4618,"BootCount":2,"BCResetTime":"2023-04-19T09:20:03","SaveCount":12,"SaveAddress":"F8000"},"StatusFWR":{"Version":"12.5.0.1(sensors)","BuildDateTime":"2023-04-19T09:13:07","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"388/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["reim29a","reim"],"TelePeriod":45,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","000000C0","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":708,"Free":292,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62,65,66","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"},"StatusNET":{"Hostname":"tasmota-EF27D2-2002","IPAddress":"192.168.5.254","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.5.1","DNSServer2":"0.0.0.0","Mac":"D8:BF:C0:EF:27:D2","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.5.41","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EF27D2","MqttUser":"reimbe","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2023-04-19T07:31:57","Local":"2023-04-19T09:31:57","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":99,"Sunrise":"06:28","Sunset":"20:25"},"StatusSNS":{"Time":"2023-04-19T09:31:57"},"StatusSTS":{"Time":"2023-04-19T09:31:57","Uptime":"0T00:00:23","UptimeSec":23,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"reim29a","BSSId":"DC:39:6F:F9:EA:CC","Channel":6,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F5, Count 15
00:00:00.056 CFG: CR 388/699, Busy 0
00:00:00.069 ROT: Mode 1
00:00:00.071 SRC: Restart
00:00:00.073 Project tasmota - Tasmota Version 12.5.0.1(sensors)-2_7_4_9(2023-04-19T09:13:07)
00:00:00.229 WIF: Checking connection...
00:00:00.230 WIF: Attempting connection...
00:00:00.232 WIF: Network (re)scan started...
00:00:01.131 WIF: Checking connection...
00:00:01.132 WIF: Attempting connection...
00:00:02.136 WIF: Checking connection...
00:00:02.137 WIF: Attempting connection...
00:00:03.143 WIF: Checking connection...
00:00:03.144 WIF: Attempting connection...
00:00:03.145 WIF: Network 0, AP-, SSId lzhqwlan01, Channel 1, BSSId 68:D7:9A:CA:ED:51, RSSI -75, Encryption 1
00:00:03.146 WIF: Network 1, AP-, SSId lzhq-wlan02, Channel 1, BSSId 6E:D7:9A:CA:ED:51, RSSI -73, Encryption 1
00:00:03.147 WIF: Network 2, AP1, SSId reim29a, Channel 1, BSSId 0C:72:74:DD:86:23, RSSI -85, Encryption 1
00:00:03.148 WIF: Network 3, AP-, SSId WLAN-776530, Channel 1, BSSId 18:83:BF:07:57:75, RSSI -80, Encryption 1
00:00:03.149 WIF: Network 4, AP1, SSId reim29a, Channel 6, BSSId DC:39:6F:F9:EA:CC, RSSI -57, Encryption 1
00:00:03.150 WIF: Network 5, AP-, SSId betha, Channel 6, BSSId DC:15:C8:5E:76:C0, RSSI -91, Encryption 1
00:00:03.151 WIF: Network 6, AP-, SSId lzhq-wlan02, Channel 11, BSSId 7A:45:58:16:A2:55, RSSI -85, Encryption 1
00:00:03.152 WIF: Network 7, AP-, SSId lzhqwlan01, Channel 11, BSSId 78:45:58:46:A2:55, RSSI -85, Encryption 1
00:00:03.489 WIF: Connecting to AP1 reim29a Channel 6 BSSId DC:39:6F:F9:EA:CC in mode 11n as tasmota-EF27D2-2002...
00:00:04.756 WIF: Checking connection...
00:00:04.757 WIF: Connected
00:00:05.013 HTP: Web server active on tasmota-EF27D2-2002 with IP address 192.168.5.254
00:00:05.464 WIF: Sending Gratuitous ARP
00:00:05.465 NTP: Sync time...
[... 11 lines cut ...]
09:41:16.466 MQT: tele/tasmota_EF27D2/STATE = {"Time":"2023-04-19T09:41:16","Uptime":"0T00:00:12","UptimeSec":12,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"reim29a","BSSId":"DC:39:6F:F9:EA:CC","Channel":6,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:41:30.760 WIF: Checking connection...
09:41:50.766 WIF: Checking connection...
09:42:01.463 MQT: tele/tasmota_EF27D2/STATE = {"Time":"2023-04-19T09:42:01","Uptime":"0T00:00:57","UptimeSec":57,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"reim29a","BSSId":"DC:39:6F:F9:EA:CC","Channel":6,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:42:01.494 MQT: stat/tasmota_EF27D2/RESULT = {"DeepSleep":{"Time":"2023-04-19T09:42:32","Epoch":1681897352}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Tested in 12.4. and 12.5. master and development

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
In the log after 09:41:16.466 the device should enter deepsleep
Worked fine in Version 11.0.0.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
none

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
none

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2023-04-19:
@stefanbode can you have a look?

stefanbode, 2023-04-20:
Correct. 10 seconds is a hard coded "as soon as possible". Any other value will keep the device up and running. This was implemented long time ago when some auxiliary devices need 30 seconds to warm up after deepsleep.

reimbe, 2023-04-20:
Hello !
Thanks for the clarification.
I am sorry, but I've been working with deepsleep since it was still a 
fork, and something is different in Version 12 and not quite right.

Usually I set deepsleeptime to 600 and logging to 600. Up to version 11 
it behaved as I expect it, it wakes up every 10 minutes, at the precise 
time announced, and goes back to sleep each time.

With Version 12 it's different.
Following the latest hints from Barbudor I set deepsleeptime to 600 and 
logging to 300. It helps in as much as the device now goes to sleep 
every cycle.
But the wakeup times are arbitrary.

12:06:02.794 CMD: deepsleeptime 600
12:06:02.797 SRC: WebConsole from 192.168.5.106
12:06:02.799 CMD: Grp 0, Cmd 'DEEPSLEEPTIME', Idx 1, Len 3, Pld 600, 
Data '600'
12:06:02.806 MQT: stat/tasmota_EF27D2/RESULT = {"DeepSleepTime":600}
12:06:03.533 CFG: Saved to flash at F9, Count 395, Bytes 4096
12:06:03.535 DSL: Reset wrong settings wakeup: 0, slip 10222
12:06:03.544 MQT: stat/tasmota_EF27D2/RESULT = 
{"DeepSleep":{"Time":"2023-04-20T12:10:30","Epoch":1681992630}}

12:19:52.484 MQT: stat/tasmota_EF27D2/RESULT = 
{"DeepSleep":{"Time":"2023-04-20T12:20:30","Epoch":1681993230}}

12:20:42.512 MQT: stat/tasmota_EF27D2/RESULT = 
{"DeepSleep":{"Time":"2023-04-20T12:30:30","Epoch":1681993830}}

12:58:40.504 MQT: stat/tasmota_EF27D2/RESULT = 
{"DeepSleep":{"Time":"2023-04-20T13:00:30","Epoch":1681995630}}

13:10:08.491 MQT: stat/tasmota_EF27D2/RESULT = 
{"DeepSleep":{"Time":"2023-04-20T13:20:30","Epoch":1681996830}}

Now it's 13:41 and the device did not wake up yet.

The announced wakeup times make sense, but the actual times when the 
device wakes up don't.

Best Regards
Benjamin Reim

Am 20.04.2023 um 12:56 schrieb stefanbode:

barbudor, 2023-04-20:
I still an outdoor device, running for more than 2 years out of a solar panel an LiIon battery with v9.5.0 and it is behaving as documented. So this not something new.

Having Teleperiod = DeepsleepTime is a non sense.
If you want to use a Teleperiod value which is not 10 or 300 (which both mean go-to-sleep-after-1st-teleperiod-message) you must have `DeepsleepTime >= Teleperiod + 30` 
Which is already useless as your device will be ON for 90% of the time.

I will redo some tests with your values this week-end but I haven't seen any change in behavior in DeepSleep since I've been using it.