arendst/Tasmota #18462: Rule cannot be deleted.

neophytemad1, 2023-04-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:44:50.271 MQT: stat/Temp_Jungs/STATUS = {"Status":{"Module":4,"DeviceName":"Temp Jungs","FriendlyName":["Temp Jungs"],"Topic":"Temp_Jungs","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:44:50.281 MQT: stat/Temp_Jungs/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://::1:80/tasmoadmin/data/firmwares/tasmota.bin","RestartReason":"Exception","Uptime":"0T02:47:13","StartupUTC":"2023-04-20T15:57:37","Sleep":50,"CfgHolder":4617,"BootCount":174,"BCResetTime":"2021-10-13T09:30:16","SaveCount":512343,"SaveAddress":"FB000"}}
19:44:50.289 MQT: stat/Temp_Jungs/STATUS2 = {"StatusFWR":{"Version":"12.1.0.1(tasmota)","BuildDateTime":"2022.08.23 07:33:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"364/699"}}
19:44:50.297 MQT: stat/Temp_Jungs/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["scharl mesh","Smart"],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["0000A009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
19:44:50.311 MQT: stat/Temp_Jungs/STATUS4 = {"StatusMEM":{"ProgramSize":625,"Free":376,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000407","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
19:44:50.319 MQT: stat/Temp_Jungs/STATUS5 = {"StatusNET":{"Hostname":"Temp-Jungs-6918","IPAddress":"192.168.0.54","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:9D:7B:06","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:44:50.326 MQT: stat/Temp_Jungs/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.103","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9D7B06","MqttUser":"mqtt.user","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:44:50.336 MQT: stat/Temp_Jungs/STATUS7 = {"StatusTIM":{"UTC":"2023-04-20T18:44:50","Local":"2023-04-20T19:44:50","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"05:49","Sunset":"19:48"}}
19:44:50.346 MQT: stat/Temp_Jungs/STATUS10 = {"StatusSNS":{"Time":"2023-04-20T19:44:50","SI7021":{"Temperature":24.0,"Humidity":50.2,"DewPoint":13.0},"TempUnit":"C"}}
19:44:50.355 MQT: stat/Temp_Jungs/STATUS11 = {"StatusSTS":{"Time":"2023-04-20T19:44:50","Uptime":"0T02:47:13","UptimeSec":10033,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Smart","BSSId":"AC:15:A2:BA:FC:14","Channel":10,"Mode":"11n","RSSI":100,"Signal":-32,"LinkCount":2,"Downtime":"0T00:00:04"}}}
19:44:50.370 MQT: stat/Temp_Jungs/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4028b090","4024144e","402659f6","40265aba","402657c1","40100a9d","40265d25","4024ef24","40276055","4027609a","4024db90","40000f49","40000f49","40000f49","40000e19","40105aa9","40105aaf","4010000d","4026ead0","4026ea81","40100a7c","401026f0","401059a3","40103287","40103468","40100a7c","40100a7c","4010598b","401026f0","40100a7c","40103287"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**
I deleted a rule with Rule1 "". However, it always restarts itself after a few hours. I can't change them either. The Rule1 that I had run for a year keeps coming up. Rule1 0 doesn't work either, it restarts anyway.

MaceMoneta, 2023-04-21:
This should be in discussions.

But the command to clear rule1 is:

rule1 "

(a single double-quote), not 

rule1 ""

(two double-quotes).

neophytemad1, 2023-04-21:
I've also tried it with just a double-quote. Don't work. If disabling the rule doesn't work either, I think that's a bug and therefore an issue.

sfromis, 2023-04-21:
If you have turned off saving of data (command `SaveData`), flash will not be updated, and next restart will revert.

neophytemad1, 2023-04-21:
That's an announcement! SaveData was set to OFF. However, I didn't set them to OFF. I have now activated it and deleted the rule again. I hope this fits now. Thank you very much!!!

sfromis, 2023-04-21:
If it was set to OFF, you **did** do that, no way around that :grin:

neophytemad1, 2023-04-21:
Ok maybe a While ago and can`t remember, maybe, only maybe. :) Thanks anyway. ;)